Marus/cortex-debug #575: Why use custom `MemoryRequest`s?

septatrix, 2022-01-18:
It seems like this extension uses a custom implementation named [`readMemoryRequestCustom`](https://github.com/Marus/cortex-debug/blob/cbfc8470db2e8440605341956ff5e13112c0bd2c/src/gdb.ts#L896) to read memory instead of using the standardized [`ReadMemoryRequest`](https://github.com/microsoft/vscode-debugadapter-node/blob/d76064871e4c542b304d518aef4eeacb7df490e5/protocol/src/debugProtocol.ts#L1549). Similar for the write variants.
Is there any particular reason to do this?

Also why does this extension have special "*Cortex* Peripherals"/"*Cortex* Registers" instead of using the normal ones without a prefix?

haneefdm, 2022-01-18:
Yes it does. History. I don't know if you are referring to the frontend or the backend? We own both and both the producer (debug adapter) and the consumer (rontend). Are you trying to write your own extension/frontend?

We helped MS add the memory request to their API. But we had implanted ours long before that. The new one doesn't add anything for us yet. We are the only client to this adapter. They are working on getting a proper memory viewer and when that happens, then we have to add the protocol version. It can wait until then. Other than changing the function name it doesn't change much. But we will lose functionality -- as in the spec does not allow an expression and we currently do.

Cortex-Registers have moved to the Local Variables and are under deprecation. You can see it in our pre-release right now.

Cortex-Peripheral: History. Mostly because it stems from the ARM specification for SVD which focussed on Peripherals. In your opinion, what does "normal ones without a prefix" mean? I mean the name of the extension is "Cortex-Debug"

We also needed to identify panels created by this extension vs the built-in ones.

haneefdm, 2022-01-18:
Similar things as memory could be said about "disassembly" as well. We have had it for years and MS is just now rolling it out and we are migrating to it. It is also in the preview releases that you have to subscribe.

septatrix, 2022-01-19:
> Yes it does. History. I don't know if you are referring to the frontend or the backend? We own both and both the producer (debug adapter) and the consumer (rontend). Are you trying to write your own extension/frontend?

No I am not writing my own frontend - at least not currently.

> We helped MS add the memory request to their API. But we had implanted ours long before that. The new one doesn't add anything for us yet. We are the only client to this adapter. They are working on [...]

Okay in that case it seems very reasonable.

> Cortex-Registers have moved to the Local Variables and are under deprecation. You can see it in our pre-release right now.

Thanks for the tip. I only started using the extension recently and will definitely check out the pre-release.

> Cortex-Peripheral: History. Mostly because it stems from the ARM specification for SVD which focussed on Peripherals. In your opinion, what does "normal ones without a prefix" mean? I mean the name of the extension is "Cortex-Debug"

I can also see panels with the name "Registers" and "Peripherals" though that might come from another extension. I though those were built in and shared by every extension.

> We also needed to identify panels created by this extension vs the built-in ones.

Not quite sure if I understand this correctly but I assume that you manually have to fill the panels as the functionality is not part of the debug-adapter-protocol and you get the correct element by the panel name?

> Similar things as memory could be said about "disassembly" as well. We have had it for years and MS is just now rolling it out and we are migrating to it. It is also in the preview releases that you have to subscribe.

Yes that is actually an issue I am already following closely (primarily the memory view - not so much the disassembly) and hope that they make good progress in that area.

<hr>

So in general it seems that these are mostly historical artefacts which are replaced as the respective interfaces in the DAP become stabilized? In that case thanks for clearing that up and I will definitely try pre-release in the future :)

haneefdm, 2022-01-19:
> Not quite sure if I understand this correctly but I assume that you manually have to fill the panels as the functionality is not part of the debug-adapter-protocol and you get the correct element by the panel name?

Kinda. We create and populate/refresh our panels (Cortex-Peripherals and Cortex-Registers). They totally belong to us. If you saw names like those without the prefix elsewhere, not sure whose extensions those are. And if you switch to another debugger (like MS cpptools or NodeJS or Python) our panels disappear. They only appear in the context of our debugger

> So in general it seems that these are mostly historical artefacts which are replaced as the respective interfaces in the DAP become stabilized? In that case thanks for clearing that up and I will definitely try pre-release in the future :)

Yup, we add stuff as time allows and need arises. We don't wait for MS but do help them move forward and merge when ready. I am the only developer here, btw, so it is not like we have an army of resources. They are currently implementing the disassembly and I gave them my implementation as a reference. Disassembly for native programs has its own challenges and I don't envy that. I have my own challenges dealing with half a dozen gdb-servers that they don't worry about.

closed 2022-01-19 by haneefdm