Marus/cortex-debug #576: What is the difference between this plugin and MS cpptools?

hongshui3000, 2022-01-19:
What is the difference between this plugin and MS cpptools? Is there a comparison table that can clearly let people know these?

haneefdm, 2022-01-19:
We are not competitors, just different needs and user bases. cpptools is more than a debugger, it also does Intellisense which we don't do but we work together. 

Perhaps someone can use them for embedded development for a month and this extension and write up a comparison table. You can even put it up in our Wiki

closed 2022-01-19 by haneefdm

hongshui3000, 2022-01-19:
Can I understand it this way, this plugin replaces some of the functions of MS tools and extends the embedded itself?

haneefdm, 2022-01-19:
No, we just provide a different debug adapter with different features. And the frontend as well for what they do not provide. We both follow the same spec.

https://microsoft.github.io/debug-adapter-protocol/
And also these people https://microsoft.github.io/debug-adapter-protocol/implementors/adapters/

Define inconsistent. There are many things that were done by us first. Two years later, they implemented some of the same things. Do you want us to break compatibility? And many things that don't make sense for them. Like `breakAfterReset`. SWO/RTT/semihosting/Peripherals. Some things don't make sense for us.

I notice you did not volunteer to do a study and write up a comparison :-)

hongshui3000, 2022-01-19:
My knowledge of debuggers is somewhat lacking, so I'm learning some basics . :-)