Marus/cortex-debug #577: C symbol names are not displayed in the Disassembly view

hongshui3000, 2022-01-20:
This is a question about function name display.
I don't know if it can be done in this plugin.

haneefdm, 2022-01-20:
I have it and did it and it does not look good. I will try again and provide it as an option.

What you see is actually a table and there isn't a column for the symbol yet. I did provide MS with the symbol info but they simply ignore it. I tried stuffing it into an existing column but it looks bad and you can't see the instruction anymore. There are only three columns I think.

I think the instruction bytes need (26 * 3  + padding) because x86 can be that long.  I faked it in various ways and basically, I lost the actual instruction completely -- pushed out to the screen and I noticed there is no horizontal scroll bar.

So, I will wait to see if MS makes any improvements Another worry I have is symbol names can get very long in C++. Maybe they can display symbol names on hover. Maybe you can request it over at MS?

I will think about it after this release. Don't want to stay in pre-release mode forever.

But 99% of the time, I felt like I didn't need it because I was looking at actual source side by side. and I also recognize my source code and what function I am in.

haneefdm, 2022-01-20:
Here is a preview @hongshui3000 I stole the column meant for opcodes and used it for func-name+offset (like gdb presents it) and combined the opcodes (no white space) and the instruction

[image]

This only works because we only do ARM/RISC type processors. It will not work for something like x86. Notice the first column is big -- that is because of 64-bit addresses which we eventually will do I guess. Best I could do for now.

haneefdm, 2022-01-20:
Super long functions will be shorted on the left they will look like `<..superlongstuff+124>`

closed 2022-01-20 by haneefdm

hongshui3000, 2022-01-21:
I think this at least solves my issue. In this way, I can easily find the function entry and exit points when disassembling