Marus/cortex-debug #578: Does the latest version affect the disassembly view of MCUs of other architectures

hongshui3000, 2022-01-21:
Does the latest version affect the disassembly view of MCUs of other architectures?
log
```
160-stack-list-frames --thread 1 0 0
GDB -> App: {"token":160,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x200009f3"],["func","main"],["file","E:\\work\\my-sdk\\samples\\hello_world\\src\\main.c"],["fullname","E:\\work\\my-sdk\\samples\\hello_world\\src\\main.c"],["line","12"],["arch","xtensa"]]]]]]}}
Debug-70: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x200009f3","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":70}
Debug-70: Dequeuing...
Debug-71: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x200009f3","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":71}
Debug-71: Dequeuing...

```
[image]

haneefdm, 2022-01-21:
Yes it can have issues. Depends on the architecture, min/max. instruction sizes, instruction alignment. Even for ARM, we do multiple checks to make sure we are doing something sane and not showing you garbage. This is because we start disassembly from an almost random spot - due to how requests come from VSCode. When you start from such an address, gdb will still disassemble but can be total garbage. This why we have multiple checks.

It absolutely will have trouble disassembling x86 for instance. That thing is a beast.

What is your processor btw? I see Xtensa. Is that a Tensillica? I am not familiar with it. Are you using one of those HiFi cores?

haneefdm, 2022-01-21:
What does the gdb command `show architecture` tell you?

haneefdm, 2022-01-21:
Can you also run the command `set architecture` with no arguments? I know it will produce an error but it will tell you what architectures your gdb supports

hongshui3000, 2022-01-22:
show 
```
show architecture
65-thread-select 1
66-stack-select-frame 0
67-interpreter-exec console "show architecture"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"Python Exception <type 'exceptions.NameError'> Installation error: gdb._execute_unwinders function is missing: \n"}]}
Python Exception <type 'exceptions.NameError'> Installation error: gdb._execute_unwinders function is missing: 
GDB -> App: {"token":65,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["new-thread-id","1"],["frame",[["level","0"],["addr","0x200009f3"],["func","main"],["args",[]],["file","E:\\work\\TP\\MY-IOT\\MY-sdk\\samples\\hello_world\\src\\main.c"],["fullname","E:\\work\\TP\\MY-IOT\\MY-sdk\\samples\\hello_world\\src\\main.c"],["line","12"],["arch","xtensa"]]]]}}
GDB -> App: {"token":66,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"The target architecture is set to \"auto\" (currently \"xtensa\").\n"}]}
The target architecture is set to "auto" (currently "xtensa").
GDB -> App: {"token":67,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
{"token":67,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
68-stack-list-variables --thread 1 --frame 0 --simple-values
GDB -> App: {"token":68,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["variables",[]]]}}

```

set architecture

```
set architecture
69-thread-select 1
70-stack-select-frame 0
71-interpreter-exec console "set architecture"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"Python Exception <type 'exceptions.NameError'> Installation error: gdb._execute_unwinders function is missing: \n"}]}
Python Exception <type 'exceptions.NameError'> Installation error: gdb._execute_unwinders function is missing: 
GDB -> App: {"token":69,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["new-thread-id","1"],["frame",[["level","0"],["addr","0x200009f3"],["func","main"],["args",[]],["file","E:\\work\\TP\\MY-IOT\\MY-sdk\\samples\\hello_world\\src\\main.c"],["fullname","E:\\work\\TP\\MY-IOT\\MY-sdk\\samples\\hello_world\\src\\main.c"],["line","12"],["arch","xtensa"]]]]}}
GDB -> App: {"token":70,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"Requires an argument. Valid arguments are xtensa, auto.\n"}]}
Requires an argument. Valid arguments are xtensa, auto.
GDB -> App: {"token":71,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Requires an argument. Valid arguments are xtensa, auto."]]}}
Requires an argument. Valid arguments are xtensa, auto. (from interpreter-exec console "set architecture")
72-stack-list-variables --thread 1 --frame 0 --simple-values
GDB -> App: {"token":72,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["variables",[]]]}}

```

hongshui3000, 2022-01-22:
Why not use “where” gdb command to query the current function and then assemble the current function

haneefdm, 2022-01-22:
The disassembly request occurs far beyond the current function in up and down directions and continues to happen when scroll.

hongshui3000, 2022-01-23:
My idea is whether we can increase the address in terms of the size of a function，so we don't care about the architecture of MCU

 this.startAddress =  function start addr
 this.endAddress =  function end addr

haneefdm, 2022-01-23:
Wish it was that simple. I get a request from VSCode for arbitrary start/end points (not physical addresses). If you start there, really think about how that translates to gdb-queries and how it can go wrong. I am not going to mess up ARM debugging for you - especially with ZERO ability to verify what I am doing.

Please, we need to stay on point. How can I tell the characteristics of your architecture? Looks like all it says is `xtensa`. I know nothing about it. I am going to close it if you can't point me to some documentation about what this is. This is a better way to help me help you. I am not looking for 1000+ page document, but rather important aspects.

hongshui3000, 2022-01-24:
> Wish it was that simple. I get a request from VSCode for arbitrary start/end points (not physical addresses). If you start there, really think about how that translates to gdb-queries and how it can [...]

doc:
[docs.zip](https://github.com/Marus/cortex-debug/files/7922240/docs.zip)

haneefdm, 2022-01-27:
Try out the new preview release (v1.3.0). II can't comment if will work or not because I have nothing to test with.

hongshui3000, 2022-01-28:
Still can't successfully disassemble

fail:
[image]
log:
[log.txt](https://github.com/Marus/cortex-debug/files/7955253/log.txt)

haneefdm, 2022-01-28:
How can your processor have 1-byte instructions? I thought the minimum was 16 bits

But, forgot about alignment. I don't expect things to be 1-byte aligned., But in Tensilica, it is 1-byte because you have 24-bit instructions. But, in the manual, I only saw 2 and 3-byte instructions (core). Does it use them for noops to pad odd-sized instructions?

In the end, you may have to enter 3-4 things about your instruction set in `launch.json`. Lot of work btw. I am not removing my error checks knowing how error-prone disassembly can be.

hongshui3000, 2022-01-28:
> But, forgot about alignment. I don't expect things to be 1-byte aligned., But in Tensilica, it is 1-byte because you have 24-bit instructions. But, in the manual, I only saw 2 and 3-byte instructions (core). Does it use them for noops to pad odd-sized instructions?

The instruction lengths I see so far are 16bit, 24bit and 48bit or 32bit? (It supports FLIX instructions, which can encode multiple independent instruction operations into a FLIX instruction word), I remember that my processor can combine two instructions, and I saw 48bit instructions in my code
This may be a complex need for alignment.

haneefdm, 2022-02-03:
I don't see anything specific or pending here...

closed 2022-02-03 by haneefdm

hongshui3000, 2022-02-07:
v1.3.3
The disassembly can be displayed normally, thanks