Marus/cortex-debug #579: Debugging suddenly broken - STM32 + STLink ISOL v2 + OpenOCD

klasyc, 2022-01-24:
First thank you for this great extension. I have been using it approximately half a year without any significant issues. Unfortunately last week I installed some updates (on both Windows and VS Code) and the extension stopped working anymore. 

Normally when I launch the project (OpenOCD with `.elf` file), `Reset_Handler` assembly opens in a new tab. Then the firmware starts. Now the `Reset_Handler` tab opens, but it's empty. VS code shows loading indicator in the tab, but its contents does not appear and firmware does not start - it seems the loading of `Reset_Handler` assembly hangs and block the firmware execution. I tried to load firmware to the board using bare OpenOCD and it works without any issues, therefore I am creating a new issue here.

My VS Code:

**Version: 1.63.2 (system setup)
Commit: 899d46d82c4c95423fb7e10e68eba52050e30ba3
Date: 2021-12-15T09:40:02.816Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19043**

Coretex-Debug version: 1.2, also tried several older versions, but the behavior is same.
OpenOCD version: 0.11

Screenshot of hanging firmware debug:

[image]

Terminal output:
```
Waiting for gdb server to start...[2022-01-24T10:14:01.108Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Users\\Vojtech\\Projekty\\CSS" -f "c:/Users/Vojtech/.vscode/extensions/marus25.cortex-debug-1.2.0/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32f3x.cfg
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-16:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J25S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.404712
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Info : device id = 0x10036446
Info : flash size = 512kbytes
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002298 msp: 0x2000fffc
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002298 msp: 0x2000fffc
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Flash write discontinued at 0x08002dac, next section at 0x08004000
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002298 msp: 0x2000fffc
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002298 msp: 0x2000fffc
```

Debug console:
```
Launching GDB: arm-none-eabi-gdb.exe -q --interpreter=mi2 "C:\\Users\\Vojtech\\Projekty\\CSS\\build\\debug\\csp.elf"
Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
undefinedc:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Reading symbols from C:\Users\Vojtech\Projekty\CSS\build\debug\csp.elf...
Launching gdb-server: openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Users\\Vojtech\\Projekty\\CSS" -f "c:/Users/Vojtech/.vscode/extensions/marus25.cortex-debug-1.2.0/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32f3x.cfg
Please check TERMINAL tab (gdb-server) for output from openocd.exe
0x08002298 in _binary_build_debug_bootloader_boot_csp_bin_start ()
Program stopped, probably due to a reset and/or halt issued by debugger
Unable to match requested speed 1000 kHz, using 950 kHz
Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002298 msp: 0x2000fffc
Unable to match requested speed 1000 kHz, using 950 kHz
Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002298 msp: 0x2000fffc
Note: automatically using hardware breakpoints for read-only addresses.
Debug-18: Enqueuing {"command":"disassemble","arguments":{"function":"Reset_Handler","file":null},"type":"request","seq":18}
Debug-18: Dequeuing...
Note: We detected the following memory regions as valid using gdb "info mem" and "objdump -h"
    This information is used to adjust bounds only when normal disassembly fails.
================================================================================
  Using following memory regions for disassembly
================================================================================
      Size   VMA Beg   VMA End   LMA Beg   LMA End  Attributes
================================================================================
  08000000  00000000  08000000  --------  --------  rw nocache
  00080000  08000000  08080000  --------  --------  flash blocksize 0x800 nocache
  f7f80000  08080000  00000000  --------  --------  rw nocache
  ------------------------------------------------------------------------------
  00002dac  08000000  08002dac  --------  --------  (.obl) contents alloc load data
  00000208  08004000  08004208  --------  --------  (.isr_vector) contents alloc load readonly data
  00008e90  08004208  0800d098  --------  --------  (.text) contents alloc load readonly code
  00000850  0800d098  0800d8e8  --------  --------  (.rodata) contents alloc load readonly data
  00000008  0800d8e8  0800d8f0  --------  --------  (.ARM) contents alloc load readonly data
  0000000c  0800d8f0  0800d8fc  --------  --------  (.init_array) contents alloc load data
  00000004  0800d8fc  0800d900  --------  --------  (.fini_array) contents alloc load data
  00000070  20000000  20000070  0800d900  0800d970  (.data) contents alloc load data
  00000304  20000070  20000374  --------  --------  (.bss) alloc
  00000600  20000374  20000974  --------  --------  (._user_heap_stack) alloc
================================================================================
```

Launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ST-Link",
            "cwd": "${workspaceRoot}",
            "executable": "build/debug/csp.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
           	"device" : "STM32F303RE",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f3x.cfg",
            ]
        }
    ]
}
```

Thank you for help!
Klasyc

haneefdm, 2022-01-24:
Thank you for reporting this problem. It appears to be our problem. Something also seems to have changed in VSCode as well but, in the end, it may be our bug. You may have to rollback to the previous release temporarily.

It will take me a day or so to address. Would you be willing to try out a pre-release?

klasyc, 2022-01-24:
Sure, I would be glad to help to test the pre-release version. Rolling a version back did not help, I had to switch to my old IDE.

haneefdm, 2022-01-24:
Our previous final release was 0.4.10.

Our versioning system changed to support pre-releases by MS. Odd minor versions are pre-releases. If a rollback did not help, then I am thinking that we have a different problem.

klasyc, 2022-01-24:
I tried versions 1.1.10 and 1.1.9 which didn't work. Downgrading to the version 0.4.10 helped.

haneefdm, 2022-01-24:
Yes all 1.1.x were pre-releases. I just prepared a hotfix release for you to try out.

https://github.com/Marus/cortex-debug/releases/tag/v1.2.1

Download the VSIX file. Here are some instruction to install the VSIX file manually.

https://stackoverflow.com/questions/42017617/how-to-install-vs-code-extension-manually

haneefdm, 2022-01-24:
Thanks for checking it out. We have two disasemblers and in this case, the old one should have been used and it didn't. We are on our way to deperate the old one but need help from MS (VSCode) to remove it completely. Sorry for the bug.

I pushed the hotfix to the MS Marketplace and you should be getting the official one from there in a few mins. They are tiny bit different from what you downloaded and should be automatically replaced by the official version. Just in case, you can revert to 1.2.0 and then install 1.2.1 from the Marketplace.

It may take an hour or two before everyone can see 1.2.1 -- how/when that happens is a mystery to me.

haneefdm, 2022-01-24:
Also, thank you for submiting **all** the info I needed to see what the problem was. It is very rare that issue reports contain relevant info.

In the next release, I am planning to integrate the new disassembler for displaying assembly code when the stack frame top has no source file/line info associated with it. Just like before but with the newer one. Do you mind trying it out when ready. I know your time is valuable but consider it a gift to the community.

In case you didn't know about the new disassembler... [Here is some info](https://github.com/Marus/cortex-debug/wiki/Disassembly-Debugging)

If you don't reply, I understand.

closed 2022-01-24 by haneefdm

klasyc, 2022-01-24:
I am OK with further testing, feel free to contact me with any updates :)