Marus/cortex-debug #582: Unable to save register preferences

zrno, 2022-01-24:
Issue Type: <b>Bug</b>

1. Start debugging
2. Debugger enters and stops at start of main()
3. Run the code. Press "Reset device", "Restart" in the debug floating pannel a few times. Continue debugging. Pause execution
4. Change the "Cortex register" panel to "Hex mode"

Not sure if caused by those steps but now every time I just start debugging and stop immediatelly I get the same error pop ups.

Extension version: 1.2.1
VS Code version: Code 1.63.2 (899d46d82c4c95423fb7e10e68eba52050e30ba3, 2021-12-15T09:39:46.686Z)
OS version: Linux x64 5.16.2-arch1-1
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz (8 x 3000)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>rasterization: disabled_software<br>skia_renderer: enabled_on<br>video_decode: disabled_software<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|1, 1, 1|
|Memory (System)|31.07GB (14.67GB free)|
|Process Argv|--no-sandbox --unity-launch --crash-reporter-id 87fed376-385e-4b70-a6e7-f62fe0f386e3|
|Screen Reader|no|
|VM|0%|
|DESKTOP_SESSION|gnome|
|XDG_CURRENT_DESKTOP|GNOME|
|XDG_SESSION_DESKTOP|gnome|
|XDG_SESSION_TYPE|wayland|
</details>

haneefdm, 2022-01-25:
Thank you for reporting this. Will look into it.

The registers window is going away soon but nevertheless, I can see the annoying popup being an issue.

haneefdm, 2022-01-28:
Should be fixed in the next release/pre-release. The same could have happened for peripherals. But remember that the Registers window will be deprecated soon.

haneefdm, 2022-02-03:
The fix should be available as a pre-release 1.3.2. Thank you for reporting this. I hope you can verify the fix.

You can opt-in and install the pre-release right from VSCode extensions tab.

closed 2022-03-28 by haneefdm