Marus/cortex-debug #583: Assembly has the same disassembly view as C/C++

hongshui3000, 2022-01-25:
https://github.com/microsoft/vscode/issues/141127
Can this be added to the plugin?

haneefdm, 2022-01-25:
What do you mean? We already do that. Otherwise, you would have soon zero disassembly. Has nothing to do with language. It can even be rust. We don't care and neither does VSCode.

You should start with a problem statement. I don't understand what the problem is?

Make sure your files (regardless of language) are compiled/linked properly and have preserved symbols. We are disassembling thing even with no symbols.  Also path names are important (like you compiled with a different directory structure).

hongshui3000, 2022-01-26:
I'll reopen this when I can fully understand the issue, I'll close it first

closed 2022-01-26 by hongshui3000

hongshui3000, 2022-01-26:
VS Code does not care about file types for the "Open Disassembly View" command.
It only looks at the currently selected stackframe that is shown in the editor. If the frame has a instructionPointerReference property, then the "Open Disassembly View" command is shown in the editor's context menu. Since the stackframe is provided by the debug extension, it is ultimately the debug extension that determines whether "Open Disassembly View" is available or not.

haneefdm, 2022-01-26:
> Since the stackframe is provided by the debug extension, it is ultimately the debug extension that determines whether "Open Disassembly View" is available or not.

Wrong. Bit more than that.

No program paused/halted, no disassembly since no stack frame. But you were saying it has something to do with the language. We provide the instruction reference ALL the time and has nothing to do with language -- it can be Fortran or Martian or no source at all for all I care.  Note that we cannot do that if the program is NOT paused or the call stack window has no stack frames -- like when the program is running or no debug session is active.

According to the DAP spec. it all comes down to how the disassembly requests are made by VSCode, around the instruction pointer an arbitrary 200 hundred instructions before and 200 after -- and this is where there is no good way to do this and why we have to verify if what we are returning to VSCode is even valid. GDB can give pure garbage, if you don't ask it right and many times even when you ask it right. Between the compiler, inlining, linker, and gdb a whole bunch of stuff gets jumbled up. Lot of work to figure this out with near-zero documentation. And, we have to do this fast. No more than a few milliseconds or user experience will be bad.

It is far easier to right-click on the correct frame in the Call Stack as well.