Marus/cortex-debug #584: New Device Support Pack for STM32WL fails to load `.svd`

zfields, 2022-01-25:
I am attempting to create a Device Support Pack for the STM32WL family.

I have started by cloning https://github.com/marus/cortex-debug-dp-stm32l4, and making the most minimal and surgical changes to target the WL family instead of the L4 family.

My change set from STM32L4 DSP is captured here: https://github.com/zfields/cortex-debug-dp-stm32wl/commit/8fb31a31e26976d0a41cf2cea0314500d3390e2f

- I have copied the WL `.svd` files from the latest STMCubeIDE (v1.8.0), and swapped them with the existing L4 `.svd` files.
- I updated `extension.ts` to scan for the appropriate files _(TBH, I am a novice at javascript/typescript, so I may have made a mistake. I am basically pattern matching the best I can.)_
- I updated the package dependencies, because it was failing to compile with the older versions.

I have compiled the `.vsix` and installed it along side Cortex-Debug, but it doesn't appear to be working correctly. The DSP plugin seems straight-forward, but I cannot understand why it is failing with:

`DSP: No active debug sessions or no SVD files specified.`

zfields, 2022-01-25:
FWIW, I also tried incorporating [this PR to the STM32L4 DSP](https://github.com/Marus/cortex-debug-dp-stm32l4/pull/1/files), in a separate branch, to enumerate and auto-load the `.svd` files in the data directory. I did this to hedge against the idea that I had written poor javascript in `extension.ts`, but it did _NOT_ appear to make any difference.

haneefdm, 2022-01-25:
Could you attach your launch.json? I don't have write permissions to those other repo's but I can see what I can do on this one.

haneefdm, 2022-01-25:
> DSP: No active debug sessions or no SVD files specified.

Such a message does not exist in our extension. Our message looks like this

```
SVD: No active debug sessions or no SVD files specified.
```

Are you sure you are using our extension and not a clone? Some people have cloned/forked ours and releasing under very similar names...

zfields, 2022-01-25:
Sorry, I mistyped the error message, it is indeed...
```none
SVD: No active debug sessions or no SVD files specified.
```
Yes, I am using your extension (`marus25.cortex-debug`). I was deliberate about this choice, because I noticed the `marus25` account was also responsible for several DSP's.

`launch.json`

```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "type": "cortex-debug",
            "executable": "${workspaceRoot}/build/sparrow.elf",
            "cwd": "${workspaceRoot}",
            "request": "launch",
            "servertype": "stlink",
        }
    ]
}
```

In case it is important, the `.elf` is compiled using the defines `STM32WL55xx` and `CORE_CM4`.

Lastly, I double checked all the other repo links I shared, to be sure they are public. I offered them, so you can "see" what I am trying to do, in case I am doing something obviously wrong or otherwise silly.

Thank you for the quick response, I can't tell you how appreciative I am!

haneefdm, 2022-01-25:
You are missing the device name in launch.json. This is what is matched against all the regular expressions. This is how a device is mapped to a particular svd file.

We don't/can't look inside your executable or source for the device name.

zfields, 2022-01-25:
Okay, in hindsight, that makes perfect sense... :man_facepalming: 

I'm sorry if I missed this in the `README.md` or some other documentation, but what is the appropriate format of the device name? Should is match the `.svd` file (i.e. `STM32WL5x_CM4`), or is it the name from the compile time definition (i.e. `STM32WL55xx`), or can it be anything that matches the pattern associated with the `.svd` in `extension.ts`?

haneefdm, 2022-01-25:
People typically follow what JLink/OpenOCD uses as a device name. STLink is kinda new and probably does not care. It also depends on how many devices are supported by your SVD file and your regex should be appropriate for those devices. I don't have a decoder ring for this family of devices, unfortunately. Note that the `"device"` is used for JLink directly as an argument to it

https://www.segger.com/supported-devices/jlink/st/stm32wl

Both JLink/OpenOCD do not require fully qualified names and it probably won't work like that. It feels like your compile-time constant seems appropriate. But not sure of the `_CM4` suffix. Maybe this?
```
/^STM32WL5[0-9].*/i
```
Or that may be too specific or too general. I don't know.

It is best that the extension document what it is assuming give guidance to the users.

zfields, 2022-01-25:
Thank you, that is the guidance I was looking for!

One last point of clarification, when you said, "device name", I assumed you we talking about the `"name"` key in the JSON. Is `"name"` the correct field for pattern matching and is also passed to JLink - or is it a JLink only parameter?

haneefdm, 2022-01-25:
No, I mean the `"device"` property. `"name"` is for VSCode to manage configurations and what comes up in the debug launch menu
[image]
https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

Yes, you can have as many svds as you have processors but for most vendors, it is kind of a waste (not always). The peripherals live in the global space and are shared between processors. But, yes you can have more than one and the right one gets updated depending on which core halted. We can't tell if the multiple-SVDs are all on the same device. Perhaps in the future...

We have examples of multi-core and multi-session (separate devices) and we don't distinguish very much. See

https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging

It will take a bit of understanding your own device but yes, it is being successfully used to debug multiple cores at the same time.

zfields, 2022-01-25:
Amazing! Thank you for everything!

closed 2022-01-25 by zfields