Marus/cortex-debug #585: Console not replaced when launching debug

SchutzSeb, 2022-01-26:
Hi,

Each time I launch a new debug session the previous debug consoles are kept open, instead of being remplaced by the new one.

[image]

Is it possible to configure the behavior ? Or if not, would it be possible to make it configurable ?

Best regards,
Sebastien

haneefdm, 2022-01-26:
First your title. Does not seem right. RTT windows and Debug Console have nothing to do with each other and have very distinct meanings.

We try to re-use the RTT windows as much as possible. The only time we don't is if the TCP connection is still active -- this should die when the debug session ends but in some cases, it won't. I use RTT all the time and have never seen this.

Then there is a possibility of a Bug. More than likely, it is a bug in our extension. Can tell me steps to reproduce and also. Attaching your launch.json is also helpful.

haneefdm, 2022-01-26:
Oh, we don't remove the RTT windows if you created them with new names perhaps, but it doesn't look like that is what you did.

mischelebuha, 2022-01-27:
I have the same problem, this appears since the last update (v1.2.1) last week.

SchutzSeb, 2022-01-27:
Hi haneefdm,

First thanks for your quick reply. I apologize for the title, I´m quite new to these tools, and probably mix some concepts ^^ (I have tried a renaming, tell me if it is still nok).

Here is the content of the launch.json file:

`{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Launch",
            "servertype": "jlink",
            "device": "ATSAMS70Q21",
            "svdFile": "${workspaceFolder}/tools/svd/ATSAMS70Q21.svd",
            "interface": "swd",
            "serverArgs": [
                "-speed",
                "50000"
            ],
            "executable": "${command:cmake.launchTargetPath}",
            "rtos": "FreeRTOS",
            "runToEntryPoint": "main",
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "port": 0,
                        "type": "console"
                    }
                ]
            }
        }
    ]
}
`
For the steps, I simply launch the debug, then end it and relaunch a new debug session.

I didn´t yet customized the names of the console, so it should be the default one.

Best regards.

haneefdm, 2022-01-27:
Hmmm. Puzzling. VSCode did change though. They seem to have changed the terminal API or how terminals are managed. I also see an issue (reported to MS) where the list of terminals on the right side go blank every once in a while. They re-appear if you reside the left/right parts.

It may also be something we did something to handle multiple sessions/cores and RTT/SWO cores can create multiple windows. Thanks for reporting this.

I am glad someone is using RTT btw. Very nice FW solution. That was a lot of work to add it to our extension.

haneefdm, 2022-01-27:
Fixed. Just published 1.2.2. You should have it in a few mins/hours -- depending on how MS pushes extensions worldwide.

SchutzSeb, 2022-01-28:
Hi,

Thank you so much for the fast fix ! And for the RTT integration, that is really useful !

I have loaded the new version of the extension but still have the issue.

[image]
[image]

Is there anything else I should do ? Or do you need any other information ?

haneefdm, 2022-01-28:
Hmmm. One reason this happens is if the RTT connection for the first run has not really ended.

After you end the first session, can you see if JLinkGDBServer has actually exited? Use Process Explorer/Task Manager on Windows or `ps` on non-Wndows to see if it really exited.

After a session ends, do you see in the `RTT CH:0 console` window, a message in magenta that the session has ended and that it is waitning for a new one? If not, JLink has not ended the connection.

Unfortunately, I was using OpenOCD as my JLink probe was busy.

SchutzSeb, 2022-01-28:
I confirm JlinkGDBServer is exited when ending the first session.

I have the magenta message indicating that gdb-server ic closed into the gdb-server window but not in `RTT Ch:0 console` window.

haneefdm, 2022-01-28:
If you didn't see a message in the RTT window, then the connection was still active. There are multiple connections going on here (half a dozen).

JLink is notorious for hanging around -- sometimes under a slightly different process name, especially on Windows. Can you see if there *any* process from SEGGER/JLink still running?

I can try closing the connection on our end but that is silly. There could still be stuff in flight.

haneefdm, 2022-01-28:
Turns out we *are* closing the connection on our end too. And you don't see a message like this..
```
RTT connection on TCP port 60003 ended. Waiting for next connection...
```
I am stumped. It is al you are still using the older version. I have to see if I can make a new build ... we have moved on to 1.3.x with more changes and enhancements

haneefdm, 2022-01-28:
Just switched to JLink and I cannot duplicate it with the newest version... It is using recycling the windows and also I do get a message indicating that Jink has closed the TCP port.
```
RTT connection on TCP port 19021 ended. Waiting for next connection...
```
In a day or so, perhaps you can try out our new version 1.3.x, but it will be a pre-release version with many other changes included.

haneefdm, 2022-01-29:
I don't know if you want to try pre-release 1.3.1 but it is now available. To use that you can use VSCode extensions area and switch to pre-release.

PhilippHaefele, 2022-01-31:
Hi together,
even with 1.3.1 i can still see multiple RTT terminal tabs when stopping and starting debugging again (get one new tab per start).
I also do not see the Magenta message in the Terminal tabs.

It seems at least that the connection is closed as in all sessions the sam RTT Telnet Port is used.

I'm using:
Windows 10 Enterprise
VSCode 1.63.2
Cortex-Debug 1.3.1
J-Link V7.60a

I did debug a little bit:

It seems that the inUse variable is not set to false after closing the sessing which will prevent the terminal from reusing in followup session (`rttCreateTerninal` in extension.ts)
[image]

This is not set because the `onClose` callback (in `rtt_terminals.ts`) isn't invoked at all. Also `dispose` is not called at all.

Will try to find out why none of them are called.

@haneefdm Any guess on that or where/how/what i can look for?

PhilippHaefele, 2022-01-31:
Hi,

me again. I maybe found something.

It seems that the terminal conenction is closed via a reset from J-Link side
[image]

In this case it seems that the `end` callback is not called in all cases.
When i set this to `close` it seem to be working propberly.
@haneefdm As i'm not into how all that works (either Node.js Net lib and also the TCP conenctions in Cortex-Debug), it's maybe better that you take a look if my solution is ok or not.

@haneefdm If you have a fix/change i can test, let me know

PhilippHaefele, 2022-01-31:
Another small addition:

It seems that we get into `error` callback in this case where i found following thing:
```js
if ((code === 'ECONNRESET') && this.connected) {
    // Server closed the connection. Done with this session. Should we emit('disconnected')?
```

So sequence is `error` callback with `ECONNRESET` and after that the `close` callback and no `end`

haneefdm, 2022-01-31:
Thank you for figuring that out. yes, you are correct. I was never seeing that situation hence the comment. Makes sense. I don't know why I put a comment there but not actual code :-)

We did not emit certain events ourselves because prior to actually connecting, you get a few and the clients (the terminals) get confused -- due to retries. The terminals themselves don't have a concept of a retry. There is a bit of chicken and egg in there...

I will test it with this but I am having trouble reproducing it. I get an very nice 'end' event every time.

```JavaScript
                    if ((code === 'ECONNRESET') && this.connected) {
                        // Server closed the connection. Done with this session
                        this.emit('disconnected');
                        this.dispose();
                    } else if (code === 'ECONNREFUSED') {
```

haneefdm, 2022-01-31:
And, yes I was expecting an 'end' callback after but looks like we are getting a 'close' as well.

Now, I have to guard against emitting 'disconnect' multiple times ... that is not good for clients and there are multiple -- not just RTT

haneefdm, 2022-01-31:
Had to rejig it quite a bit. The retries we are doing -- which was not needed until recently -- causes the 'close' even to be called because we are the ones destroying a connection that never got connected in the first place.

I pushed just the change to the socket.ts file.

https://github.com/Marus/cortex-debug/blob/master/src/frontend/swo/sources/socket.ts

Not sure if you will have the time to use it but ...

haneefdm, 2022-02-03:
Could not have fixed it without help. I just couldn't reproduce it. I had to fake it eventually

It is out in pre-release 1.3.2 (fingers crossed)

closed 2022-02-03 by haneefdm

SchutzSeb, 2022-02-03:
Hi,

We have tested the pre-release 1.3.2 for the whole morning and confirm that the fix is working for us too.

Thank you so much for your support !