Marus/cortex-debug #586: Environments and Symbol lookup mangled in newer versions

Plasmarobo, 2022-01-26:
I've noticed that newer versions of Cortex-Debug seem to be mangling symbol lookup for ST-Link and tools.
```
Operating System: Debian
Shell: zsh
IDE: Visual Studio Code
Versions: 1.2.1 (broken), 1.1.10 (broken), 0.4.10 (working)
```

When invoking the ST-Link tools via the below launch config, ST-Link complains about missing symbols (dynamic linking errors?), specifically the `GetLibApiVerv` symbol.
```
  {
        "name": "My App (cortex-debug/stlink)",
        "executable": "${workspaceRoot}/app/build/app_debug/app_debug.elf",
        "request": "launch",
        "type": "cortex-debug",
        "servertype": "stlink",
        "armToolchainPath": "/opt/miniconda3/envs/build/bin/",
        "stm32cubeprogrammer": "/opt/stm/stm32Cube/STM32CubeProgrammer/bin",
        "serverpath": "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.100.202109301221/tools/bin/ST-LINK_gdbserver",
      },
```

Note: this works just fine with 0.4.10. I'm assuming there was some environment changed which overrides or breaks dynamic linking in this case.

haneefdm, 2022-01-26:
Hmmm. Thank you for reporting this problem.

What changed is that there is no longer an option to de-mangle. It should be done all the time as it does not hurt anything.

Are you using C++ or Rust? I am assuming that you are using the same version of GNU tools (objdump, gdb, etc.) with all the versions?

Plasmarobo, 2022-01-26:
Using C in this case.
The toolchain is the same `arm-none-eabi` chain pinned to a specific version, so no other tooling or environment changes.
The same commands work in my normal shell, and the linker is able to resolve.

I am using a miniconda environment, but the ST tools work both inside and outside of my conda env.

haneefdm, 2022-01-26:
We do two things. Run objdump to get symbol names demangled
```
arm-none-eabi-objdump --syms -C -h -w <elf-file>
```
Output for the above can be pretty large. Please make sure that your symbols are indeed demangled. The `-C` is the option that does that.

And for gdb, we do the following.
```
set print demangle on
set print asm-demangle on
```
The above should also force GDB to demangle when printing names. You will see these commands if you enable debug -- in the Debug Console. In your launch.json, add the following
```
            "showDevDebugOutput": "raw",
```

haneefdm, 2022-01-27:
I am sorry, I misread the original issue. It says

> When invoking the ST-Link tools via the below launch config, ST-Link complains about missing symbols (dynamic linking errors?), specifically the GetLibApiVerv symbol.

We don't even communicate with ST-Link tools directly or even indirectly. We are not involved in compilation either. Only thing w do is launch the stlink gdb-server and let gdb talk to it.

It sounds like you have a bad st-link executable. See if you can run the st-link server from the command line (you will find exact command line in Debug Console).

Please re-open if this still an issue with details of what exactly is not working.

Plasmarobo, 2022-01-27:
Sorry, let me clarify: I can run everything successfully from a different shell.
The executable isn't at fault. The shell environment that is running through Cortex-Debug is causing my issue (not sure how or why). 

Version 0.4.10 works as expected.
Versions 1.2.1 and 1.1.10 both exhibit the following failure mode (gdbserver terminal)
[image]

Here is an example of running the same command in my native zsh
[image]

haneefdm, 2022-01-27:
Okay, that makes sense.

What was the current dir. when you ran from command-line? For us, the current dir. has to be near the workspace (or the one specified in launch.json) or things will not work.

With 1.1.x (preview releases for 1.2.x), more directories were added to the library path (DYLD_FALLBACK_LIBRARY_PATH for Mac and LD_LIBRARY_PATH for Linux). This was a suggestion/change from another user. What was in 0.4.10 was that we relied on your own env. being right for Mac and Linux. But what is done should be additive. You can see it here...

https://github.com/Marus/cortex-debug/blob/4e167cb8854e6356a2f7e22490e48c132a468906/src/stlink.ts#L195

I will see if I can reproduce this. Do you have the ST tools installed in multiple places? It is possible the code is combining paths from different installations that have a good chance of failure. But, if your env. was already setup properly, it should not have failed because of the additive changes.

I was not thrilled with this change in our code.  This should have been done properly by ST themselves and there are standard ways of handling dynamic libraries. No professional tool does this.

One way to debug this is to replace the executable with a shell script and dump the env. variables and see what you are getting. Would you be willing to do that?

Plasmarobo, 2022-01-27:
Yep! I'll mess around and see what data I can get back to you.

I do, in fact, have 3 different ST toolchains installed (long story, agree ST does things... _interestingly_).
I bet there is some strangeness coming from what Visual Studio Code considers the "environment" vs what zsh is pulling in.
Do you have suggestions for an env dumping script? Want to make sure I get a useful snapshot.

haneefdm, 2022-01-27:
It is all done using the DYLD_LIBRARY_PATH environment variable. Nothing else. But it is scanning some familiar places to find the ST stuff.

I really hated adding that to our code but I finally agreed because of so many asks.

Plasmarobo, 2022-01-31:
After some digging, setting LD_LIBRARY_PATH seems to be overriding settings in my `/etc/ld.so.cache`.

I'll see what happens if I overload my library path to contain my cache.

haneefdm, 2022-02-03:
Any progress? I am assuming you used `ldd` to find out what the dependencies are. Is it an issue with 64/32 bit binaries maybe?

Plasmarobo, 2022-02-03:
Turns out ST-LINK ships a bunch of incompatible garbage. Still trying to sort out out.

Also working on a pr that just adds a boolean option to disable modifying the system environment. I'll let you know when I work it all out!

haneefdm, 2022-02-03:
More config options? STLink is becoming a testing nightmare. I am tempted to rip the whole thing out. It is so much easier to handle this on the user side by using a shell script that can do everything needed.

Plasmarobo, 2022-02-03:
I support this. Just let me know what I've got to call.
In all seriousness I was considering adding a global flag that can be used to let the user do basically that.

haneefdm, 2022-02-04:
@Plasmarobo Do you mean what to call the script? Didn't understand your question

We have inadvertently created a backward compatibility issue as well, so I can't rip out everything I don't like. Too late. I can rip out the whole LD_LIBRARY thing and replace with something equivalent. At least on a Mac/Linux this should be simple. Windows NOPE, but we have fewer issues there wrt. this.

haneefdm, 2022-02-04:
I need to find out how the ST programs are linked before they ship first. I think I can figure this out.

haneefdm, 2022-02-04:
What seems to work is to run the gdb-server with the CWD set to the directory where the executable resides. Can anyone confirm that? Nothing in the env. has to be set this way

We would be totally violating our rule for what the current directory should be but this is how ST themselves seem to be using it in Eclipse. This means any logfiles/config-files you specify as extra arguments to the server have to be full-path names and not relative to where you work. Never seen anything like that.

In any given install of ST tools, do you all see multiple versions of gdb-servers, openocd's, programmers, etc.? Wonder what the significance of those is. They can't just be cruft?!?!

I usually manually pick out the latest (for my purposes) from what looks like the latest version number but why do they do that? Must be a reason. But every time I get an update, I have to remember to check the plugins directory.

Plasmarobo, 2022-02-04:
I think they are taking the "shotgun" approach, but I haven't had time to analyze the directory structure.
I do see the specific driver that is causing this issue is the `libSTLinkUSBDriver.so`

My system likes the one in `/lib/` and breaks when loading the one included with `STM32CubeProgrammer`.

Testing the `cwd` workaround:
Success: works great. I think VS Code variables for paths expand so they can be used appropriately. Wont help someone with custom logging etc setup, but works fine for remote debugging.

**NOTE**
I'm a big idiot.
I had a custom path for STMCubeProgrammer pointing to an external tool (installed separately) instead of the one shipped with the CubeIDE. Version mismatches are tough. Don't need the workaround when using the correct tools.

**FOR ANYONE ELSE HAVING THIS ISSUE**
Make sure the versions on your tools match.

Thanks so much for your help!

haneefdm, 2022-02-04:
So, I should leave the LD_LIBRARY_PATH change in? You said it worked fine with 0.4.X before that change was introduced.

Plasmarobo, 2022-02-04:
You can, it wasn't the root cause of the issue, it just exposed it.
Essentially the root cause was always "not using the libraries that shipped", but my system (debian) had `libSTLinkUSBDriver.so` installed (a compatible version), so would find it from cache when run from my shell.

However the correct thing to do was to load the `libSTLinkUSBDriver.so` that shipped with my CubeIDE install.
I had set the STM32Programmer path, so it was loading that library instead.

the `LD_LIBRARY_PATH` was set correctly, but not actually resolving, the `stm32cubeprogrammer` option was taking precedence, but I didn't notice it. Apparently adding to `LD_LIBRARY_PATH` was causing it to try a load from that path first.

haneefdm, 2022-02-08:
I pushed out a new pre-release (1.3.4) where we no longer mess with the env. vars. But, we `cd` to the exe location of the stlink gdb server and run from there.

You can override using `serverCwd` setting (not yet documented but you can use it). This is something that will be needed when we add support for WSL and remote VSCode. Again, messing with the env. vars would not have worked with WSL/Remoting anyways.