Marus/cortex-debug #587: Watch window display characters

MaJerle, 2022-01-26:
It would be nice to have a char/string displayed when having variable (or char array) in watch window, next to hex value.

Was this considered already?

haneefdm, 2022-01-26:
It already does work like that. And gdb is the one that handles this for us.

[image]

It may be that your variable declaration does not agree with gdb? What does it look like? Note that in C, there is no such thing as a string (not a native data type). They are done by convention.

https://stackoverflow.com/questions/14709323/does-c-have-a-string-type/14709667#:~:text=There%20is%20no%20string%20type,one%20additional%20zero%20terminating%20character.&text=Both%20Java%20and%20Python%20have,concept%20of%20a%20%22string%22.

MaJerle, 2022-01-26:
Its not a native string, its a 'unsigned char' aka 'uint8_t'. But it only shows bytes when expanded. Temporary solution is to start "memory view", then I can see info next to hex.

haneefdm, 2022-01-26:
Yup, that is gdb for ya. Did you try casting to (char *) in the watch window? The problem is 'unsigned char' is not the same as 'char'.  uint8_t is not part of the C language either right. It is generally a #define or typedef that translates to 'unsigned char'

Values 127-255 and above are kind of undefined. And gdb does not do translations to anything but 7-bit ascii. We have to convince gdb that it is okay by using a typecast.

And, we do not try to interpret or re-interpret what gdb says. We pass it straight to VSCode.

haneefdm, 2022-01-26:
Here is my example...
[image]

MaJerle, 2022-01-26:
Tried char, doesnt work either.

Uint8_t is more than well parsed by gdb, just not well displayed here. Maybe I am missing a setting somewhere?

haneefdm, 2022-01-26:
Maybe you missed it in the picture above. In my program hello is defined as `uint8_t hello[]`. The same variable, I cast it to `(char *)`. You can see this same variable with and without casting in the watch window.

closed 2022-01-26 by haneefdm

MaJerle, 2022-01-27:
I do confirm that adding `(char *)` improves the situation, thanks.
Eclipse, on the other hand, always shows *user-readable* char value, same is with VS.

haneefdm, 2022-01-27:
One day we will get there. I don't have an army of people like Visual Studio or Eclipse. Just me. Yet, some things, we do better.