Marus/cortex-debug #590: Disassembly shows some issue with identifying constants in code

hongshui3000, 2022-01-29:
Constants are displayed as instructions

[image]

haneefdm, 2022-01-29:
Yup. I know that. That is what gdb gives you. I didn't make that up. Why do you think that is? You tell me.

haneefdm, 2022-01-29:
Some instructions can have immediate operands. Which processor is this?

haneefdm, 2022-01-29:
Sometimes compilers put constants right next to the code so that they can be referenced relative to the PC. This is very common with ARM especially with optimizations that asks each function to be its own elf section. That is also done with short branches.

Unless you can prove that I am doing something wrong...

closed 2022-01-29 by haneefdm

hongshui3000, 2022-01-29:
I don't mean to say what you are doing wrong, I just want to know if this is caused by the disassembly given by gdb or by vscode or cortex debug extension.
Because I see that ARM's own debugger is able to recognize these as data

haneefdm, 2022-01-29:
It is 100%  gdb. You can give gdb a random address and it will produce random, but good looking assembly.  And, we don't know that it is a random address. If you keep scrolling into the data area then it might be garbage or it can be code loaded from an external source. We don't know. Only you know. We can kind of recognize some of the data areas as well but VSCode asked for disassembly in that area -- because you scrolled into that area or the PC stopped there. How come gdb is giving us disassembly in that area is another question.

What is that symbol that everything is relative to? Do you know what is (or should be) at that address? It really helps if you do some investigation. I am the only person maintaining this and it is free work. It would be nice if people help.

In the Debug Console you can see requests coming from VSCode, which I translate to queries to gdb and then send the results back to VSCode with minor  formatting. That is it. You have ALL the info right there.

Also, GDB may not have the information that ARM does. For instance, I think Visual Studio C++ compiler creates side files  like *.pdb (not in the executable) that contains additional information. Where did you see ARM show you more/correct information? If it did it while compiling, then so does GCC and it is far more accurate (but not with absolute addresses).

Can you provide the executable? I get nothing to work with from you. No one wants to share anything.

BTW, I have also seen gdb produce right assembly but totally wrong source information.

hongshui3000, 2022-01-29:
[image]
This kind of issues seems to be caused by the disassembler of gdb is not very perfect
my elf/lst：
[elf.zip](https://github.com/Marus/cortex-debug/files/7963203/elf.zip)

haneefdm, 2022-01-29:
Working backward is a LOT harder than spitting stuff out while compiling. GDB works backward. AS/GCC work with a LOT more info. Even so, gdb could have done a better job and maybe it will in the future.

But I have to work with gdb that is 10 years old. You as a user can jump to newer versions. I can't.