Marus/cortex-debug #594: Use gdbinit file

CoreyWilliamson, 2022-02-03:
Hi,

I'm trying to start a debug session but the process is failing when attempting to load executable for some reason.  I tried getting things to work by simply running JLinkGDBServerCL.exe and gdbserver(arm version) separately and it works.  It makes me think that an incorrect or unneeded command is being ran in the gdbserver startup.  Is there any way to create a .gdbinit file and use that instead?  I didn't see an attribute in the documentation for that.

haneefdm, 2022-02-03:
May as some version info of Cortex-Debug? Also, output from Debug Console and gdb-server terminal tab can be helpful

haneefdm, 2022-02-03:
Attaching a launch.json can also be helpful

CoreyWilliamson, 2022-02-03:
Sure! v0.4.10, I can see in the gdb-server terminal tab that the halt command is being ran then a "failure to prepare for programming".  It seems to be a gdb-server init issue so I'm wonder if it's possible to control the gdb commands being ran.

haneefdm, 2022-02-03:
We are now at version 1.2.2 for release version and 1.3.3 for pre-release versions.

Yes, you have near-complete control of commands sent to jlink. Basic commands for all gdb-server is reset+halt, load-program, reset+halt

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

You can see everything going on in the debug console as well if you enable the debug flag `showDevDebugOutput`

CoreyWilliamson, 2022-02-03:
Ahhh ok, it looks like the reset command is breaking it.  Is there any way to disable that?

haneefdm, 2022-02-03:
Your question was if you can create a gdbinit. Sure you can. Just do it the normal gdb way (.gdbinit) or use the `debuggerArgs` to pass additional commands but I am pretty sure it will interfere with what we are doing. I recommend first looking at what commands we are sending and then decide. Maybe you can use on of the pre/override/post commands.

haneefdm, 2022-02-03:
If the reset is breaking, not much will work (restart/reset/etc.) We need the ability to reset the device or some way to get the program to start at the starting point. If you have another way, fine, use it. I have rarely seen a device that cannot be reset. May I ask what device?

Perhaps you want to attach to a running program rather than start at the beginning. You do that by changing 
```
            "request": "attach",
```
In this case, a reset is not used but once you press the reset/restart buttons, a reset needs to work.

haneefdm, 2022-02-03:
https://github.com/Marus/cortex-debug/wiki/Overview

haneefdm, 2022-02-03:
I am going to close this one as I don't really see an issue. You can keep commenting

closed 2022-02-03 by haneefdm

CoreyWilliamson, 2022-02-03:
Hi thank you for the info! As for my case yes unfortunately it can't be reset then halted.  The chip is a RSL15, I went back and double checked with their IDE setup and in fact they request the debug option "Initial Reset and Halt" is unchecked.  I'm guessing it's something with the boot PROM secret code that messes everything up.  Thank you for your help and pointing me to the docs where I found the command to override and run custom commands!  Everything is working well, I did notice in the documentation that the commands "target-download" and "monitor reset halt" were shown but weren't recognized when I tried them in my override commands.  Thanks again for your time!

haneefdm, 2022-02-03:
It may be a syntax issue or how/where to use quotes for instance. Overrides should work and we take them blindly after VSCode substitutions and pass it on. Remember that the format is JSON. Here is a complex example (not JLink) but everything below goes to gdb. And anything that starts with "monitor" gdb forwards to the gdb-server (OpenOCD in this case). `target-download` is a gdb command and `monitor program` is an OpenOCD command. You can use whichever works.
```JSON
            "overrideLaunchCommands": [
                "set mem inaccessible-by-default off",
                "-enable-pretty-printing",
                "set remotetimeout 15",
                "monitor reset init",
                // Comment this next line out if you don't want to reload program
                "monitor program {./build/CY8CKIT-062-WIFI-BT/Debug/mtb-example-psoc6-hello-world.hex}",
                "monitor reset run",
                "monitor sleep 200",
                "monitor psoc6 reset_halt sysresetreq"
            ],
```
Yes, it required a complex song and dance routine to get the program started.