Marus/cortex-debug #595: VS Code's generic debugger now supports viewing and editing binary data

hongshui3000, 2022-02-07:
Will cortex debug be updated accordingly?

haneefdm, 2022-02-07:
We already have it. It is called Memory View. Based on addresses (which can be C expressions).

We are waiting for a new GUI for it coming from MS. See my TODO list. I modified the MS Hex editor they used, fixed a couple of issues but then they changed the GUI to use the side panel and the debug experience went to hell. MS debugger people are trying to integrate it into the debugger as well.

So, use what we have for right now. It has been there for more than three years now

closed 2022-02-07 by haneefdm

hongshui3000, 2022-02-08:
I have nothing to complain about MEM view, it's good enough. I'm looking forward to MEM write/edit. The MEM view is currently read-only

haneefdm, 2022-02-08:
Yes, I am waiting for MS to add it properly. We don't have the resources to do GUI stuff. Creating editors is something I am not doing. Priority is less than zero for me.

https://github.com/microsoft/vscode/issues/126268

In the meantime, use GDB commands in the Debug Console.

hongshui3000, 2022-02-08:
[image]

Looks like their latest release already supports editing?

haneefdm, 2022-02-08:
Yes, that is good news. I was thinking it would be here in Dec. So, I will add it to the queue, give it a couple of releases for it to mature. They make monthly releases.

Unlike the disassembly, I really doubt they did the memory view like all the other debuggers -- the ability to have multiple memory views. We will see.

We have had the main part of the protocol part done for a couple of years now