Marus/cortex-debug #597: GDB executable "docker exec -it armdev \arm-none-eabi-gdb.exe" was not found

JohnBerg60, 2022-02-08:
My situation:
I provide docker containers for my team to run arm cortex projects.  In vscode, a docker container get started (and or build) when vscode launches from a folder. 
In the container are al tools to compile, flash etc. Remote is not a option, because I do not wan't to distrubute git account info, and othere ssl sensitive data. Also swiching every time to a remote container was to much of a hastle.

Every thing is working fine, my team can compile, flash, test with the Dockerfile/docker-compose provided over git.

Except debugging .. 

My launch.json:
```
{
            "cwd": "${workspaceRoot}",
            "name": "Debug (default)",
            "request": "launch",
            "type": "cortex-debug",
            "showDevDebugOutput": "both",
            "servertype": "stutil",
            "armToolchainPath": "docker exec -it armdev ",
            "executable": "./bin/firmware.elf",
            "svdFile": "STM32F091x.svd"
 }
```
As cortex-debug assumes correctly it is on windows, it places a trailing slash before the ```arm-none-eabi-gdb``` and adds a ```.exe``` extension. 

Is there a possibility to set the gdb runtime to something like: ```gdb = docker exec -it armdev /usr/bin/arm-none-eabi-gdb``` and the let cortex-debug capture the output?

haneefdm, 2022-02-08:
`gdbPath` needs to be a full pathname (not an additional command that has to run) to gdb that is available from within docker. GDB always runs locally I am assuming. We do check to see if the executable exists. We pass arguments and open pipes to gdb executable, btw. So, it should look like a normal program. There is no extra checking but the string you give us, we give to NodeJS to run. If it runs, it runs.

Haven't used docker in a long long time but if you want to run `ls` you simply do an `ls` right? Should be same with gdb?

Note: AFAIK, when you run VSCode in a Docker/WSL folder, our extension is actually running in the remote (docker) side. We are not considered a UI extension, so we don't run on the host.

haneefdm, 2022-02-08:
You can also set `gdbPath` as a just `arm-none-eabi-gdb` instead of a fullpath, but then it has to be in your PATH

JohnBerg60, 2022-02-08:
Just to clarify: I am running vscode out of windows 10. The compiler and all of the tools are in a docker container. Nowadays, that also could be a **azure** or **aws** container.

When I run ```docker exec -it armdev``` I get a bash prompt into the dev container.

A compiling task (in tasks.json) looks like this:
```
"label": "build",
            "type": "shell",
            "command": "docker",
            "args": [
                "exec",
                "-it",
                "armdev",
                "make"
            ],
```
this basicly does a make of my project. Flashing the firmware is executing a ```make flash``` from the docker command line. In the Makefile for make flash it says:  ```st-flash --reset write $(BUILDDIR)/$(TARGET).bin 0x8000000```
This all works perfectly fine, executing in the container, without local install in windows.
We have containers for AVR, ESP32 and ESP32 RISC-V. Installing all this stuff in windows is not a option.

haneefdm, 2022-02-08:
I never said you should use the windows version. That would totally make docker/WSL useless. Please use the docker installed version of gcc/gdb/etc. That is why I was using the `ls` example.

I don't know how the tasks work, but I am thinking, they also do not need the `docker exec -it armdev make` stuff. Just 'make' should have probably worked.

Here is an example from Microsoft for a container...

https://github.com/microsoft/vscode-remote-try-cpp/blob/main/.vscode/tasks.json

I see no complicated prefixes running command wrappers and such.

JohnBerg60, 2022-02-08:
Those examples run vscode inside the container .. (so remote). That has al kinds of issues with git, ssl, vscode not correctly showing git changes, compile speed .. to name a few.

I run vscode from windows. When I open a terminal, I get a ```cmd``` prompt, or ```powershell```, but not bash. ```ls``` is a linux command, and does not work from cmd or powershell.

haneefdm, 2022-02-08:
Okay, then you are using the remote development env. backward from how I imagined. I have reached the limit of how I can help you.

Maybe someone else can help. Pretty sure others are already using it successfully.

As an aside -- may not apply to you:

I do launch VSCode from windows but open a folder that is in WSL and all of a sudden, it becomes a "remote" thing, and I do get shells that are unix like and `ls ~` works just right like in Ubuntu (no MSYS/cygwin involved). I can have two VSCodes side by side, both started from Windows showing different folders (one from WSL and one native Windows) and they are totally different in what shells they show. And compile speed is precisely why this makes sense to me. I had no issues with git either.

Key is what folder did I open within VSCode and it morphs into something different. Again, my experience with docker+VSCode is non-existent.

Also, if we cannot see all the folders and files from within VSCode, then your debugging experience will be pretty useless.

haneefdm, 2022-02-08:
Oh, I would appreciate if you can post your learning here -- whatever made it work, to educate me and others. Educating me could mean adding something helpful to the extension.

JohnBerg60, 2022-02-09:
As the c/c++ extension from MS has the same problem, I posted a feature request:

[feature request #8814](https://github.com/microsoft/vscode-cpptools/issues/8814)

lets see what the guys over there have to say. 

note: missing support in the c/c++ can be overcome by putting a portable version of mingw64 on every machine of my team. A awful hack, be for now we can move on.

haneefdm, 2022-02-09:
Even if they fix it (just running gdb using the command you specified), it may not work for us.

Sometimes we have to kill gdb because it gets stuck. Well, which process do we kill? We expect to run in the same env. as gdb and talk to it via stdout/stderr/stdin. Since you are using `-it` it may work. But when a breakpoint is hit, we/VSCode look for source code. gdb will point us at a path in docker and we most likely won't find it. Then, you would be limited to assembly debug and not even sure if that will work.

PS: You can replace the `gdbPath` with a shell script that does an `exec` with your command and all arguments provided to it. Just to see how well it works.

PhilippHaefele, 2022-02-09:
Did you already tried using https://code.visualstudio.com/docs/remote/containers which should allow you to easily build and debug inside the container and still executing VSCode (and most likely Git management) on Windows.
The only problem you will have is on how to connect to your (USB?) Debug probe with stutil.
There seem to be a solution named usbipd-win for WSL out now, but not sure how (well) this works!?
At least for Segger debuggers there's another solution available i know of (J-Link Remote Server)

haneefdm, 2022-02-10:
> Those examples run vscode inside the container .. (so remote). That has al kinds of issues with git, ssl, vscode not correctly showing git changes, compile speed .. to name a few.

A few comments ago, this was mentioned. Doesn't want to open the container in VSCode. I am not sure how that can ever work for interactive debugging if we can't see source files with same paths that gdb does.

Btw, I am working on the WSL solution so all gdb-servers can work. It has been slow going. Same thing could work (with minor mods) for Docker and SSH as well.

haneefdm, 2022-02-10:
That means gdb could not find the path used by VSCode. It is not clear to me which gdb you used. If you are using the. VSCode (and thus us) gave gdb the file and line number but gdb could not find it. This can happen if the current working dir. is not right or code was compiled in way that sources cannot be associated with instructions.

closed 2022-03-29 by haneefdm