Marus/cortex-debug #599: Segger RTT-Support: _SEGGER_RTT not found automatically

Rehcsif, 2022-02-11:
Hi,
I tried out rttConfig and it works fine. However, the automatic search for the address of symbol _SEGGER_RTT does not seem to work.
when I set `"address": "auto"` I get the following message in the debug console:

```
Could not find symbol '_SEGGER_RTT' in executable. Make sure you compile/link with debug ON or you can specify your own RTT address
```
However calling `info address _SEGGER_RTT` manually in the debug console gives me the correct address I can use to specify in the config. On top of that completely deleting the the entry `address` in the launch.json that it looks like the following:

```json
            "rttConfig": {
                "enabled": true,
                "decoders": [
                    {
                        "port": 0,
                        "type": "console"
                    }
                ]
            },
```

Does also work. In the console I see `Executed SetRTTAddr undefined` but the RTT console opens and works just fine. So the issue I want to address is that setting `"address": "auto"` does not work but setting nothing does work? Maybe this could be added to the docs or changed?

haneefdm, 2022-02-11:
Ummm. That is serious. I am using RTT and test with that before every release and it does work for me. Not sure what is going on.

On your other issue, you generated a couple of text files after running objdump and nm. I was wondering if you could grep in there.

You are using JLink so it (kinda) worked because it will search the whole memory. With OpenOCD, it would have failed.

If you can attach those files, that would be even better.

haneefdm, 2022-02-11:
Attaching the executable would be even better of course

haneefdm, 2022-02-11:
FYI: Here is my config and this is what I see with JLink
```JSON
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "port": 0,
                        "timestamp": true,
                        "type": "console"
                    }
                ]
            },
```
In this build, if you do not set the address to anything, it will become "auto" instead of remaining "undefined". Had to do that because we also support OpenOCD.
```
Resetting target
Created RTT terminal for channel 0 on tcp port 19021
Executed memU32 0x8003338 = 0
Executed memU32 0x800333c = 0
Executed memU32 0x8003340 = 0
Executed SetRTTAddr 0x8003338
```
The second line above is from the extension. Rest are from JLink

Rehcsif, 2022-02-11:
I looked at the output of `objdump` and `nm` in both I cannot find any symbol _SEGGER_RTT.
However, I guess I know the issue, and it's more a problem in our project structure than in the plugin.

```json
        {
            "executable": "../build/outer.elf",
             //....
            "preLaunchCommands": [
                "add-symbol-file ../build/inner.elf 0x80701B0",
            ]          
        },
```

The initialization of SEGGER RTT is done in the inner.elf. If I produce some output for this file I can find a _SEGGER_RTT symbol and the corresponding address. This also explains why I am able to get an address when executing `info address _SEGGER_RTT` manually in the debug console, because then the other file is already loaded ...

If I call the initialization in the "outer.elf" the plugin is able to find the address when using `"address": "auto"`

haneefdm, 2022-02-11:
You can try the following.
* Use the `inner.elf` as *the* `"executable"`. This way both gdb and us can use the same symbols and this is what you will be debugging anyways. For globals, statics, disassembly, this is what matters.
* You can either use `"preLaunchCommands"` or even better perhaps `"loadFiles"` to say what is actually programmed. I see you used `0x80701B0` not sure what that does or why the elf file does not already have that address.

FYI: With load files, you can load multiple hex/elf files if you wish.

What do you think?

haneefdm, 2022-02-11:
I am going to close it because I currently don't see an issue with RTT

closed 2022-02-11 by haneefdm

Rehcsif, 2022-02-12:
Yep this shouldn't be an issue with RTT, it is our configuration. 
When specifying `inner.elf` as the "executable" plugin is able to correctly determine address ;)