Marus/cortex-debug #601: Support cJTAG interface

albertskog, 2022-02-24:
J-Link supports the cJTAG interface (`JLinkGDBServerCLExe -if cjtag `), but cortex-debug does not accept this value for `interface`.

Happy to provide a PR if I could get a hint where to look.

haneefdm, 2022-02-24:
It is already supported. See `interface`

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

albertskog, 2022-02-24:
My bad, a separate configuration error on my part made me think this was a bigger problem. Yes, it does work, but with yellow squiggles and a warning. Hovering over `cjtag` shows a popup saying "Value is not accepted. Valid values: "swd", "jtag"."

[image]

[image]

[image]

haneefdm, 2022-02-24:
I can add `cjtag` in the next release. Wondering if this goes beyond JLink. We have half a dozen other probes/gdb-servers to check. Even for JLink, trouble is that it is not documented (not in Wiki or in Manual).
[image]

For jlink it is simple to just pass it along but for others, there is some logic to convert to appropriate command-line options or other methods.

albertskog, 2022-02-24:
Yes, it took me a while to figure out how to enable it! I have been going through today's browser history trying to figure out how I found it, only thing I can think of is that there is a drop-down menu for target interface in the GUI version, `JLinkGDBServerExe`. Could not make a screenshot while the menu was open, but the available options are:
* JTAG
* cJTAG
* SWD
* FINE
* ICSP

[image]

albertskog, 2022-02-24:
I had a look at a couple of the other servers. If you want to trust their documentation is more up to date, I found this: 
## OpenOCD
https://openocd.org/doc-release/html/Debug-Adapter-Configuration.html#Transport-Configuration
Not sure are all relevant for debugging, but supports JTAG, SWD, SPI and SWIM

## PyOCD
https://pyocd.io/docs/options.html 

dap_protocol | str | 'default' | Wire protocol, either 'swd', 'jtag', or 'default'.
-- | -- | -- | --

closed 2022-03-28 by haneefdm