Marus/cortex-debug #602: JLink RTT only allows a single port/channel

stephendpmurphy, 2022-02-24:
Hi all,

I just have a question regarding multiple channels when using a JLink and RTT. In the Wiki the following can be found:
```
Channel sharing: You can use the same RTT channels in multiple ways. Cortex-Debug reads the channel data from OpenOCD/JLink once and distributes to all subscribers (terminals & graphs & logfiles). For instance, you can plot a channel and also look at its binary data in a terminal. Just use two decoders with the same channel (actually called port) number.
```

However, if I attempt to add more than a single decoder (as the above would indicate is possible) where I send data to a binary decode as well as a graph (Both from port 0) - I get the error that states
```
Currently, JLink RTT can have a maximum of one decoder and it has to be port/channel 0
```

Is there a way to have the same channel decoded in multiple ways or am I stuck with a single decoding type of a single data channel.

Relevant config used:
```json
{
	"version":"0.2.0",
	"configurations":[
	{
		"name":"LPC5526",
		"type":"cortex-debug",
		"request":"launch",
		"servertype":"jlink",
		"cwd":"",
		"executable":"./output/debug/controller.elf",
		"device":"LPC5526",
		"runToMain":true,
		"v1":false,
		"serverArgs":[
			"--target",
			"LPC5526"
		],
		"rttConfig":{
			"enabled":true,
			"address":"auto",
			"decoders":[
				{
					"port":0,
					"type":"binary",
					"encoding": "unsigned",
				},
				{
					"port":0,
					"type":"graph",
					"encoding": "unsigned",
					"graphId":"1",
					"scale":1
				},
			]
		},
		"graphConfig": [
			{
				"label": "Graph 1",
				"timespan": 30,
				"type": "realtime",
				"annotate": false,
				"maximum": 10,
				"minimum": 0,
				"plots": [
					{
						"graphId": "1",
						"label": "data 1",
						"color": "#53753c"
					},
				]
			},
		]
	}]
}
```

haneefdm, 2022-02-24:
Thanks for reporting this. The error checking was wrong in the front end. The backend does it properly but unfortunately, you can't get to it. So, yes, you can have as many decoders as you want but only one RTT channel in FW and it has to be '0'. This is a limitation of JLinkGDBServer that I wish did not exist.

It will be fixed soon. Maybe as early as this week.

stephendpmurphy, 2022-02-24:
Thanks @haneefdm :+1:

haneefdm, 2022-03-28:
This should be fixed in latest release. Let me know it isn't

closed 2022-03-28 by haneefdm