Marus/cortex-debug #605: Adding support for custom windows

microwavesafe, 2022-03-04:
I have no idea how complicated this would be, but I recently started using a gdb script to compile a list of properties of my FreeRTOS tasks [gdbinit-FreeRTOS](https://github.com/microwavesafe/gdb-freertos-helper/blob/main/.gdbinit-FreeRTOS). This covers what I want to know about tasks and I like that it only runs when I call it as it is slow (calculating free stack space requires iterating over a loop per task).

It got me thinking though. A while ago I opened a pull request trying to hack the gdb code to do the same thing, but it wasn't generic and maintaining would be difficult. But if we could run custom gdb scripts with a button and have the output displayed in the UI, that would cover everything we need to do. A different script can be created for each RTOS and users can create custom scripts. The script could print as a JSON string and the cortex-debug extension shows that as a table in the UI.

I'm not sure how feasible any of that is, but I liked the idea so thought I'd mention it.

haneefdm, 2022-03-10:
@microwavesafe Yes, you can have custom windows and they don't even have to be part of this extension. This is because you can interact with any debugger (you can get events for start/stop and also any customer events that a specific debugger might have). You can also make custom requests. That is how our own front end works. Peripheral/Register/Memory windows  for instance. Peripheral window is a good example because it has near zero support from the backend. Only thing our backend provided was a generic read-memory request.

But on the subject of RTOS windows though. I will see if I can start a framework for that. We can start with FreeRTOS but others can add their own. Are you good with CSS/HTML? I am pretty bad at it, but this is what is needed to display a table.

hongshui3000, 2022-03-11:
Microsoft Tools
https://devblogs.microsoft.com/cppblog/vscode-embedded-development/
https://github.com/microsoft/vscode-embedded-tools/issues/3

microwavesafe, 2022-03-11:
This new extension looks interesting, it may provide a quick avenue to more RTOS features. I'll close this for now and subscribe to the other issue.

haneefdm, 2022-03-18:
Hi @microwavesafe, While waiting for MS folks to allow our extension to work with theirs, I implemented something. It was not that simple, too many things can go wrong. We have to do many small reads and the user could have stepped or continued and we have to fail gracefully. I will merge upstream soon so you can look at what I did. Hopefully, adding more RTOSes is not as hard as the first one and others could help. I know it is not pretty but we can do that using CSS externally (baby steps)
[image]

There can also be garbage or empty data at the start
[image]

Collecting data from gdb can go wrong in a hundred different ways because VSCode/users are interacting while I am collecting data. This is unlike Registers/Peripherals windows where very few requests to gdb are made. Here, we have many many tiny ones.

The infrastructure I have will scan for all known OSes and quickly eliminate any that don't qualify. Haven't figured out when to enable this view. Right now, it is always there even if not RTOS is detected.

Looking for a better name for the Panel. Don't lie 'RTOS INFO'

haneefdm, 2022-03-18:
If anyone wants to play with it, clone the repo and uncomment the following line.

https://github.com/Marus/cortex-debug/blob/4b33de620f4caa661c0999ae9e442e216cb7aae7/src/frontend/extension.ts#L57

All relevant code is in `./src/frontend/rtos` and generic stuff that gets embedded into HTML will end up in `./resources` at the top level

microwavesafe, 2022-03-18:
This looks good! It has all the information I need. I don't need it all the time, but when I do it's incredibly useful. Having it run manually with the refresh button rather than on every pause like the other windows makes sense. It's expensive so only run it when when definitely required.

How about just 'RTOS' for Panel name?

haneefdm, 2022-03-18:
@microwavesafe I thought about a simple name like 'RTOS' but wondered if it too generic to collide with someone else's. I think I will change it to RTOS and see if it collides.

For everyone's information, for tracking purposes, here was my interaction with the MS folks.

https://github.com/microsoft/vscode-embedded-tools/issues/3

Our backend debug adapter provides zero special support for this. It is all done using the Debug Protocol that everyone has to follow. But implementations can vary (especially in error situations) so our extension may not work out of the box with what MS has.

Hey, maybe our implementation is actually better.

haneefdm, 2022-03-19:
A few more columns (optional if those features are not enabled in RTOS FW). Not sure how to right/left justify columns yet

[image]

Very rare, but I sometimes see wrong info in the RTOS window and/or the CALL STACK window. They should agree 100% but they don't. Super rare. Once I saw the same task listed twice. I may be pausing during a task switch where RTOS structures are not coherent.

haneefdm, 2022-03-20:
Latest snapshot. It is going live shortly
[image]

PhilippHaefele, 2022-03-21:
Awaited this feature as the last main one missing in Cortex Debug.

I will look into adding support for uC/OS-II soon.

embOS is another one that will have a lot of fans, but i'm personally not into it...

@haneefdm Thank you very much for your efforts making this extension so useful

haneefdm, 2022-03-22:
It is now alive as a regular release. Your VSCode may update shortly. I do have a question. Can you teach me how to get the "Event Object" column? I can probably figure it out but if you are a pointer...

[image]

It is from [this page](https://mcuoneclipse.com/2017/03/18/better-freertos-debugging-in-eclipse/)

I do have a bug in what is released. I was not guarding against having a thread name of size `pcTaskName[0]` -- actually not sure what various compilers do.

haneefdm, 2022-03-22:
Yes, you have to enable it. One of several ways...

https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#new-features

microwavesafe, 2022-03-22:
I should really of checked there first, thanks. It's looking good! Really useful information. For "Stack Peak" I'm getting "Read mem fail" not sure if there is something I should be implementing my end?

A couple of ideas that you can feel free to ignore. There are times where the GDB overhead may be an issue, is it an idea to add a pause button? I realise we can turn off the panel, but it would be nice to able to just pause it temporarily.

I was also wondering if the column order could to changed to help make sense of the numbers to something like this

``Stack Beg / Stack Top / Stack End / Stack Size / Stack Used / Stack Free / Stack Peak``

Obviously this is just personal preference and purely subjective.

Thank you for implementing this!

haneefdm, 2022-03-22:
Sorry, you should not have to do anything for top/peak. Should have been in my TODO notes. I don't have an example with stack end yet and have not tested that. Will push something out soon.

Yes, I like the order of the stack info.

Not sure if you noticed but if you are not literally looking at the RTOS panel, zero updates occur. You don't have to remove the RTOS panel, just don't expose it. Expose something else. The minute you expose it, it will do an update if the program is in paused state.

I will think of a pause button implementation. It may be quite a bit of work and may not be worth it. You can however step/continue and RTOS updates will be (silently) aborted -- ihowever,  It should tell you that the data you see is stale. The last line has the time stamp, amount of time it took to gather data, and if the information could be stale. So, nny buttons you press for the main debugger operation has precedence over this.

I have already sped the gdb interaction a bit, but with the peak info, it will slow down again.

microwavesafe, 2022-03-22:
I did not know it has zero overhead when not viewed, that is good to know and may well work around pause the issue.

haneefdm, 2022-03-22:
You can enable gdb traces and you won't see any related to RTOS. When you expose the RTOS from being obscured, you will see for a brief moment that it has just one line saying something like "nothing here to see".

You can also. 1) Expose Terminal 2) Continue the program with no breakpoints and 3) Expose RTOS and you will see a nearly empty window.

Not sure if Microsofts version took the pain to do this :-)

haneefdm, 2022-03-22:
In ARM, does the stack always grow down? I knew it is customizable in HW but most people use it growing down. Yet my stack calculations were backwards I think.

I need to figure out how to detect (reliably) which way the stack grows. I think the ABI only says stack grows down.

fdischner, 2022-03-22:
> In ARM, does the stack always grow down? I knew it is customizable in HW but most people use it growing down. Yet my stack calculations were backwards I think.
> 
> I need to figure out how to detect (reliably) which way the stack grows. I think the ABI only says stack grows down.

Just came here to report exactly this bug. Otherwise, it's working well for me.

The ARM [Cortex-M3 Devices Generic User Guide](https://developer.arm.com/documentation/dui0552/a/the-cortex-m3-processor/programmers-model/stacks) says this about the stacks:

> The processor uses a full descending stack. This means the stack pointer holds the address of the last stacked item in memory. When the processor pushes a new item onto the stack, it decrements the stack pointer and then writes the item to the new memory location.

The M0/M4/M7 documentation has the same wording. I'm not sure if this applies to all Cortex-M processors or if it's selectable by the vendor, but I've never seen one where the stack grows upward. A quick search didn't turn up anything for the A or R series, but all the FreeRTOS ports (A, R and M series) define `portSTACK_GROWTH` to `(-1)`. Additionally, all the thread-aware debuggers I've looked at (openocd, j-link gdbserver, and ozone) assume the stack grows down. So unless someone reports otherwise, it's probably safe to assume that the stack will always grow down (at least for Cortex-A/R/M parts).

haneefdm, 2022-03-22:
@fdischner Thanks for checking. Even the high end Cortex-A devices do the same. I think you pretty much have to follow that standard even if the HW supports the other way. Reason is binary compatibility with libraries, Operating Systems, etc. Same can be said about endianness. HW is configurable for either but only 'little' works in the real world.

Here is the updated screenshot. Will go public shortly as this is the only change. Oh I enabled launching memory view from the variables Window but I don't like it. Have to have a chat with MS. They request 256KB every time.

[image]

Sorry, with VSCodes `grid` I could not figure out how to span columns (for stack label) for the headers. I know how to do it with regular HTML but then a lot of work to make it work with themes.

@PhilippHaefele , I refactored quite a bit and move code into common area so it will be less work for other OSes. I need to do more but you can take a look at your convenience.

closed 2022-03-26 by haneefdm