Marus/cortex-debug #606: Cortex-Debug can't find ST-LINK GDB Server, ARM toolchain, or anything as far as I can tell.

JonathanCarter990, 2022-03-05:
Apologies in advance if I'm a clueless noob here. I'm struggling to find any documentation regarding the stutil/stlink functionality, or any examples of doing this in Windows.

I'm attempting to debug an STM32 Discovery board using the ST tools on Windows 10. The board has a built-in ST-Link.

In my settings.json I have:
```
"cortex-debug.gdbPath": "ST-LINK_gdbserver.exe",
"cortex-debug.stutilPath": "st-util.exe",
```

In my PATH, I have the bin folders for the ST-Link GDB server and ST-Link, as well as the bin folder for my GCC.

In my launch.json I have:
```
"version": "0.2.0",
    "showDevDebugOutput": true,
    "configurations": [
        {
            "name": "STlink launch",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/location/of.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "stutil",
            "device": "STM32F407VG",
            "interface": "swd",
            "runToEntryPoint": "main", // else it starts at reset handler - not interested
            "v1": true,
```

However, I receive the error message "GDB executable "ST-LINK_gdbserver.exe" was not found. Please configure "cortex-debug.armToolchainPath or "cortex-debug.gdbPath" correctly.

I also tried absolute paths to no avail.

I should also add that I tried commenting out the gdbpath and stutil path and defining an arm toolchain path instead. The arm toolchain GDB exe was also not found.

Why can't cortex-debug find anything?

Thanks

haneefdm, 2022-03-05:
`"stutil"` and `"stlink"` are different (competing) tools for the same purpose. They are both gdb-servers. `"stlink"` is newer and mainatined by ST as you know. The `servertype` determines which one you want to use. I recommend `"stlink"`. It appears you are mixing the two. Also, you have gdb confused with gdb-server -- easy to do

So, use the following in launch.json

```
            "servertype": "stlink",
```

If you have everything in your PATH, no other settings are needed. Note that we don't use gcc/gdb, but use tools that look like `arm-none-eabi-gdb`, `arm-none-eabi-objdump`. These should be in the bin folder you mentioned. So, remove the "cortex-debug.gdbPath" -- which btw, should have been a path to `gdb` and not the gdb-server.

If you do want to use the settings the following to matter
```JSON
"cortex-debug.stlinkPath": "<path-to-ST-LINK_gdbserver.exe>",
// Following is the path ARM executables gdb, obdjump, nm, etc.
"cortex-debug.armToolchainPath": <path-to-bin-dir-containing-arm-tools`",
```

If you want to learn more visit our WiKi. An overview of how our extension works is captured here
https://github.com/Marus/cortex-debug/wiki/Overview

haneefdm, 2022-03-05:
Btw, "v1" only applies to "stutil" and for very old boards.

If you have everything in your PATH, just the template from 'Add Configuration...' button should have sufficed (except for editing the `executable` name. Not even sure if the `device` name was needed -- pretty sure stlink should already know that). The following button appears when editing `launch.json`

[image]

Reopen if you are still having issues.

JonathanCarter990, 2022-03-07:
Thanks for the quick reply, Haneefdm. I'm unable to reopen the issue.

I switched to using the default configuration. Unfortunately, GDB was not found, even though I've added it to my PATH.

I tried adding to settings.json:
`"cortex-debug.armToolchainPath": "C:/Program Files (x86)/GNU Tools ARM Embedded/9 2019-q4-major/bin"`

I see arm-none-eabi-gdb.exe in the /bin/ folder.

But I receive "GDB executable "C:/.../bin" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.

Otherwise I am using the automatic configuration except for my elf:
       ```
 {
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/path/to/my.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "showDevDebugOutput": true,
            "servertype": "stlink"
        },
```

haneefdm, 2022-03-07:
It appears that you also (inadvertently) set `cortex-debug.gdbPath`. Check both your workspace and user settings to make sure that there are no extraneous settings.

JonathanCarter990, 2022-03-07:
I don't see any gdbPath in my User or Workspace settings.

However, maybe one thing that might be causing a problem is that I'm accessing the source code remotely via the SSH extension. The source code and ELF are on a remote Linux machine, but my discovery board is plugged into my local Windows machine.

haneefdm, 2022-03-07:
Oh, that setup is very difficult to make it work. But it has nothing to do with why we are seeing this message.
```
"GDB executable "C:/Program Files (x86)/GNU Tools ARM Embedded/9 2019-q4-major/bin" was not found. Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.
```
The path displayed should have `arm-none-eabi-gdb.exe` in it according to your source code
https://github.com/Marus/cortex-debug/blob/a23be8f1ce0d07d2eb3a74dedb345381fc069f07/src/gdb.ts#L619

If you using VSCode remote setup, then we don't support it out of the box. We are working on that. Some people have made it work with WSL but I can't say it will work with SSH and how VSCode even deals with it.

https://github.com/Marus/cortex-debug/issues/467

If you are using the the VSCode remote extensions, then our debugger is really running on the Linux side and not on Windows and your settings in VSCode linux side may be what matters. Still, that message does not make sense. How all that works with VSCode remoting is not something I am an expert at.

JonathanCarter990, 2022-03-07:
Haneefdm, thanks for all the help. It seems like my best bet is to install linux locally and proceed with local debugging.

closed 2022-03-07 by JonathanCarter990

JonathanCarter990, 2022-03-09:
I finally got the debugger working and figured I should come back to conclude this thread.

After installing Linux, I created the default ST-Link launch file and filled in my details:

launch.json
```
"cwd": "${workspaceRoot}",
"executable": "path/to/my.elf",
"name": "Debug STM32",
"request": "launch",
"type": "cortex-debug",
"showDevDebugOutput": true,
"servertype": "stlink"
```

I received the same message about GDB not being found, so I added armToolchainPath to the settings file:

settings.json
```
"cmake.sourceDirectory": "${workspaceFolder}/my-workspace",
"cortex-debug.armToolchainPath": "/usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/",
"cortex-debug.stlinkPath": "/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.200.202202231230/tools/bin/ST-LINK_gdbserver",
```

And now it's good times in firmware land.

Thanks for your help, Haneefdm!