Marus/cortex-debug #607: Adding new register command

crowlec8, 2022-03-05:
I am trying to add a new command that simply asks a question when debugging. However I keep getting the following error when I try to run the command in the debug window.

"Command 'Ask Question' resulted in an error (command 'cortex-debug.ask' not found)"

I have added
vscode.commands.registerCommand('cortex-debug.ask', () => {
    vscode.window.showInformationMessage("How was today?", "good", "bad");
}),
inside the context.subscriptions.push, in the constructor of extension.ts

I have also added "onCommand:cortex-debug.ask", to activationEvents and 
{
    "command": "cortex-debug.ask",
    "title": "Ask Question"
},
to the commands, both which exist in package.json

haneefdm, 2022-03-05:
This is a "how to write a VSCode Extension" question. I suggest you go through VSCode's tutorials and build some small stuff, try to emulate what you are trying to do in this extension.

Also, put your repo (fork) on github and then you can also point to source code in issues like this
https://github.com/Marus/cortex-debug/blob/b8cb820f1b4bf010fbaf0743f24c38c809fbe490/package.json#L4

or formatted with tripple-back-quotes like this (recommend learning github markdown syntax).
```JSON
    "activationEvents": [
        "onDebugResolve:cortex-debug",
        "onStartupFinished"
    ],
```

It may be that you never compiled after making changes. Not clear to me how you are running the extension and exactly where you are seeing the error. Do you know if the extension is indeed activated?

crowlec8, 2022-03-05:
I have emulated what I am trying to do on a sample extension. I have done the exact same in the cortex-debug code too. I have cloned the repository to my laptop and I am editing and running it through vscode. The extension is activated also, as I am able to change a few things on the debugger itself (i.e. register names).  I compile and debug the code which returns a debug window. Here I have tried using ctrl+shift+p to open the command pallet.Doing this I search “Ask Question“, which appears, however when I hit enter the error appears.  Sent from Mail for Windows From: Haneef MohammedSent: Saturday 5 March 2022 19:20To: Marus/cortex-debugCc: crowlec8; AuthorSubject: Re: [Marus/cortex-debug] Adding new register command (Issue #607) This is a "how to write a VSCode Extension" question. I suggest you go through VSCode's tutorials and build some small stuff, try to emulate what you are trying to do in this extension.Also, put your repo (fork) on github and then you can also point to source code in issues like thishttps://github.com/Marus/cortex-debug/blob/b8cb820f1b4bf010fbaf0743f24c38c809fbe490/package.json#L4or formatted with tripple-back-quotes like this (recommend learning github markdown syntax).    "activationEvents": [        "onDebugResolve:cortex-debug",        "onStartupFinished"    ],It may be that you never compiled after making changes. Not clear to me how you are running the extension and exactly where you are seeing the error. Do you know if the extension is indeed activated?

haneefdm, 2022-03-05:
How am I supposed to read what you have posted? I am not a bot.

crowlec8, 2022-03-05:
Apologies, I dont know how that happened, 

I have emulated what I am trying to do on a sample extension. I have done the exact same in the cortex-debug code too. I have cloned the repository to my laptop and I am editing and running it through vscode. The extension is activated also, as I am able to change a few things on the debugger itself (i.e. register names). 
 
I compile and debug the code which returns a debug window. Here I have tried using ctrl+shift+p to open the command pallet.Doing this I search “Ask Question“, which appears, however when I hit enter the error appears.

haneefdm, 2022-03-05:
You must have replied to an email notification which is why it got all garbled.

Follow what is going on for `examineMemory` as an example. It is registered for the `commandPalette` as well as `commands` in package.json besides in the code. All of this is required by VSCode

How do you know that the version that got activated is yours and not one that is released? Only way to know is to put a breakpoint in the initialization code or display a message.

Curious, what are you trying to do anyway?

crowlec8, 2022-03-05:
Thanks you for the advise. I will try this. 
I uninstall the released version as I was not able to make any simple changes to the code. Once uninstalled I was able to see changes I made to the code in the debug window.
I am attempting to create a visualization for the debugger, to help beginners learn more about ARM assembly. I had hoped to create a command which allows me to open a webview which would take in updated data as the debugger executes and shows graphics accordingly.

haneefdm, 2022-03-05:
What you were doing should have worked even with extension installed. Maybe you kissed a step in how to debug.

https://github.com/Marus/cortex-debug#how-to-debug

Closing issue.

Btw, what you sound like doing, you can do it in your own extension. You can see what every debugger is doi ng without modifying those extensions. Anyway, good luck.

closed 2022-03-05 by haneefdm