Marus/cortex-debug #608: separate scope for arguments (from DAP)

GitMensch, 2022-03-09:
So far there are Local, Global, Static and Register https://github.com/Marus/cortex-debug/blob/b8cb820f1b4bf010fbaf0743f24c38c809fbe490/src/gdb.ts#L2139-L2146

Looking at the [DAP](https://microsoft.github.io/debug-adapter-protocol/specification#scope), there is also a scope for Arguments.

Checking on how GDB returns those shows that there's an optional argument "arg" which is only set for those to "1".
Would it be possible/useful to split those out of Locals?

Additional from the DAP: wouldn't it be reasonable to set the optional `presentationHint?: 'arguments' | 'locals' | 'registers' | string;`? It seems those aren't set currently.

haneefdm, 2022-03-09:
The DAP evolves due to many reasons besides VSCode and we can't really keep up with it. Many times, they add stuff after we already implemented something and we don't have the time to constantly monitor their changes. The DAP also adds stuff hoping it is useful to someone/someday. In this case though here is the VSCode class for a Scope in `debugSession.d.ts` (does not define all of DebugProtocol.Scope)
```typescript
export declare class Scope implements DebugProtocol.Scope {
    name: string;
    variablesReference: number;
    expensive: boolean;
    constructor(name: string, reference: number, expensive?: boolean);
}
```
As you can see many of the DAP properties are missing. Generally, if you don't see it here, it will not be used. Just for kicks I added a `presentationHint` and it did nothing. I down't want to upcast/downcast just to add the property that has no benefit. I would also like a formal definition first because it doesn't even list half of the typical scopes out there (closures, etc.).

Abiout separating Locals from Args, doable, but I am not going to. Too much work for very little benefit. In fact some people (including myself) may not even like it. Too much work because of how variable references are handled and how frame and thread info is bundled all into one 32 bit value. It will require a ton of testing to add another dynamic range. Adding Registers was already a pain.

GitMensch, 2022-03-09:
So is this "open to let others add the Argument split" (maybe with an extension setting, if you've implemented that already)?

haneefdm, 2022-03-09:
> So is this "open to let others add the Argument split" (maybe with an extension setting, if you've implemented that already)?

Not sure what you mean. Splitting Locals is not a good idea even from a UX point of view. VSCode auto opens the first scope and splitting it means I either get to see Arguments or Other Locals but not both. Nearly 100% of the time, I want both. Dividing into two groups what generally is a small set is what I don't like.

closed 2022-03-09 by haneefdm

GitMensch, 2022-03-09:
Agreed :-)