Marus/cortex-debug #612: honor frame selection from GDB

GitMensch, 2022-03-10:
When the debugger halts vscode always selects the first frame with debug information, which is reasonable.
But even when there is explicit frame selection the frame in vscodium stays the same and the `--frame` sent to GDB is also as it was before.

Th easiest way to reproduce the issue:

* start debugging
* set a breakpoint to a sub-sub-function
* run (the breakpoint is reached and vscode stops)
* execute "up" on the debug console two times

Current result:
* you see the line of the upper frame 2 times (because it is sent with `--frame 0`) in the debug console
* vscode is still on frame 0

Expected result:
* you see what you get in the the console, and get up each time
* vscode shows the correct source position each time

Currently the frame can only be switched by the vscode ui. This is especially bad when you run GDB scripts which change the frames itself.

GDB communication per logging:

~~~
59-thread-select 1
60-stack-select-frame 0
61-interpreter-exec console "up"
GDB -> App: {"token":59,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["new-thread-id","1"],["frame",[["level","0"],["addr","0x00005555555552f5"],["func","sub_sub"],["args",[[["name","entry"],["value","0"]]]],["file","TEST.c"], ...}}
GDB -> App: {"token":60,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"#1  0x0000555555555214 in sub () at TEST.c:46\n"}]}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-selected","output":[["id","1"],["frame",[["level","1"],["addr","0x0000555555555214"],["func","sub"],["args",[]],["file","TEST.c"],..]}
GDB -> App: {"token":61,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
~~~

ToDo:
* add parsing of `frame->level` to the parsing of `thread-selected` message https://github.com/Marus/cortex-debug/blob/ee4f98361073ad4a1568399fc7ad68abc682ea2a/src/gdb.ts#L596
* send this to the frontend either with a separate or with the same `customEvent`
* on the fronted handle it in https://github.com/Marus/cortex-debug/blob/63479a6c35733325cd39e272ddfef048cd60bb96/src/frontend/extension.ts#L615

What I did not found: how to select the frame? The `ThreadEvent` seems to not have a frame attribute, it possibly needs to be added first? If this is the case, then it could still be useful to implement the parsing and let the frontend receive it along the thread id.

haneefdm, 2022-03-10:
I will look into it. Perhaps gdb did not pass the frame at the time that code was written. GDB's thread handling has evolved and still not right, btw.

Also, why may I ask you are quoting code from 2018?

haneefdm, 2022-03-10:
`ThreadEvent` is part of the MS DAP protocol and yes, it does not have a frame-id in it. So, we can't make one up.

Additionally, what you are thinking may not be right. Here are some details
* All evaluate requests come with the stack/frame context to run the command with. Because gdb is so unreliable with the maintenance of this, we make sure the right one is used.
* We *HAVE* to respect what VSCode tells us when evaluating expressions. Or things like hover won't work. Also 'Watch' variables won't work and neither will the Variables window.
* When you type commands in the console REPL window, they are handled as expression evaluations by VSCode.
* There is no way to communicate back to VSCode that your want the GUI to update itself based on what ever else occurred -- other than exit. And, I agree that it is the right behavior. I think the whole GUI will barf and shimmering will occur

Finally, you are on your own if you are modifying the program context. Sometimes you may get wrong/unexpected results.

At this point, I am not understanding what you are trying to do overall? Not just this issue.

GitMensch, 2022-03-10:
The reason for the quote is easy: the threading code was added back then and I've used the first permalink I've found :-)

Thanks for looking into it, the most interesting question is: how to select the frame in the UI (but having that info on the frontend would be a good thing in any case).

haneefdm, 2022-03-10:
> The reason for the quote is easy: the threading code was added back then and I've used the first permalink I've found :-)

Please don't do that. The code does not resemble that at all, took me a while to figure out what you were talking about.

Programmatically, you cannot select/query the thread/frame in the GUI. Using your mouse, it is easy. Just click on it  ... all relevant windows will react (and get updated) because of that click.

No, we cannot keep it in the frontend because it would be pointless ... it will go stale in an instant and is unreliable to track thread/frame and the gdb-way is a terrible, unreliable way. There is no API (or event) in VSCode either when the user clicks on a frame. I requested such an API a while ago but it was a no-go.

closed 2022-03-10 by haneefdm

GitMensch, 2022-03-11:
> I requested such an API a while ago but it was a no-go.

Can you try to find this request, please? It seems that is currently missing in the dap (and after it is in we can include it).