Marus/cortex-debug #616: Allow for full configurations in "chainedConfigurations" entry

michaeljones, 2022-03-15:
I'm working on an extension that uses this extension and we'd like to dynamically generate debug configurations that are chained together. At the moment, it seems that the "chainedConfigurations" feature uses the `name` field to reference the chained configurations which means that the configurations have to be written to disk in some kind of `launch.json` file for the VS Code API to find them when calling the `startDebugging` function here:

https://github.com/Marus/cortex-debug/blob/6cb42803409fc585a21b2d73e47589a1af83b4ef/src/frontend/extension.ts#L697

As the `startDebugging` API supports either receiving a name or a whole configuration, I was wondering if it was possible to extend the chained configurations definition so that it could either be a name or a whole configuration.

Does that seem like a reasonable thing for this extension to support? I can look at doing the changes if that would be useful.

haneefdm, 2022-03-15:
Sure. I would add another option `"config"` to `launches`. This would override the `"name"`.

But, I think it should still go through the sanitization that occurs in configprovider.ts. Haven't looked up the API but not sure what happens to the folder argument.

haneefdm, 2022-06-17:
I have not made any progress on this. We need to spec this out. I don't completely understand how your extension works. Can you point me to it?

michaeljones, 2022-06-18:
Thank you for checking back in on this. I'm afraid our plans have gone elsewhere now so we aren't so much in need of this. I'm sorry to not make better use of or contribute more to the extension you're helping to maintain. 

For what little it is worth, we experienced some friction with the idea of trying to contribute as we felt like we'd want prettier, eslint and strict mode typescript compilation in place to help with such a task and they are currently absent. Though I realise pointing that out when we're not in a place to contribute resources to tackling those issues is not that helpful.

haneefdm, 2022-06-18:
> we'd want prettier, eslint and strict mode typescript compilation in place

I tried a couple of times and the automated tools totally messed up the code base. I wish I had more time.

closed 2022-06-18 by haneefdm