Marus/cortex-debug #617: Stuck on "Launching GDB: ....." on versions more current than 0.4.10

rift-labs-developer, 2022-03-16:
How am I using Cortex-debug?
Developing in WSL2 in VSCode using gcc-arm-none-eabi-9-2019-q4-major toolchain and jlink.
Using usbip to get usb debugger access in WSL2

launch.json:
```
{
    "version": "0.5.0",
    "configurations": [
        {
            "name": "proj_debug",
            "cwd": "${workspaceRoot}",
            //workspaceRoot is parsed incorrectly by GDB (workspaceRoot uses two \ , which get removed. Need to be defined using forward slashes)
            "executable": "${workspaceRoot}/mypaths/merged.hex",
            "armToolchainPath": "/opt/gcc-arm-none-eabi-9-2019-q4-major/bin",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
            "device":"nRF52840_xxAA", 
            "interface": "swd",
            "showDevDebugOutput":"raw"
        }
    ]
}
```

On cortex-debug 0.4.10 im able to start the gdb server.
The output when launching is:

Reading symbols from '/home/mypaths/merged.hex'
Finished reading symbols
Please check TERMINAL tab (gdb-server) for output from /opt/SEGGER/JLink/JLinkGDBServerCLExe
Launching server: "/opt/SEGGER/JLink/JLinkGDBServerCLExe" "-singlerun" "-nogui" "-if" "swd" "-port" "50000" "-swoport" "50001" "-telnetport" "50002" "-device" "nRF52840_xxAA"

On versions more current than 0.4.10:
- no output at all in debug console on some versions
- on 1.3.5:
-Reading symbols from /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-objdump --syms -C -h -w /home/mypaths/merged.hex
Reading symbols from /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/mypaths/merged.hex
Launching GDB: /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/mypaths//merged.hex
1-gdb-set target-async on

And then nothing happens. Is there something that must be done differently when using versions more recent than 0.4.10 ?

haneefdm, 2022-03-16:
The executable cannot be a hex file -- never expected that. It HAS to be an elf file. Hex files have zero debug information. We now also use the `arm-none-eabi-nm` program to get more debug information.

haneefdm, 2022-03-16:
You can program using a hex file(s) though. See `loadFiles` property for launch.json.

haneefdm, 2022-03-16:
Btw, if it is stuck at that point, that means we're waiting on a response from gdb for `gdb-set target-async on` GDB is either also confused about the hex file (probably not) or something else is wrong. This should never fail.

The next command you should see in 1.3.x is `gdb-version` and we have not seen that. Since you can see the *exact* gdb command line you can execute those commands yourself and see why gdb is not responding.

rift-labs-developer, 2022-03-16:
Thanks so much for helping me, even though it turned out it had nothing to do with cortex-debug. 

The problem was that arm-none-eabi-gdb could not find libncurses5 . Everything works after installing that.

closed 2022-03-16 by rift-labs-developer