Marus/cortex-debug #618: Debuging not working anymore with JLINK

ParalaX002, 2022-03-23:
Hi,

Configuration:
- host: Macbook pro M1 max
- target: NXP MKW41Z, cortex m4
- probe: jlink base
- VSCODE: 1.65.2, extension in version 1.4.1

Since 1.4.0 (it worked ok on version 1.2.2), it's not possible anymore to debug with the jlink probe. The jlink start, connect, then halt at "Waiting for GDB connection".
After that, it hangs for several seconds (like 30s), then program the micro successfully, and everything seems to be working, except that breakpoints are not working anymore. 
Reverting to version 1.2.2 of the extension and everything is back to normal.

Is there some new configuration to do with the new extension? Here is my launch.json:

 ```
 {
      "configurations": [
          {
              "name": "JLink",
              "request": "launch",
              "type": "cortex-debug",
              "cwd": "${workspaceRoot}",
              "interface": "swd",
              "servertype": "jlink",
              "cortex-debug.armToolchainPath": "/opt/homebrew/bin",
              "serverpath": "/usr/local/bin/JLinkGDBServerCL",
              "executable": "TinyIoT.elf",
              "device": "MKW41Z512xxx4",
              "svdFile": "${workspaceRoot}/../../../../Libraries/Embedded/SDK/v223/devices/MKW21Z4/MKW21Z4.xml",
              "rtos": "FreeRTOS",
              "preLaunchTask": "Build Project",
              "serialNumber": "821008201"
          }
      ]
  }
```

Regards

haneefdm, 2022-03-23:
Thanks for reporting this. Could you enable debug (set `"showDevDebugOutput": "raw"` i launch.json and include the output you see in the `Debug Console`. I am mainly interested in the command-line used to launch JLink gdb server. And whatever else is going on with gdb. I also want to  make sure that you can launch gdb from the command line in an external shell.

There is one thing that changed for JLink. A new interface called `cjtag` was added.

haneefdm, 2022-03-23:
Hmmm. All we are doing is a continue and not sure why gdb thinks there is no active thread. Surely there is one and it just reported on thread.

Can you try setting either `breakAfterReset` or `runToEntryPoint`. This is a case where neither is set and something is not working with gdb itself. It just paused and all we are asking is for it to do is `continue`. Differece from before and now is that we used to do more useless work (only because we were slow) before a continue.
```
14-thread-list-ids
-> =thread-created,id="2",group-id="i1"
-> ~"[New Remote target]\n"
[New Remote target]
-> 14^done,thread-ids={thread-id="1",thread-id="2"},current-thread-id="1",number-of-threads="2"
15-thread-info 1
16-thread-info 2
-> =thread-exited,id="1",group-id="i1"
-> 15^done,threads=[]
-> 16^done,threads=[{id="2",target-id="Remote target",frame={level="0",addr="0x000004d8",func="Reset_Handler",args=[],file="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Parameters/Startup.S",fullname="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Parameters/Startup.S",line="120",arch="armv6s-m"},state="stopped"}]
17-exec-continue
-> 17^error,msg="Cannot execute this command without a live selected thread."
```

ParalaX002, 2022-03-23:
I've tried with `breakAfterReset = True`: this time I still have the freezing (around 30s-40s at the same place), but now it seems to be running, I can see the list of tasks (thumbs up by the way this is great! I juste don't have the runtime but I guess this is some config on my side to do).

I also tried with `"runToEntryPoint": "main"`: it works too, but still with the freeze.

haneefdm, 2022-03-23:
I am glad it is (kinda) working but it is not right. It should take no time at all to get started unless programming takes a while which we cannot avoid. Unfortunately, I cannot duplicate what you are seeing. Could you tell were it is freesing (what was the last thing you saw before the freeze in Debug Console)

Oh, you tried the RTOS thing? Hot off the oven. To enable runtime, it is a bit involved. If you have a HW timer/counter you can spare, run it at a 1us frequency and setup a macro to sample the counter avoiding ISRs. The counter needs to be at least 10x faster than your OS scheduler Tick for it to be meaningful. Of course you have to handle wraparound/overflow but since it is not for production, you can ignore that part. Wonder if the FreeRTOS itself handles the wraparound.

In your `FreeRTOSConfig.h`
```C
/* Modify the following existing macro */
#define configGENERATE_RUN_TIME_STATS             1 /* 1: generate runtime statistics; 0: no runtime statistics */

/* Define these two macros or you will get helpful errors asking you to do it */
#define portCONFIGURE_TIMER_FOR_RUN_TIME_STATS() /* Define this to initialize your timer */
#define portGET_RUN_TIME_COUNTER_VALUE() /* Define this to sample the counter */
```

ParalaX002, 2022-03-23:
Thanks for the tip. I'll check it out sometimes :)

Regarding the freeze, it occurs here:
```
-> 1^done
Suppressing output for '2-gdb-version'
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Suppressing output for '5-symbol-info-variables'
Finished reading symbols from nm: Time: 45 ms

```

and then on `Waiting for GDB connection`

haneefdm, 2022-03-23:
We are waiting for (and this did not change in eons) a string that looks like `Waiting for GDB connection...` to know that JLink started.

What I see you said was `Waiting for GDB connection` without the three trailing dots. Is that right?

If that is the case, then you can override the like we are looking for in launch.json
```JSON
    "overrideGDBServerStartedRegex": "Waiting for GDB connection"
```
and see how that works. The dots `...` seemed to have been buffered up by the c-library in JLink SW or in NodeJS on VSCode side.

ParalaX002, 2022-03-24:
Unfortunately the dots appears it's just me that didn't copied them apparently, sorry for the false hope :s I also tried directly copy pasting the correct line in the Regex, but it doesn't work either.

haneefdm, 2022-03-24:
May I have the full output of the Debug Console when it appears to hang 30-40 seconds? Wondering if `objdump` is taking too long or it is something else. I see that `nm` returned pretty fast.

ParalaX002, 2022-03-24:
Here is the output.

As for info variables, when I enter the command after everything is successfully initialized (in the debug console of VSCode), it seems to be pretty fast, about 100ms if I read the timestamps correctly.

```
0000000007: Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf
0000000010: Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf
0000000017: Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf
0000000018: 1-gdb-set target-async on
0000000021: Launching gdb-server: /usr/local/bin/JLinkGDBServerCL -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MKW41Z512xxx4 -select usb=821008201 -rtos GDBServer/RTOSPlugin_FreeRTOS.dylib
0000000021:     Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/JLinkGDBServerCL
0000000036: Finished reading symbols from objdump: Time: 29 ms
0000000043: -> =thread-group-added,id="i1"
0000000043: -> ~"Reading symbols from /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf...\n"
0000000043: Reading symbols from /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf...
0000000054: -> 1^done
0000000054: Suppressing output for '2-gdb-version'
0000000055: 3-interpreter-exec console "set print demangle on"
0000000055: 4-interpreter-exec console "set print asm-demangle on"
0000000055: -> 3^done
0000000055: -> =cmd-param-changed,param="print asm-demangle",value="on"
0000000055: -> 4^done
0000000055: Suppressing output for '5-symbol-info-variables'
0000000079: Finished reading symbols from nm: Time: 69 ms
0000036076: 6-interpreter-exec console "source /Users/mremacle/.vscode/extensions/marus25.cortex-debug-1.4.1/support/gdbsupport.init"
0000036076: 7-interpreter-exec console "source /Users/mremacle/.vscode/extensions/marus25.cortex-debug-1.4.1/support/gdb-swo.init"
0000036076: 8-target-select extended-remote localhost:50000
0000036076: 9-interpreter-exec console "monitor halt"
0000036077: 10-interpreter-exec console "monitor reset"
0000036077: 11-target-download
0000036077: 12-interpreter-exec console "monitor reset"
0000036077: 13-enable-pretty-printing
0000036089: -> 6^done
0000036091: -> =cmd-param-changed,param="language",value="c"
0000036095: -> =cmd-param-changed,param="language",value="auto"
0000036097: -> 7^done
0000036104: -> =thread-group-started,id="i1",pid="42000"
0000036104: -> =thread-created,id="1",group-id="i1"
0000036109: -> ~"0x000038ba in HardFault_Handler () at /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Parameters/ISRManagerC.c:12\n"
0000036109: 0x000038ba in HardFault_Handler () at /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Parameters/ISRManagerC.c:12
0000036109: -> ~"12\t  __asm volatile (\n"
0000036109: 12	  __asm volatile (
0000036109: -> *stopped,frame={addr="0x000038ba",func="HardFault_Handler",args=[],file="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Parameters/ISRManagerC.c",fullname="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Parameters/ISRManagerC.c",line="12",arch="armv6s-m"},thread-id="1",stopped-threads="all"
0000036109: mi2.status = stopped
0000036110: Program stopped, probably due to a reset and/or halt issued by debugger
0000036112: -> 8^connected
0000036112: -> 9^done
0000036182: -> @"Resetting target\r\n"
0000036182: Resetting target
0000036182: -> 10^done
0000036183: -> 11+download,{section=".interrupts",section-size="192",total-size="2190939"}
0000036183: -> 11+download,{section=".interrupts",section-sent="192",section-size="192",total-sent="192",total-size="2190939"}
0000036183: -> 11+download,{section=".flash_config",section-size="16",total-size="2190939"}
0000036183: -> 11+download,{section=".text",section-size="82404",total-size="2190939"}
0000036185: -> 11+download,{section=".ARM.extab",section-size="728",total-size="2190939"}
0000036185: -> 11+download,{section=".ARM",section-size="2504",total-size="2190939"}
0000036186: -> 11+download,{section=".init_array",section-size="8",total-size="2190939"}
0000036186: -> 11+download,{section=".fini_array",section-size="4",total-size="2190939"}
0000036186: -> 11+download,{section=".data",section-size="244",total-size="2190939"}
0000036403: -> 11^done,address="0x000004d8",load-size="86100",transfer-rate="229600000",write-rate="6623"
0000036473: -> @"Resetting target\r\n"
0000036473: Resetting target
0000036473: -> 12^done
0000036473: -> 13^done
0000036503: 14-break-insert "/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Utils/CLI.cpp:139"
0000037193: -> 14^done,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x0000877e",func="CLI_cmd_GPIOr(char*, unsigned int, char const*)",file="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Utils/CLI.cpp",fullname="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Utils/CLI.cpp",line="140",thread-groups=["i1"],times="0",original-location="/Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/Utils/CLI.cpp:139"}
0000037263: 15-thread-list-ids
0000037263: -> =thread-created,id="2",group-id="i1"
0000037263: -> ~"[New Remote target]\n"
[... cut ...] 0000b801002000000000000000000000000003f600000000000000000000b8500020398e000060000020b8500020120000001b00000030000000578f000001000000f0500020d00000202f000000f8310020000000000000000001000000a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5000000009c3100200000001004ed00e0e850002003f90000d00901000000000100000010ffffffff0000000015140000d0090100245f002030e5ff1fb0e5ff1f685f00207358000000000000a5a5a5a5a5a5a5a5685f0020a5a5a5a5170302a5a5a5a5a5a5a5a5a5245f0020a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5f35c0000a5a5a5a549090100"}]

```

haneefdm, 2022-03-24:
Any chance you can share your executable?

It appears that reading symbol information from GDB is what is taking time. We have not found a reliable way of figuring out which file has what variables defined and gdb has that info. Even with thousands of symbols, it should not take that long. All we do is the equivalent of `info variables`. 36 seconds is an eternity.

I created new release, that should be coming your way shortly. MS takes an hour or two to push the release out to users. May take up to a day. But you can manually force it from the Extension Panel.

In this release, I did two things related to this.
* Start reading the symbols a little early on
* Print additional debug messages to see which aspect is causing the slowdown

We could running it using the new release. I don't think it will change anything but at least, it will tell us what is causing the issue.
Note that while we start the gdb-server, we do not connect to a gdb-server (JLink in your case) until everything else is ready.

ParalaX002, 2022-03-24:
No problem to send you the file, I'll do it by mail :)

In the meantime here is the latest log:

```
0000000009: Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf
0000000011: Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf
0000000012: Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf
0000000013: 1-gdb-set target-async on
0000000016: Launching gdb-server: /usr/local/bin/JLinkGDBServerCL -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MKW41Z512xxx4 -select usb=821008201 -rtos GDBServer/RTOSPlugin_FreeRTOS.dylib
0000000016:     Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/JLinkGDBServerCL
0000000034: Finished reading symbols from objdump: Time: 25 ms
0000000041: -> =thread-group-added,id="i1"
0000000041: -> ~"Reading symbols from /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf...\n"
0000000041: Reading symbols from /Users/mremacle/Documents/ParalaX/Projects/Sources/TinyIoT/Embedded/KW41/TinyIoT.elf...
0000000050: -> 1^done
0000000050: 2-gdb-version
0000000051: -> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.07) 10.2.90.20210621-git\n"
0000000051: -> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
0000000051: -> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
0000000051: -> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
0000000051: -> ~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\n"
0000000051: -> ~"Type \"show configuration\" for configuration details.\n"
0000000051: -> ~"For bug reporting instructions, please see:\n"
0000000051: -> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
0000000051: -> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
0000000051: -> ~"\n\n"
0000000051: -> ~"For help, type \"help\".\n"
0000000051: -> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
0000000051: -> 2^done
0000000051: 3-interpreter-exec console "set print demangle on"
0000000051: 4-interpreter-exec console "set print asm-demangle on"
0000000053: -> 3^done
0000000053: -> =cmd-param-changed,param="print asm-demangle",value="on"
0000000053: -> 4^done
0000000053: Suppressing output for '5-symbol-info-variables'
0000000053: 5-symbol-info-variables
0000000058: Finished reading symbols from nm: Time: 48 ms
0000000230: Debug Time: GDB Ready...
0000036267: Debug Time: GDB info variables done...
0000036768: Debug Time: GDB Server post start events done...
0000036768: Debug Time: objdump and nm done...
0000036768: Debug Time: All pending items done, proceed to gdb connect...
0000036769: 6-interpreter-exec console "source /Users/mremacle/.vscode/extensions/marus25.cortex-debug-1.4.3/support/gdbsupport.init"
0000036769: 7-interpreter-exec console "source /Users/mremacle/.vscode/extensions/marus25.cortex-debug-1.4.3/support/gdb-swo.init"
0000036769: 8-target-select extended-remote localhost:50000
0000036769: 9-interpreter-exec console "monitor halt"
0000036769: 10-interpreter-exec console "monitor reset"
0000036769: 11-target-download
0000036769: 12-interpreter-exec console "monitor reset"
0000036769: 13-enable-pretty-printing
0000036779: -> 6^done
0000036781: -> =cmd-param-changed,param="language",value="c"
0000036785: -> =cmd-param-changed,param="language",value="auto"
0000036787: -> 7^done
0000036797: -> =thread-group-started,id="i1",pid="42000"
0000036797: -> =thread-created,id="1",group-id="i1"
0000036804: -> ~"World::itemGet (this=0x20005d9c, item=item@entry=WorldItem::INTERNAL_TIME) at /Users/mremacle/Documents/ParalaX/Projects/Common/Base/World.cpp:373\n"
0000036804: World::itemGet (this=0x20005d9c, item=item@entry=WorldItem::INTERNAL_TIME) at /Users/mremacle/Documents/ParalaX/Projects/Common/Base/World.cpp:373
0000036804: -> ~"373\t    if(m_storage[i]->getType() == item)\n"
0000036804: 373	    if(m_storage[i]->getType() == item)
0000036804: -> *stopped,frame={addr="0x00009750",func="World::itemGet",args=[{name="this",value="0x20005d9c"},{name="item",value="WorldItem::INTERNAL_TIME"},{name="item@entry",value="WorldItem::INTERNAL_TIME"}],file="/Users/mremacle/Documents/ParalaX/Projects/Common/Base/World.cpp",fullname="/Users/mremacle/Documents/ParalaX/Projects/Common/Base/World.cpp",line="373",arch="armv6s-m"},thread-id="1",stopped-threads="all"
0000036804: mi2.status = stopped
0000036804: Program stopped, probably due to a reset and/or halt issued by debugger
0000036807: -> 8^connected
0000036807: -> 9^done
0000036875: -> @"Resetting target\r\n"
0000036875: Resetting target
0000036875: -> 10^done
0000036876: -> 11+download,{section=".interrupts",section-size="192",total-size="2190804"}
[... 245 lines cut ...]
0000039864: 128-var-create  hover_d9b2577d20680e9cc202dba094b8786f8601d7e78fa193ebdf8851b4b97b47f0 @ "(char *)((TCB_t*)0x20004018).pcTaskName"
0000039865: -> 128^done,name="hover_d9b2577d20680e9cc202dba094b8786f8601d7e78fa193ebdf8851b4b97b47f0",numchild="1",value="0x2000404c <ucHeap+21096> \"Tmr Svc\"",type="char *",has_more="0"
[... cut ...] 00000b8010020000000000000000000000000fff500000600000000000000b8500020358e000060000020b85000201e000000bd00000030000000538f000001000000f0500020dc0000202f000000f8310020000000000000000001000000a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5000000009c3100200000001004ed00e0e8500020fff80000d00901000000000100000010ffffffff00000000981f0100d0090100245f002030e5ff1fb0e5ff1f685f00207358000000000000a5a5a5a5a5a5a5a5685f0020a5a5a5a5170302a5a5a5a5a5a5a5a5a5245f0020a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5a5f35c0000a5a5a5a545090100"}]
```

haneefdm, 2022-03-24:
I wish more people were like you to share the elf file. Thanks.

Some data. `info variables` ran pretty fast for you in the Debug Console because gdb cached the results. But starting from scratch, a very different story. Here is what I saw with your elf file
```
hdm@HaneefMac-2021 Archive % time (echo "info variables" | arm-none-eabi-gdb TinyIoT.elf > vars.txt)
( echo "info variables" | arm-none-eabi-gdb TinyIoT.elf > vars.txt; )  36.70s user 1.08s system 99% cpu 37.809 total
hdm@HaneefMac-2021 Archive % ls -l vars.txt TinyIoT.elf 
-rwxr-xr-x@ 1 hdm  staff  2364972 Mar 24 09:19 TinyIoT.elf
-rw-r--r--  1 hdm  staff    17709 Mar 24 14:04 vars.txt
hdm@HaneefMac-2021 Archive % 

hdm@HaneefMac-2021 Archive % arm-none-eabi-gdb --version
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
...
```
Mine is a pretty fast Intel Mac. It takes forever for gdb to gather the info to respond and, running it interactively, there is nothing for quite a long time and then it trickles. Never seen anything like that. 37 seconds is obscene, and after all that, just 18KB of output. I am thinking gdb is doing a `call a friend` :-)

Not many C++ programs that I run across though, so I am not surprised. Wondering if we should file an issue with the gdb folks. The elf file is just 2.4MB so what's happening.

Still trying to find a solution...thanks for your patience.

ParalaX002, 2022-03-25:
Hi,

If it can help, I have the arm-gcc version 10.3 on my Mac. I can’t upgrade to version 11.2 because there is a bug in the compiler that doesn’t work any more on Mac M1, however, I tested gdb, and the time was the same.

Also, on version 1.2.2 of your extension, I don’t have the problem, so there surely is something there :) So what I do is that I use 1.4.x to help with your debug, than revert to 1.2.2 to work, and it works everytime. Maybe there is a call that is a bit different or something?

Regards,

Matthieu

haneefdm, 2022-03-25:
>
> Also, on version 1.2.2 of your extension, I don’t have the problem, so
> there surely is something there :) So what I do is that I use 1.4.x to help
> with your debug, than revert to 1.2.2 to work, and it works everytime.
> Maybe there is a call that is a bit different or something?

That is because in 1.2.2, the static variables were not showing up as
expected in the Variables Window. So, we started using gdb to get that
info. We used to use GDB a long time ago as well, but we never saw this
problem. We rely 1000% on GDB doing the right thing, but find bugs in it
once in a while and all we can do is find a workaround. I will have some
sort of solution over the weekend.

haneefdm, 2022-03-28:
Please try version 1.4.4. If it is still not addressed, I can reopen this issue.

closed 2022-03-28 by haneefdm

Felix1165, 2022-03-28:
Just tried the new release and now the debugging with a JLink Debugger seams to work again.
I will try it over the next days.

ParalaX002, 2022-03-29:
Hello, I confirm that it is now working. Thanks a lot again!

ehntoo, 2022-04-01:
I think this may be the same GDB bug I ran into a while back - the bug is fixed with binaries produced by newer builds of binutils/GCC (binutils >=2.37), and may also be fixed by building binaries with `-g2` instead of `-g3`.
GDB issue that still remains open:
https://www.cygwin.com/bugzilla/show_bug.cgi?id=27754
Binutils bug:
https://sourceware.org/bugzilla/show_bug.cgi?id=27590

haneefdm, 2022-04-01:
Thank you for letting me know @ehntoo