Marus/cortex-debug #619: Unable to debug with v1.4.0 or v1.4.1 and GNU ARM Embedded Toolchain v10.3 -2021.10

goldwake, 2022-03-23:
I am unable to debug using the GNU ARM Embedded Toolchain v10.3-2021.10 and v1.4.0 or 1.4.1 of the cortex-debug extension.

The error i get printed is 
`/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-260_20210727_1627371386/src/gdb/gdb/utils.c:671: internal-error: virtual memory exhausted: can't allocate 4064 bytes.`

If I try and use STM32CubeProgrammer to program the hex generated by VS Code, it warns me that the Core is locked up after programming.

If i revert to use v10.2 or v9.3.1 of the toolchain and v1.4.1 of the extension, everything works as expected

Rolling back the extension version to v1.2.2 and using v10.3 -2021.10 of the tool chain works as expected

haneefdm, 2022-03-24:
So, gdb crashed with our extension versions 1.4.x. It is not us but we need to figure out which command(s) triggered the crash. Please enable debug (set `"showDevDebugOutput": "raw"` in launch.json)and include the contents of the Debug Console in the issue.

Sounds very much like a bug in gdb and they will most likely want to hear about it.

goldwake, 2022-03-24:
here's the Debug console log with `"showDevDebugOutput": "raw" in launch.json `. I gathered this using v1.4.3 of the plugin

```Reading symbols from C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-objdump.exe --syms -C -h -w C:\ORG\Source\my-project/output/build/cmake/source/myOutput.elf
Reading symbols from c:/program files (x86)\gnu arm embedded toolchain\10 2021.10\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:\ORG\Source\my-project/output/build/cmake/source/myOutput.elf
Launching GDB: "C:\\Program Files (x86)\\GNU Arm Embedded Toolchain\\10 2021.10\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "C:\\ORG\\Source\\my-project/output/build/cmake/source/myOutput.elf"
1-gdb-set target-async on
Launching gdb-server: openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\ORG\\Source\\my-project" -f "c:/Users/goldwake/.vscode/extensions/marus25.cortex-debug-1.4.3/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32l4x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd.exe
Finished reading symbols from objdump: Time: 203 ms
Finished reading symbols from nm: Time: 195 ms
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
-> =thread-group-added,id="i1"
-> ~"Reading symbols from C:\\ORG\\Source\\my-project/output/build/cmake/source/myOutput.elf...\n"
Reading symbols from C:\ORG\Source\my-project/output/build/cmake/source/myOutput.elf...
-> 1^done
2-gdb-version
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=i686-w64-mingw32 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Suppressing output for '5-symbol-info-variables'
5-symbol-info-variables
/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gdb/gdb/utils.c:671: internal-error: virtual memory exhausted: can't allocate 4064 bytes.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

Quit this debugging session?
(y or n) [answered Y; input not from terminal]

This is a bug, please report it.
  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>.
/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gdb/gdb/utils.c:671: internal-error: virtual memory exhausted: can't allocate 4064 bytes.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

Create a core file of GDB?
(y or n) [answered Y; input not from terminal]
OpenOCD GDB Server Quit Unexpectedly. See gdb-server output for more details.

```

and the gdb-server output
```[2022-03-24T22:50:54.621Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
    openocd.exe -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\ORG\\Source\\my-project" -f "c:/Users/goldwake/.vscode/extensions/marus25.cortex-debug-1.4.3/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32l4x.cfg
    Open On-Chip Debugger 0.10.0 (2020-12-21) [https://github.com/sysprogs/openocd]
    Licensed under GNU GPL v2
    libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
    For bug reports, read
            http://openocd.org/doc/doxygen/bugs.html
    CDRTOSConfigure
    Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
    Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
    Info : Listening on port 50001 for tcl connections
    Info : Listening on port 50002 for telnet connections
    Info : clock speed 500 kHz
    Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
    Info : Target voltage: 3.124111
    Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
    Info : starting gdb server for stm32l4x.cpu on 50000
    Info : Listening on port 50000 for gdb connections
    [2022-03-24T22:51:43.146Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
    GDB server session ended. This terminal will be reused, waiting for next session to start...```

haneefdm, 2022-03-24:
Thanks for getting back.

Yup, looks like newer versions of gdb is broken. This is the 2nd report I see where the equivalent of `info variables` fails spectacularly. The other one was a C++ program.

See https://github.com/Marus/cortex-debug/issues/618#issuecomment-1078392055

Not sure what to do because we rely on that for figuring out where static symbols are defined. We have to find another way. Quickly.

VitPilot, 2022-03-25:
Thanks for posting this -  I have the same problem on an STM32 board, VSCode based project , using a JLink or ST-Link 
 for debugging... 

Got a similar error message 

Finished reading symbols from objdump: Time: 122 ms
-> 1^done
Suppressing output for '2-gdb-version'
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
-> 3^done
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Suppressing output for '5-symbol-info-variables'
Finished reading symbols from nm: Time: 565 ms
/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gdb/gdb/utils.c:671: internal-error: virtual memory exhausted: can't allocate 4064 bytes.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

A roll back to 9.x seems to work, but is a little unstable.

haneefdm, 2022-03-25:
We have resolved this as a bug in gdb. A thousand percent sure now. I am trying to work around it as a fix from gdb is unlikely, and going back for gdb versions is not nice (other bugs). I will be making a new release of Cortex-Debug shortly. Perhaps over the weekend, Monday at the latest.

CIPop, 2022-03-27:
+1 having the same issue
Thanks for looking for a workaround @haneefdm !

goldwake, 2022-03-28:
I am able to debug again using v1.4.4, thank you. 

I can't use the new RTOS panel, I'm guessing this is expected?

haneefdm, 2022-03-28:
No, RTOS panel was untouched and it does work for me. Remember, it is only FreeRTOS and you may have to refresh the panel the first time after starting your program and exposing the RTOS panel.

If there is an issue, lets do a separate issue with details. I am closing this issue as it is probably addressed. Please report if the original issue still persists

closed 2022-03-28 by haneefdm

VitPilot, 2022-03-29:
So,  I can now get debugging going with 1.4.4 and GNU Arm Embedded Toolchain - 10 2021.10   over JLINK , however when you run the target and then press the pause button on the debugger panel,  the GDB crashes with the same virtual memory error 

_/mnt/workspace/workspace/GCC-10-pipeline/jenkins-GCC-10-pipeline-338_20211018_1634516203/src/gdb/gdb/utils.c:671: internal-error: virtual memory exhausted: can't allocate 4064 bytes.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

Quit this debugging session?_

Any ideas?

haneefdm, 2022-03-29:
Looks like a bug in gdb ... it is running out of memory. You should report this to them.

smohekey, 2022-04-04:
Using -g2 when compiling solves this for me.