Marus/cortex-debug #622: ST Link not working since update 1.4.0

WillieVisagie, 2022-03-28:
I'm using
- Windows 10 (latest)
- GNU Arm embedded toolchain Ver 10.3-2021.10
- STM32CubeProgrammer Ver 2.10.0
- STM32CubeIDE Ver 1.9.0
- ST Link V3Set (Firmware upgraded to latest via STM32CubeIDE in the single UART mode)
- VSCode (latest)

Since version Cortex-Debug 1.4.0 I could not have stepped through the code.  It seems like the server does not respond.

When I rolled back to ver Cortex Debug 1.2.2 everything worked again.
I hope the info I gave is enough to trace the problem.

Kind regards

haneefdm, 2022-03-28:
A bit more. What are the contents of the `Debug Console` and the output of the gdbserver (in the Terminal tab)? We are now at 1.4.4, btw and STLink works for me (but I am on a Mac).

WillieVisagie, 2022-03-28:
Ok - this was Saturday night while still running 1.4.3.  Here are the two
windows.  After upgrading to 1.4.4 it works again.  The terminal stops
right there, nothing further.
So it it reproducible with 1.4.3.

[image: image.png]

[image: image.png]

Willie Visagie
082-561-5201

haneefdm, 2022-03-28:
Thanks, @WillieVisagie for reporting. What was fixed in 1.4.4 were actually workarounds for bugs in gdb and VSCode. It was universal (STLink, JLink, OpenOCD, etc.). We triggered those bugs of course by using known features of gdb and VSCode. Only affected people using specific versions of gdb. VSCode was universal. Hopefully, it is behind us.

Btw, your images did not display at all. Must be a github issue.

closed 2022-03-28 by haneefdm

WillieVisagie, 2022-03-28:
Oh.  I'm replying from GMAIL.  Let me attach them...

Willie Visagie
082-561-5201

WillieVisagie, 2022-03-28:
But I'm glad you solved it!  You guys do a wonderful job really helping
people!!!

Willie Visagie
082-561-5201

WillieVisagie, 2022-03-28:
Retrying...
[image]
[image]

haneefdm, 2022-03-28:
Yup, the gdb crash (or running out of memory) with certain versions gdb is what caused a whole bunch of consternation. It was a perfect storm for me. Three things going wrong all at one time.

Anyways, happy debugging :-)