Marus/cortex-debug #623: Reset oddity

lbthomsen, 2022-03-30:
I have got quite a lot of experience with STM32CubeIde debugging using a wide array of different ST-Link devices and STM32 targets.  In most cases I just hook up GND and the CLK/DIO lines of the SWD interface and it works.  I am however quite fed up with STM32CubeIde and been looking for alternatives.  One such alternative is of course VSCode and cortex debug (both latest version).

Experimenting with a number of different options, I came up with the best working one:

```
        {
            "cwd": "${workspaceFolder}",
            "executable": "./build/bp_blink.elf",
            "name": "Debug with ST-Link",
            "request": "launch",
            "type": "cortex-debug",
            "device": "STM32F411CEU6",
            "showDevDebugOutput": "none",
            "servertype": "stutil", 
            "v1": false
        }
```

The oddity is that unless I connect the NRST signal from the st-link to the target, early breakpoints are ignored.

Debugging the following snippet:

```
int main(void)
{
  /* USER CODE BEGIN 1 */
   //for (uint32_t i = 0; i < 1000000; ++i) asm("NOP"); // Insane
  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();
```

Trying to set a breakpoint at HAL_Init() it gets ignored.  If I add the above for loop throwing a million NOP's in before HAL_Init and the breakpoint works.

If I connect NRST everything works as expected.

The problem is I got a number of targets where the NRST is not connected to a physical pin.

Just to make it clear that in STM32CubeIde it works flawlessly.

lbthomsen, 2022-03-30:
In Stm32CubeIde I usually use the ST-Link (OpenOCD) option (the first option fails if the firmware in the ST-Link is not 100 % up-to-date).

I just checked and it will fire up OpenOCD like this:

```
/home/lth/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.2.0.202202231230/tools/bin/openocd "-f" "pwm_ic Debug.cfg" "-s" "/home/lth/src/blackpill/pwm_ic" "-s" "/home/lth/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.200.202202161333/resources/openocd/st_scripts" "-s" "/home/lth/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mpu.debug.openocd_2.0.200.202202231231/resources/openocd/st_scripts" "-c" "gdb_report_data_abort enable" "-c" "gdb_port 3333" "-c" "tcl_port 6666" "-c" "telnet_port 4444"
```

I tried the following launch.json:

```
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceFolder}",
            "runToEntryPoint": "main",
            "executable": "./build/bp_blink.elf",
            "device": "STM32F411CEU6",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f4x.cfg"
            ], 
            "showDevDebugOutput": "none",
        }
```

But the regular openocd (standard Debian) refuse to start which is why I moved to st-util (which works except the oddity that early breakpoints are ignored).

haneefdm, 2022-03-30:
If you prefer OpenOCD, so be it. But make sure you use what ST distributes and not something random from the Web. There is no such thing as "regular openocd" -- there are so many snapshots, customizations, we can't keep up.

You can use `"serverpath"` in launch.json to specify the exact path of OpenOCD. Or you can setup in User VSCode settings so it can apply to all paths. For openocd, you need to setup the search paths. In the end, you will see the exact command line(s) used in the Debug Console.

You also need to have the armToolChainPath in User settings (see all other relevant settings) for us to find all the arm gcc tools (specifically arm-none-eabi-gdb, arm-none-eabi-objdump and arm-none-eabi-nm). Again, I would use the one distributed by ARM or ST and not something random.

[image]

There are a ton of opeionts that can go into the lanch.json to customize your setup.

https://github.com/Marus/cortex-debug
https://github.com/Marus/cortex-debug/wiki

haneefdm, 2022-03-30:
All of those other pieces of SW do not belong to us. What combination works for you is up to you.

Me, I always use the STLink from ST and rarely OpenOCD, sometimes JLink, but never "stutil". I also periodically make sure that the STLink FW on the Nucleo/Discovery boards is up to date.

And, just as a reminder, all these other tools should work even from the command line. If they don't they will not work with our extension either. We don't do anything special that you cannot do from the command line. This is why we show you **exactly** which command line we use.

closed 2022-03-30 by haneefdm

lbthomsen, 2022-03-31:
I think you were a bit quick to close this one.  Speaking of comment line, st-util is executed as:

```
st-util -p 50000 --no-reset
```

which leaves whatever firmware is running running.  However, if executed like:

```
st-util -p 50000 --connect-under-reset
```

I suspect it would work.  One suggestion would be to make the st-util arguments configurable.

lbthomsen, 2022-03-31:
Because the `--no-reset` appears to be hardcoded:
```
st-util -p 50000 --no-reset --connect-under-reset
```