Marus/cortex-debug #624: Debugging STM32H755ZI-Q Nucleo Problem

marco-polo-github, 2022-03-30:
Somehow the debugging does not start.  ubeProgrammer and also CubeIDE works fine, and also the ST-LINK GDB seems to start. Somehow it is stopping and I can't see the reason. Is there a possibility to get more information in the debug console?

**DEBUG_CONSOLE output**
Reading symbols from /home/stein/.conan/data/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-objdump --syms -C -h -w /home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755
Reading symbols from /home/stein/.conan/data/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755
Launching GDB: /home/stein/.conan/data/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb -q --interpreter=mi2  /home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755
1-gdb-set target-async on
Setting GDB-SErver CWD: /opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.200.202202231230/tools/bin
Launching gdb-server: /opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.200.202202231230/tools/bin/ST-LINK_gdbserver -p 50000 -cp /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin --swd --halt "-l 64" -v
    Please check TERMINAL tab (gdb-server) for output from /opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.200.202202231230/tools/bin/ST-LINK_gdbserver
-> =thread-group-added,id="i1"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
-> &": No such file or directory.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":": No such file or directory.\n"}]}
: No such file or directory.
-> &"\"/home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755\" is not a core dump: file format not recognized\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"log","content":"\"/home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755\" is not a core dump: file format not recognized\n"}]}
"/home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755" is not a core dump: file format not recognized
-> 1^done
GDB -> App: {"output":"","token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
2-gdb-version
Finished reading symbols from objdump: Time: 126 ms
-> ~"GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git\n"}]}
-> ~"Copyright (C) 2019 Free Software Foundation, Inc.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Copyright (C) 2019 Free Software Foundation, Inc.\n"}]}
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."}]}
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nType \"show copying\" and \"show warranty\" for details.\n"}]}
-> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"}]}
-> ~"Type \"show configuration\" for configuration details.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"show configuration\" for configuration details.\n"}]}
-> ~"For bug reporting instructions, please see:\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For bug reporting instructions, please see:\n"}]}
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"<http://www.gnu.org/software/gdb/bugs/>.\n"}]}
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."}]}
-> ~"\n\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\n\n"}]}
-> ~"For help, type \"help\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For help, type \"help\".\n"}]}
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"apropos word\" to search for commands related to \"word\".\n"}]}
-> 2^done
GDB -> App: {"output":"","token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
WARNING: Cortex-Debug will deprecate use of GDB version 8 after July 2022. Please upgrade to version 9+
3-interpreter-exec console "set print demangle on"
-> 3^done
GDB -> App: {"output":"","token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","print asm-demangle"],["value","on"]]}]}
-> 4^done
GDB -> App: {"output":"","token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
Finished reading symbols from nm: Time: 329 ms
5-interpreter-exec console "source /home/stein/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdbsupport.init"
-> 5^done
GDB -> App: {"output":"","token":5,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
6-interpreter-exec console "source /home/stein/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","language"],["value","c"]]}]}
-> =cmd-param-changed,param="language",value="auto"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","language"],["value","auto"]]}]}
-> 6^done
GDB -> App: {"output":"","token":6,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
7-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
[... 27 lines cut ...]
GDB -> App: {"output":"","token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","No executable file specified.\nUse the \"file\" or \"exec-file\" command."]]}}
10-interpreter-exec console "monitor reset"
Failed to launch GDB: No executable file specified.
Use the "file" or "exec-file" command. (from target-download)

**gdb-server output:** 
/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.200.202202231230/tools/bin/ST-LINK_gdbserver -p 50000 -cp /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin --swd --halt "-l 64" -v

STMicroelectronics ST-LINK GDB server. Version 6.1.0
Copyright (c) 2022, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 64
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Enabled
        SWD Debug                  : Enabled

Waiting for debugger connection...
Debugger connected
Waiting for debugger connection...
Shutting down...
[2022-03-30T15:45:09.334Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

haneefdm, 2022-03-30:
Looks like you have nothing to debug.  GDB just quit

```
Launching GDB: /home/stein/.conan/data/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/stein/vide-mounts/projects/fmp-io1-at-stm32h755/build/bin/run_at_stm32h755
.....
9-target-download
-> 9^error,msg="No executable file specified.\nUse the "file" or "exec-file" command."
GDB -> App: {"output":"","token":9,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","No executable file specified.\nUse the "file" or "exec-file" command."]]}}
```

In your launch.json, what did you use for an `"executable"`. It must be a valid elf file (this was passed on the gdb command line). If this does not exist, then there is nothing to debug

marco-polo-github, 2022-03-31:
In the launch.json I have set it:  
`"executable": "./build/bin/run_at_stm32h755",`
and the _file_ command says this: 
`run_at_stm32h755: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped`

I have also set an absolute path to the executable, and renamed the file to run_at_stm32h755.elf which did not help.

haneefdm, 2022-03-31:
I can't tell what is wrong. You can run gdb manually on the command line and see what is wrong. As you can see, we are only reporting what gdb reports. We are not the debugger. We are only the front end to it

haneefdm, 2022-03-31:
Could you attach your launch.json? Also, make sure you are using the same version of GDB (GNU ARM Tools) as you are using with STM32 IDE.

If you are on Linux, make sure you can run gdb from the command line without crashing.

If we can run everything using command-line tools, we can see where things are going wrong. I know it is painful but in cases like this, it can't be avoided ... especially since I cannot duplicate your environment

haneefdm, 2022-04-04:
Yes, things look normal when you ran it manually. Silly as it sounds could we try the following.

I think you probably already did this but ... set executable to full path-name this way
```
      "executable": "${workspaceFolder}/build/bin/run_at_stm32h755",
```
Another silly thing to try is to rename the executable. Let us add an extension to your executable and call it `run_at_stm32h755.elf` and try like this.

```
      "executable": "${workspaceFolder}/build/bin/run_at_stm32h755.elf",
```

Another way is to not change anything else but do the following
```
"loadFiles": [
    "${workspaceFolder}/build/bin/run_at_stm32h755"
],
```

Please let me know if any (or none) of the ways work. I have never seen this before. There is nothing wrong with STLink ... it is gdb misbehaving.

haneefdm, 2022-04-04:
Just in case, I don't think your server args are correct. It should be
```
"serverArgs": ["l", "64", "-v"],
```
The -l and 64 should be two separate arguments but none of that has anything to do with why gdb is not working.

marco-polo-github, 2022-04-05:
ok, `"serverArgs": ["-l", "64", "-v"],` I have changed. and this works fine starting it manually. 

Changing the name of the binary does not help:
run_at_stm32h755
run_at_stm32h755.elf
output
output.elf

also in my home path:
```
      "loadFiles": [
        "${env:HOME}/output.elf"
      ],
```
but that does not work. Here Visual Studio Code outputs: Unable to find executable file at undefined.
The vscode DEBUG CONSOLE is empty.

marco-polo-github, 2022-04-05:
ok with the following debugging seems to start.
```
...
     "executable": "${env:HOME}/output.elf",
      "loadFiles": [
        "${env:HOME}/output.elf"
      ],
...
```
The Disassembly is shown but the symbols are missing, so` "runToEntryPoint": "main",` will also not work.

haneefdm, 2022-04-05:
Yes, loadFiles still requires an executable. A session won't even start.

It appears you compiled your program with zero debug info. Even when using optimizations you must at least use `-g` to have some sort of an experience.