Marus/cortex-debug #625: Can not find how to remove default OpenOCD arguments

EvgeniyVasyliev, 2022-03-31:
Hello, everyone!

I am facing a problem with setting the OpenOCD parameters to make debugging when working with STM32H7 custom board. I have set the launch.json the following way:

```
{
    "version": "0.2.0",
    "configurations": [        
        {
            "name": "openocd",
            "request": "launch",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "servertype": "openocd",
            "armToolchainPath": "C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.0.202111181127/tools/bin",            "serverpath":"C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.win32_2.2.0.202202231230/tools/bin/openocd.exe",
            "serverArgs": [
                "-f", "/Device.cfg",
                "-s", "D:/Device_Firmware",
                "-s", "C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.200.202202161333/resources/openocd/st_scripts",
                "-s", "C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mpu.debug.openocd_2.0.200.202202231231/resources/openocd/st_scripts",
                "-c", "gdb_report_data_abort enable",
                "-c", "gdb_port 3333",
                "-c", "tcl_port 6666",
                "-c", "telnet_port 4444",
                "-c", "st-link serial 52FF72066578555442370767"
            ],
            "executable": "${command:cpptools.activeConfigName}/Device.elf",
            "interface": "swd",
            "device": "STM32H743VI",
            "svdFile": "STM32H743x.svd",
            "configFiles": [
                "/Device.cfg"
            ],
            "preLaunchTask": "Make All",
            "showDevDebugOutput": "both"
        }
    ]
}
```

When it generates the code it makes the following:

Launching gdb-server: "C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.win32_2.2.0.202202231230/tools/bin/openocd.exe" **-c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:\\Device_Firmware" -f "c:/Users/Evgeniy Vasyliev/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl" -f /Device.cfg** -f /Device.cfg -s "D:/Device_Firmware" -s "C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.200.202202161333/resources/openocd/st_scripts" -s "C:/ST/STM32CubeIDE_1.9.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mpu.debug.openocd_2.0.200.202202231231/resources/openocd/st_scripts" -c "gdb_report_data_abort enable" -c "gdb_port 3333" -c "tcl_port 6666" -c "telnet_port 4444" -c "st-link serial 52FF72066578555442370767"

Here the arguments marked in bold are generated automatically and they are incorrect. How can I get rid of them, so that only the arguments set by "serverArgs" property are used? I was struggling by trying various properties described on [https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md](https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md), but could not make any of them work as I need.

Thank you!

haneefdm, 2022-03-31:
Sorry, you cannot set the tcp ports. This extension will determine those. We support multiple probes and cores which all require port conflict resolution and we do that automatically. For OpenOCD, you should look at options for `searchDir`, `configFiles`, etc.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

Also peruse our Wiki https://github.com/Marus/cortex-debug/wiki/OpenOCD-Specific-Configuration

closed 2022-03-31 by haneefdm

EvgeniyVasyliev, 2022-04-16:
Thank you, @haneefdm for your answer, but still this seems to be a bug. 
I can not override the values generated automatically! Please watch my question again...

The values
**_-c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002"_**
are generated automatically by the cortex-debug extension. Why? I do not need them!!!

I need to use 
**_-c "gdb_port 3333" -c "tcl_port 6666" -c "telnet_port 4444"_**
I am putting them to "serverArgs" attribute.

So, is there any possible way to remove those values generated automatically? I do not need them! They are bringing the effect that I can not use cortex-debug extension...

haneefdm, 2022-04-16:
Like I said it is by design. There are more tcp ports than what you see there (SWO, RTT) and all are managed by us. So, it is NOT a bug and we insist on that. 

You can always start the gdb-server from the command line and use the servertype "external" and use gdbTarget to specify gdb port. You may also have to specify other customizations. Now you have control of all the TCP ports.