Marus/cortex-debug #627: VSCode 1.66.0 appears to have broken Native Debug (at least for darwin arm64)

zfields, 2022-04-01:
The VSCode team made multiple updates to debugging, and it appears one is incompatible with Native Debug.

Changes:
- https://code.visualstudio.com/updates/v1_66#_debugger-extension-authoring
- https://code.visualstudio.com/updates/v1_66#_debug-adapter-protocol

Experience: No errors are thrown on any of the VSCode windows, and the debugger appears to work momentarily, but it immediately detaches before hitting any breakpoints.

When I revert to VSCode 1.65.2, then everything works as expected.

PhilippHaefele, 2022-04-01:
Just to mention it here, because this could also be the reason (depending on the used launch.json settings / debug probe): They updated Electron to V17 and Chromium version.

@zfields I think it's worthy to share your launch.json content (so other people can try to reproduce this)

haneefdm, 2022-04-01:
Aaaarg. Not in a good position to debug. Traveling.

But, please do share details on how I can duplicate any issues you are seeing.

zfields, 2022-04-01:
I am using an ESP32 and the ESP-Prog. Here is a repository I followed to get setup.
https://github.com/makercrew/esp_prog_vscode_debug/

Specifically, here is the `launch.json`
```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ESP-Prog Debug",
            "type": "gdb",
            "request": "launch",
            "target": "./build/esp_prog_vscode_debug.elf",
            "cwd": "${workspaceRoot}",
            "env": {
                "PATH": "${config:esp_toolchain_paths}:${env:PATH}"
            },
            "gdbpath": "xtensa-esp32-elf-gdb",
            "preLaunchTask": "Launch OpenOCD GDB Server",
            "autorun": [
                "target extended-remote :3333",
                "mon reset halt",
                "flushregs",
                "thb app_main",
                "c",
                "monitor [target current] configure -event gdb-detach { shutdown }",
            ],
        }
    ]
}
```

haneefdm, 2022-04-01:
This launch.json does not use our extension. Not even sure which extension handles this. It is unfortunate they used something generic like `"type": "gdb"`. My guess is

https://github.com/espressif/vscode-esp-idf-extension

They may appreciate knowing about this.

zfields, 2022-04-01:
:facepalm: I'm sorry, I use both your extension, "Cortex-Debug" and "Native Debug" (by Web Freak). I got confused, please accept my apologies.

closed 2022-04-01 by zfields