Marus/cortex-debug #628: Disassembling added on start?

hurricanefrog, 2022-04-04:
Hello,

in a new version of Cortex-Debug (can't tell exactly which version, must be one of the 1.4.x series - I should've disabled auto updates of the extensions. Can't connect to GDB server with 1.4.0-1.4.3) the start of the debug session takes a very long time. During this period, I can't do anything.

The RTOS panel is disabled, so it shouldn't be this.

Is there a way to disable this behavior?

Log output is:

```
Debug-43: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00043f4c","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":43}
Debug-43: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00043b68 -e 0x0004409c -- 5     1332 bytes  (SystemInit)
Debug: Gdb command: -data-disassemble -s 0x0004409c -e 0x00044140 -- 5      164 bytes  (func1)
Debug: Gdb command: -data-disassemble -s 0x00044140 -e 0x000442a0 -- 5      352 bytes  (func2)
Debug-43: Elapsed time for Disassembly Request: 37349 ms
```

- Cortex Debug version: 1.4.4
- VScode: 1.65.2
- Platform: Windows
- SEGGER J-Link GDB Server V6.88a Command Line Version
- arm-none-eabi-gdb.exe: GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git

hurricanefrog, 2022-04-04:
To be more specific, this seems to happen after a crash / SIGTRAP of the firmware. The debugger is in the "wrong state", seems to think the program is paused, although it is already running again. The behavior appears after a "Restart" (using the button in the debug toolbar).

haneefdm, 2022-04-04:
May I ask where you got your gdb version from? A lot of recent bug reports recently are revolving around versions of gdb/gcc-toolchain distributed by various sources.

The disassembly stuff is triggered by VSCode if/when the program halts in a location that has no source code. Currently, there is no way to disable that (and if we do, then have to disable disassembly completely). Also, we didn't add this at the start. VSCode did. But lets take a look
```
Debug-43: Elapsed time for Disassembly Request: 37349 ms
```
37 seconds for a 2KB of disassembly? Obscene. That should take less than 100ms. Again, something is going on gdb that is not explainable. The stuff above that like are actual gdb requests we made. Even if it takes 37 seconds, no one is waiting for the results (does block other requests if gdb is busy).

If your program starts normally and continues to main or even halts at the reset handler (that has source), you will never see disassembly.

As an aside, we rely on GDB for every aspect of debugging. If it cannot be relied upon, then we don't have a debugger. The recent number of bugs were mostly attributable to gdb.

About your "Restart" thing. Are you using the "Reset" button or "Restart". The latter adds a rebuild and the former simply does a reset of your CPU with the help of the gdb-server. Can you enable debug tracing and attach the results just after you did a "Restart". Output from both the gdb-server and Debug Console are helpful.

Could I ask you to try an older version of gdb and report back?

https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

haneefdm, 2022-04-04:
```
  Unfortunately, No memory information from gdb (or gdb-server). Will try to manage without
```
This is not fatal but it raises a curious point. Why your elf file has very little or no information. You can get this information by doing `arm-none-eabi-objdump -h <elf-file>`

Also, did you try our version 1.4.4? We had to put workarounds for bad versions of gdb in it. Above has nothing to do with that.

haneefdm, 2022-04-04:
See also #619 and a few other recent issues. They all went to some issue with the gnu toolchain.

haneefdm, 2022-06-17:
With 1.5.1, at least we avoid the initial request for disassembly from VSCode -- which generally happens at reset.

closed 2022-06-17 by haneefdm