Marus/cortex-debug #629: Chip stops running if I disconnect debugger with v1.4.x

goldwake, 2022-04-05:
I have noticed that with v1.4.x of the extension, I can no longer disconnect the debugger (hold Alt and press the stop button) and have the chip keep running with the debugger disconnected. It now just stops running as if I had hit the stop button.

When I revert back to v1.2.2 of the extension, it behaves as expected

EvilEli, 2022-04-06:
I experience the same issue (using v1.4.4)

haneefdm, 2022-04-06:
Which gdb-server? The detach/disconnect requests handled very differently (wrongly) by some gdb-servers. JLink vs STLink vs OpenOCD ...

@EvilEli Thanks for the command trace. I see that we are trying to detach followed by disconnect (which is what we should do) but I see a second disconnect and not sure where that is coming from.

goldwake, 2022-04-07:
my servertype is "openocd"

EvilEli, 2022-04-07:
I checked the output again -> does indeed disconnect twice

I also rolled back to version 1.2.2 which works properly for me as well.
Here is the output from that version:
```
[...]
0000065042: 35-exec-continue --thread 1
0000065054: GDB -> App: {"token":35,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
0000065055: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
0000065055: mi2.status = running
0000083049: 36-break-delete
0000083049: 37-target-detach
0000083050: GDB -> App: {"token":36,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000083051: GDB -> App: {"token":37,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Cannot execute this command while the target is running[.\nUse]() the \"interrupt\" command to stop the target\nand then try again."]]}}
0000083052: 38-target-disconnect
0000083066: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-exited","output":[["id","i1"]]}]}
0000083067: GDB -> App: {"token":38,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000083067: -gdb-exit
0000083081: GDB -> App: {"outOfBandRecord":[],"resultRecords":{"resultClass":"exit","results":[]}}
```

EvilEli, 2022-05-10:
Tested disconnecting from a debug session with version 1.5-pre.
I can confirm that the issue is still present.
The output differs slightly from version 1.4.4:
```
0000027205: 39-exec-continue --thread 1
0000027206: GDB -> App: {"output":"","token":39,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
0000027207: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
0000027207: mi2.status = running
To client: {"seq":0,"type":"event","event":"continued","body":{"threadId":1,"allThreadsContinued":true}}
To client: {"seq":0,"type":"event","event":"custom-continued","body":{"threadID":1,"allThreads":true}}
To client: {"seq":0,"type":"response","request_seq":18,"command":"continue","success":true,"body":{"allThreadsContinued":true}}
From client: disconnect({"restart":false,"terminateDebuggee":false})
0000055840: 40-exec-interrupt
0000055850: GDB -> App: {"output":"","token":40,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000055878: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nProgram"}]}
0000055878: 
Program
0000055878: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":" received signal SIGTRAP, Trace/breakpoint trap.\n"}]}
0000055878:  received signal SIGTRAP, Trace/breakpoint trap.
0000056097: GDB never responded to an interrupt request. Trying to end session anyways
0000056098: 41-break-delete

[...]

0000056156: mi2.status = stopped
0000056156: GDB -> App: {"output":"","token":41,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000056156: 42-target-detach
0000056199: GDB -> App: {"output":"","token":42,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Remote doesn't know how to detach"]]}}
0000056200: 43-target-disconnect
0000056222: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-exited","output":[["id","i1"]]}]}
To client: {"seq":0,"type":"event","event":"thread","body":{"reason":"exited","threadId":1}}
0000056223: GDB -> App: {"output":"","token":43,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000056223: 44-target-disconnect
0000056223: -gdb-exit
0000056237: GDB -> App: {"output":"","token":44,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","You can't do that when your target is `exec'"]]}}
0000056240: GDB -> App: {"output":"","outOfBandRecord":[],"resultRecords":{"resultClass":"exit","results":[]}}
To client: {"seq":0,"type":"response","request_seq":19,"command":"disconnect","success":true}
To client: {"seq":0,"type":"event","event":"terminated"}
```

Just from a superficial comparison I see the presence of `<n>-exec-interrupt` command in the non working versions, wherase it is not present in the output from the working version 1.2.2.

haneefdm, 2022-05-10:
1.2.2 was actually a bug that made it work for you. i.e., not doing an `exec-interrupt` before doing the rest. In 1.2.2 we did not wait for commands to complete...just fired and hoped it worked and killed everything. That made it look like the program kept running -- which it was.

My question is why did `exec-interrupt` fail? At the point you did the disconnect, does the pause button work?

> 0000056097: GDB never responded to an interrupt request. Trying to end session anyways

We HAVE to do an `exec-interrupt` to be able to delete all breakpoints. Some gdb-servers do not delete breakpoints and that could prevent program from running.

Flow is, we request gdb to interrupt the program, it forwards the request to gdb-server, server tries to interrupt on SWD/JTAG. After this, the program is supposed to halt and server is supposed to respond to gdb which in turn responds to us. It appears that the server is not doing its job or the MCU is not responding. We have to look at server debug logs (JLink?) to see what is going on. Why did our request never get a response? It should at least failed....not just 'no response'

goldwake, 2022-05-10:
how can we get the logs needed? I am using openocd and @EvilEli is using JLink

EvilEli, 2022-05-10:
> My question is why did exec-interrupt fail? At the point you did the disconnect, does the pause button work?

I think this message (`GDB never responded to an interrupt request. Trying to end session anyways`) was a hoax.
Tried it again and that particular message is not printed anymore. Not sure what went wrong there but it is not the usual case.
The pause button works as well:
```
From client: pause({"threadId":1})
0002464846: 52-exec-interrupt 
0002464849: GDB -> App: {"output":"","token":52,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
To client: {"seq":0,"type":"response","request_seq":25,"command":"pause","success":true}
0002465063: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nProgram"}]}
0002465063: 
Program
0002465064: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":" received signal SIGTRAP, Trace/breakpoint trap.\n"}]}
0002465064:  received signal SIGTRAP, Trace/breakpoint trap.
[...]
0002465080: mi2.status = stopped
```

The server logs when trying to disconnect in Version 1.5:
```
Debugger requested to halt target...
...Target halted (PC = 0xC009D2DE)
Reading all registers
Read 4 bytes @ address 0xC009D2DE (Data = 0x709EF5A0)
GDB closed TCP/IP connection (Socket 1120)
[2022-05-10T10:44:50.065Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
in Version 1.2.2:
```
GDB closed TCP/IP connection (Socket 1032)
[2022-05-10T10:56:02.834Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
Somehow the `exec-interrupt` causes the target to be halted before the `target-disconnect` happens.
Wouldn't it be possible to do an `exec-continue` after the `exec-interrupt` so we have
`mi2.status = running` when the `target-disconnect` happens?

EvilEli, 2022-05-10:
I was curious too see how Eclipse IDE does the GDB handling (since the disconnect is working properly here).
This is the output:

```
Debugger requested to halt target...
...Target halted (PC = 0xC0122DAA)
Reading all registers
Read 4 bytes @ address 0xC0122DAA (Data = 0x0B94F8D4)
Reading 64 bytes @ address 0x2FFC2E80
Read 4 bytes @ address 0xC0062E96 (Data = 0x0094F8D5)
GDB closed TCP/IP connection (Socket 1136)
Restoring target state and closing J-Link connection...
```
This suggests that cortex-debug version 1.5 is actually doing the same thing except the last line:
```
Restoring target state and closing J-Link connection...
```
Not sure if there actually happens something since the GDB session is supposed to be closed.
Don't know how I could find out what's the magic behind that.
Probably a Wireshark trace?

EvilEli, 2022-05-10:
Being even more curious I tried to do the whole process manually.
I started the GDB Server in one Terminal and GDB itself in a separate one and typed in the sequence of commands to attach GDB, load and start the program by hand.
Interestingly the exact same sequence works when I do it by hand.

1. Start `JLinkGDBServerCL`
2. Start arm-none-eabi-gdb --interpreter=mi2
3. `(gdb) target remote 127.0.0.1:2331`
4. `(gdb) load <path-to-firmware>`
5. `(gdb) monitor halt`
6. `(gdb) monitor reset`
7. `(gdb) set $pc=__init__`  // or whatever the start address of the program is
8. `(gdb) continue`
9. Ctrl-C which causes a pause to happen (otherwise I cannot enter more commands)
10. `(gdb) interrupt`
11. `(gdb) delete`
12. `(gdb) detach` // <- Doesn't work since we never attached (Remote doesn't know how to detach)
13. `(gdb) disconnect` // Target keeps running

Only difference is that I halt the program explicitly and then disconnect.

haneefdm, 2022-05-10:
Can you try this private build I made for you and let me know?

https://github.com/Marus/cortex-debug/releases/tag/v1.5.1-pre1

EvilEli, 2022-05-11:
Hi @haneefdm thanks for the quick implementation.
Unfortunately the build does not fix the issue.
I tried the trick with the pause button again and that seemed to be a hoax as well :(.
The manual approach however always works (tested it like 5 times to be sure).
Is the sequence I provided similar to what the extension is doing?

haneefdm, 2022-05-13:
I thought the `pause` button was working. It that does not, then I have no idea how to fix this. That has to work. You are probably still seeing this

> GDB never responded to an interrupt request. Trying to end session anyways

From the Debug Console, can you type `-exec-interrupt` and see if it pauses the program? One thing about the manual sequence is that you are not in `async` mode. For machine interface (any debugger), we have to be in `async` mode. Also you don't have to attach to be able to use a detach. A launch + detach is perfectly acceptable.

To me, we need to look at a different version of GDB -- move one/two versions back or forward. I just cannot duplicate this at all.

EvilEli, 2022-05-13:
> I thought the pause button was working. It that does not, then I have no idea how to fix this. 
So the pause button itself is working. Only the procedure 1. pause 2. stop does not leave the program running (as it does with the manual sequence)

> From the Debug Console, can you type -exec-interrupt and see if it pauses the program? One thing about the manual sequence is that you are not in async mode. For machine interface (any debugger), we [...]

I type in `interrupt/-exec-interrupt`  (not sure what's the difference) and it halts the code execution.
Also there i can type `disconnect`/`-target-disconnect` (not sure what's the difference either) and the program continues without stopping.

```

interrupt
From client: evaluate({"expression":"interrupt","context":"repl"})
0000038106: 33-interpreter-exec console "interrupt"
0000038119: GDB -> App: {"output":"","token":33,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
To client: {"seq":0,"type":"response","request_seq":23,"command":"evaluate","success":true,"body":{"result":"{\"output\":\"\",\"token\":33,\"outOfBandRecord\":[],\"resultRecords\":{\"resultClass\":\"done\",\"results\":[]}}","variablesReference":0}}
{"output":"","token":33,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000038126: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nProgram"}]}
0000038126: 
Program
0000038126: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":" received signal SIGTRAP, Trace/breakpoint trap.\n"}]}
0000038127:  received signal SIGTRAP, Trace/breakpoint trap.

[...]

-break-delete
From client: evaluate({"expression":"-break-delete","frameId":258,"context":"repl"})
0000047706: 51-stack-select-frame --thread 1 2
0000047717: GDB -> App: {"output":"","token":51,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000047717: 52-break-delete
0000047732: GDB -> App: {"output":"","token":52,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
To client: {"seq":0,"type":"response","request_seq":30,"command":"evaluate","success":true,"body":{"result":"{\"output\":\"\",\"token\":52,\"outOfBandRecord\":[],\"resultRecords\":{\"resultClass\":\"done\",\"results\":[]}}","variablesReference":0}}

[...]

-target-disconnect
From client: evaluate({"expression":"-target-disconnect","frameId":258,"context":"repl"})
0000057466: 59-stack-select-frame --thread 1 2
0000057467: GDB -> App: {"output":"","token":59,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000057467: 60-target-disconnect
0000057482: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-exited","output":[["id","i1"]]}]}
To client: {"seq":0,"type":"event","event":"thread","body":{"reason":"exited","threadId":1}}
0000057483: GDB -> App: {"output":"","token":60,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
To client: {"seq":0,"type":"response","request_seq":34,"command":"evaluate","success":true,"body":{"result":"{\"output\":\"\",\"token\":60,\"outOfBandRecord\":[],\"resultRecords\":{\"resultClass\":\"done\",\"results\":[]}}","variablesReference":0}}
{"output":"","token":60,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000057781: 61-target-disconnect
0000057792: GDB -> App: {"output":"","token":61,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","You can't do that when your target is `exec'"]]}}
0000058293: kill failed for -21660Error: kill ESRCH
To client: {"seq":0,"type":"event","event":"terminated"}
From client: disconnect({"restart":false,"terminateDebuggee":false})
0000058313: 62-break-delete
0000058313: Could not delete all breakpoints. TypeError: i.result is not a function
0000058314: 63-target-detach
To client: {"seq":0,"type":"response","request_seq":35,"command":"disconnect","success":true}
```

What I found out: When I type in `disconnect` instead of `-target-disconnect` I get an additional message in the log:
```
To client: {"seq":0,"type":"event","event":"thread","body":{"reason":"exited","threadId":1}}
0000036384: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Ending remote debugging.\n"}]}
0000036384: Ending remote debugging.
0000036384: GDB -> App: {"output":"","token":60,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```
Despite that both commands lead to a running state after disconnect.

haneefdm, 2022-05-13:
The commands that start with a `-` are machine interface (gdb MI language) commands that have a guaranteed parseable result format but command names are very different and options can be very different. In this case, `-exec-interrupt` and `interrupt` should be the same. We have to 100000% rely on GDB responses. We cant guess. Or every operation we do is impossible to see what happened.

Let us focus on this comment

https://github.com/Marus/cortex-debug/issues/629#issuecomment-1122077385

* I see that between `-exec-interrupt` and the response is 316 ms later, but we did get one
* We wait for 250 ms after `-exec-interrupt` at which point we assume it is not going to work (happens when gdb is buggy or wedged). This is why you saw the `-break-delete` happen earlier
* We then continue on with the rest of the process and eventually kill GDB to finish the session.

What I changed in the custom build is that the wait time was increased from 250 ms to 2000 ms. In the failure case, if you did not see approx two seconds elapse before a break delete is issued, then you are not using the custom build (or my build failed).

When you snip the trace output too quickly, it is possible that I am not seeing async. messages coming from gdb after it actually responded. This can happen.

I am not convinced that you are running the modified extension. Could I see the full trace starting from the point you hit the disconnect button (nothing removed). I normally clear the Debug Console, before I start the experiment and then use the whole trace output.

haneefdm, 2022-05-13:
I also need the `cortex-debug-server-exiting.log` file that I mentioned in this comment

https://github.com/Marus/cortex-debug/issues/629#issuecomment-1122128270

There may be some valuable information there. You can clear/delete the file before you start a debug session to remove old history.

EvilEli, 2022-05-16:
> Could I see the full trace starting from the point you hit the disconnect button (nothing removed). I normally clear the Debug Console, before I start the experiment and then use the whole trace output.

```
From client: disconnect({"restart":false,"terminateDebuggee":true})
0000036001: 38-exec-interrupt
0000036005: GDB -> App: {"output":"","token":38,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000036012: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nProgram"}]}
0000036013: 
Program
0000036013: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":" received signal SIGTRAP, Trace/breakpoint trap.\n"}]}
0000036013:  received signal SIGTRAP, Trace/breakpoint trap.
0000036014: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"0xc021ffaa in memcpy ()\n"}]}
0000036014: 0xc021ffaa in memcpy ()
0000036014: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["reason","signal-received"],["signal-name","SIGTRAP"],["signal-meaning","Trace/breakpoint trap"],["frame",[["addr","0xc021ffaa"],["func","memcpy"],["args",[]],["arch","armv7"]]],["thread-id","1"],["stopped-threads","all"]]}]}
0000036014: mi2.status = stopped
0000036014: 39-break-delete
0000036037: GDB -> App: {"output":"","token":39,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000036037: 40-target-disconnect
0000036052: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-exited","output":[["id","i1"]]}]}
To client: {"seq":0,"type":"event","event":"thread","body":{"reason":"exited","threadId":1}}
0000036052: GDB -> App: {"output":"","token":40,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000036053: -gdb-exit
0000036067: GDB -> App: {"output":"","outOfBandRecord":[],"resultRecords":{"resultClass":"exit","results":[]}}
To client: {"seq":0,"type":"response","request_seq":23,"command":"disconnect","success":true}
To client: {"seq":0,"type":"event","event":"terminated"}
```

> I also need the cortex-debug-server-exiting.log file that I mentioned in this comment

```
[2022-05-16T06:07:17.714Z] ppid=26572 pid=26572 Cortex Debug (no rebuild): ******* Starting new session request type="launch"
[2022-05-16T06:07:53.685Z] ppid=26572 pid=22216 Cortex Debug (no rebuild): Begin disconnectRequest
[2022-05-16T06:07:53.956Z] ppid=26572 pid=22216 GDB: exited
[2022-05-16T06:07:53.958Z] ppid=26572 pid=22216 Cortex Debug (no rebuild): quitEvent: Killing server
[2022-05-16T06:07:53.959Z] ppid=26572 pid=22216 GDBServer(6848): forcing an exit with kill()
[2022-05-16T06:07:53.967Z] ppid=26572 pid=22216 Cortex Debug (no rebuild): disconnectRequest sendResponse 3
[2022-05-16T06:07:53.971Z] ppid=26572 pid=22216 Cortex Debug (no rebuild): quitEvent: sending VSCode TerminatedEvent
[2022-05-16T06:07:53.974Z] ppid=26572 pid=22216 GDBServer(6848): exited code=null signal=SIGTERM
```

goldwake, 2022-05-16:
Here's the output on the terminal after I disconnect the debugger. I have -d3 severArgs set and am using v1.2.2 of the extesion

```
Debug: 6686 19870 gdb_server.c:1055 gdb_connection_closed(): GDB Close, Target: stm32l4x.cpu, state: running, gdb_actual_connections=0
Debug: 6687 19870 target.c:1804 target_call_event_callbacks(): target event 8 (gdb-end) for core stm32l4x.cpu
Debug: 6688 19870 target.c:1804 target_call_event_callbacks(): target event 23 (gdb-detach) for core stm32l4x.cpu
Info : 6689 19871 server.c:576 server_loop(): dropped 'gdb' connection (error -400)
[2022-05-16T21:13:36.277Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

and the corresponding exit log
```
[2022-05-16T21:13:16.166Z] ppid=21100 pid=22748 GDB started ppid=21100 pid=22748
[2022-05-16T21:13:36.222Z] ppid=21100 pid=22748 Begin disconnectRequest
[2022-05-16T21:13:36.238Z] ppid=21100 pid=22748 target-detach hung: target probably running, thats okay, continue to stop()
[2022-05-16T21:13:36.239Z] ppid=21100 pid=22748 target-detach failed: target probably running, thats okay, continue to stop()
[2022-05-16T21:13:36.255Z] ppid=21100 pid=22748 target-stop failed with exception:You can't do that when your target is `exec' (from target-disconnect)
[2022-05-16T21:13:36.273Z] ppid=21100 pid=22748 GDB: exited
[2022-05-16T21:13:36.274Z] ppid=21100 pid=22748 quitEvent: Killing server
[2022-05-16T21:13:36.274Z] ppid=21100 pid=22748 GDBServer: forcing an exit with kill()
[2022-05-16T21:13:36.276Z] ppid=21100 pid=22748 GDBServer: exited null SIGTERM
[2022-05-16T21:13:36.277Z] ppid=21100 pid=22748 disconnectRequest sendResponse 2
[2022-05-16T21:13:36.285Z] ppid=21100 pid=22748 quitEvent: sending VSCode TerminatedEvent

```

and here is the terminal output when using the v1.5.0 pre release version
```
Debug: 6672 27721 gdb_server.c:3292 gdb_input_inner(): received packet: ''
Debug: 6673 27721 hla_target.c:418 adapter_halt(): adapter_halt
Debug: 6674 27723 hla_target.c:602 adapter_read_memory(): adapter_read_memory 0xe000edf8 4 1
Debug: 6675 27723 target.c:2566 target_read_u32(): address: 0xe000edf8, value: 0x00000000
Debug: 6676 27725 armv7m.c:341 armv7m_read_core_reg(): read r0 value 0x00000000
Debug: 6677 27726 armv7m.c:341 armv7m_read_core_reg(): read r1 value 0xa5a5a5a5
Debug: 6678 27726 armv7m.c:341 armv7m_read_core_reg(): read r2 value 0xa5a5a5a5
Debug: 6679 27727 armv7m.c:341 armv7m_read_core_reg(): read r3 value 0x00000000
Debug: 6680 27728 armv7m.c:341 armv7m_read_core_reg(): read r4 value 0xa5a5a5a5
Debug: 6681 27729 armv7m.c:341 armv7m_read_core_reg(): read r5 value 0xa5a5a5a5
Debug: 6682 27730 armv7m.c:341 armv7m_read_core_reg(): read r6 value 0xa5a5a5a5
Debug: 6683 27731 armv7m.c:341 armv7m_read_core_reg(): read r7 value 0x200017c8
Debug: 6684 27732 armv7m.c:341 armv7m_read_core_reg(): read r8 value 0xa5a5a5a5
Debug: 6685 27733 armv7m.c:341 armv7m_read_core_reg(): read r9 value 0xa5a5a5a5
Debug: 6686 27734 armv7m.c:341 armv7m_read_core_reg(): read r10 value 0xa5a5a5a5
Debug: 6687 27735 armv7m.c:341 armv7m_read_core_reg(): read r11 value 0xa5a5a5a5
Debug: 6688 27736 armv7m.c:341 armv7m_read_core_reg(): read r12 value 0xa5a5a5a5
Debug: 6689 27737 armv7m.c:341 armv7m_read_core_reg(): read sp value 0x200017c8
Debug: 6690 27738 armv7m.c:341 armv7m_read_core_reg(): read lr value 0x08022d11
Debug: 6691 27738 armv7m.c:341 armv7m_read_core_reg(): read pc value 0x08022e00
Debug: 6692 27739 armv7m.c:341 armv7m_read_core_reg(): read xPSR value 0x01000000
Debug: 6693 27740 armv7m.c:341 armv7m_read_core_reg(): read msp value 0x20017fd0
Debug: 6694 27742 armv7m.c:341 armv7m_read_core_reg(): read psp value 0x200017c8
Debug: 6695 27742 armv7m.c:341 armv7m_read_core_reg(): read pmsk_bpri_fltmsk_ctrl value 0x02000000
Debug: 6696 27743 armv7m.c:339 armv7m_read_core_reg(): read d0 value 0x000000004023c73e
Debug: 6697 27746 armv7m.c:339 armv7m_read_core_reg(): read d1 value 0x0000000000000000
Debug: 6698 27747 armv7m.c:339 armv7m_read_core_reg(): read d2 value 0x0000000000000000
Debug: 6699 27749 armv7m.c:339 armv7m_read_core_reg(): read d3 value 0x0000000000000000
Debug: 6700 27751 armv7m.c:339 armv7m_read_core_reg(): read d4 value 0x0000000000000000
Debug: 6701 27753 armv7m.c:339 armv7m_read_core_reg(): read d5 value 0x0000000000000000
Debug: 6702 27754 armv7m.c:339 armv7m_read_core_reg(): read d6 value 0x0000000000000000
Debug: 6703 27756 armv7m.c:339 armv7m_read_core_reg(): read d7 value 0xbd1f37803f800000
Debug: 6704 27758 armv7m.c:339 armv7m_read_core_reg(): read d8 value 0x0000000000000000
Debug: 6705 27759 armv7m.c:339 armv7m_read_core_reg(): read d9 value 0x0000000000000000
Debug: 6706 27762 armv7m.c:339 armv7m_read_core_reg(): read d10 value 0x0000000000000000
Debug: 6707 27763 armv7m.c:339 armv7m_read_core_reg(): read d11 value 0x0000000000000000
Debug: 6708 27765 armv7m.c:339 armv7m_read_core_reg(): read d12 value 0x0000000000000000
Debug: 6709 27765 armv7m.c:339 armv7m_read_core_reg(): read d13 value 0x0000000000000000
Debug: 6710 27768 armv7m.c:339 armv7m_read_core_reg(): read d14 value 0x0000000000000000
Debug: 6711 27770 armv7m.c:339 armv7m_read_core_reg(): read d15 value 0x0000000000000000
Debug: 6712 27771 armv7m.c:341 armv7m_read_core_reg(): read fpscr value 0x00000000
[... 13 lines cut ...]
Debug: 6726 27775 target.c:2420 target_read_buffer(): reading buffer of 64 byte at 0x200017c0
[... cut ...]
[2022-05-16T21:21:59.901Z] ppid=25500 pid=2060 GDB: exited
[2022-05-16T21:21:59.902Z] ppid=25500 pid=2060 MyApp: quitEvent: Killing server
[2022-05-16T21:21:59.902Z] ppid=25500 pid=2060 GDBServer(21276): forcing an exit with kill()
[2022-05-16T21:21:59.904Z] ppid=25500 pid=2060 GDBServer(21276): exited code=null signal=SIGTERM
[2022-05-16T21:21:59.905Z] ppid=25500 pid=2060 MyApp: disconnectRequest sendResponse 2
[2022-05-16T21:21:59.919Z] ppid=25500 pid=2060 MyApp: quitEvent: sending VSCode TerminatedEvent
```

EvilEli, 2022-05-17:
Hi @haneefdm,

I just tried to compile and test the extension myself and started playing around a bit.
I found out that adding a short delay between disconnecting and exiting gdb fixes the issue (at least for me).

I made a pull request for it. I'm not sure if this implementation is actually a good solution since I'm by no means a TypeScript developer.
Let me know if it is working for you.

EvilEli, 2022-05-18:
> We HAVE to do an `exec-interrupt` to be able to delete all breakpoints. Some gdb-servers do not delete breakpoints and that could prevent program from running.

I just have a follow up question regarding the `exec-interrupt` because it came up in this issue (probably a separate issue is appropriate for this). Doing this before actually disconnecting does heavily impact the real time behavior of the running application. I think there might be use cases (at least I have some) where we want to disconnect GDB without impacting the program flow. I see that there is a problem for some gdb-servers and it is preferable to have a disturbed real-time behavior instead of a completely halted firmware. Unfortunately, for the cases where the GDB server does the right thing this feature diminishes the debugging quality quite a bit. 
Wouldn't it be possible to do the breakpoint clearing sequence optional?
Or make the disconnect sequence overrideable kind of like the `overrideLaunchCommands` option?

haneefdm, 2022-05-18:
First, I don't think you can rely on a debugger to preserve real-time behavior. It is a very invasive process. Almost zero operations occur in gdb without interrupting the program. I cannot even detach without interrupting
```
-target-detach
0000063780: 37-target-detach
0000063780: -> 37^error,msg="Cannot execute this command while the target is running.\nUse the \"interrupt\" command to stop the target\nand then try again."
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
```

You might be confusing `disconnect`(close TCP port) with `detach`. It is `detach` (optionally) followed by `disconnect` that causes the program to continue to run. Provided there are no breakpoints set. Another reason to interrupt are to remove all breakpoints because many servers don't. I have yet to see a well behaved gdb-server implemented per spec.

JLink does not even support a detach. It does not follow spec. Openocd, also has issues with detach. It in fact does nothing...

https://github.com/openocd-org/openocd/blob/7d2ea186cf212c64a8e5b10725e0a2512a137fbe/src/server/gdb_server.c#L3406

I have my own custom openocd which prepares for a proper detach. This is what mine looks like

```
static int gdb_detach(struct connection *connection)
{
	struct gdb_connection *gdb_con = connection->priv;

	if (!gdb_con) {
		/* When a detach is immediately followed by a disconnect (port closed) we can actually
		 * get a detach after a disconnect has already taken place. Not sure how this is possible.
		 * Very hard to duplicate, but protect ourselves just in case */
		LOG_ERROR("Detach received aftr a disconnect.");
		return ERROR_FAIL;
	}

	if (gdb_con->attached) {
		struct target *target = get_target_from_connection(connection);

		breakpoint_clear_target(target);
		watchpoint_clear_target(target);
		if (target->state == TARGET_HALTED)
			target_resume(target, 1, 0, 0, 0);

		gdb_con->attached = false;
		target_call_event_callbacks(target, TARGET_EVENT_GDB_DETACH);
	}

	return gdb_put_packet(connection, "OK", 2);
}
```

The `target_resume` is the key (besides removing breakpoints automatically which the spec does not say but should be done for a proper resume). This can only be done by the server. Even the above is wrong because the detach should be unconditional (not relying on a previous attach). But technically, we always attach to a gdb-server

If one does a disconnect without doing a detach first, then the program is left in whatever state it is in and on non-embedded systems, the process is terminated.

https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_22.html

Technically, detach and disconnect are unrelated. And it is `detach` that should do all the right stuff.

So, that is the protocol. And, I have never seen an embedded gdb-server do this properly

Also `disconnect` is supposed to an orderly closing of the TCP connection so that that TCP Port is available for next time. And cause single-connect servers to also exit (after disconnecting from SWD) cleanly.

closed 2022-07-23 by haneefdm