Marus/cortex-debug #630: Path length issues complicate checkout on Windows

bmcdonnell-fb, 2022-04-07:
I get several `Filename too long` error messages when trying to checkout on Windows. Even with the local repo at the root of the C-drive, named after the project, as quoted below. (Many more when I have it nested a few levels down.)

I changed my working copy dir to `/c/cdbg` for the time being, and was able to checkout (to the latest `master`, `94450f9`). But this is problematic (making me relocate to a non-uniform and crypticly-named location), and liable to break harder later.

Can you flatten these directories a bit, or abbreviate the SHAs in the filesnames, or something to reduce these path lengths?

```
MINGW64 /c/cortex-debug (master)
$ git merge origin/master
error: unable to create file webview/node_modules/@vscode/webview-ui-toolkit/.vscode-test-web/vscode-web-insider-b7730c807c1e0fdacdcc91a854bb963715b2a89b/extensions/typescript-language-features/dist/browser/node_modules/typescript/lib/pt-br/diagnosticMessages.generated.json: Filename too long
error: unable to create file webview/node_modules/@vscode/webview-ui-toolkit/.vscode-test-web/vscode-web-insider-b7730c807c1e0fdacdcc91a854bb963715b2a89b/extensions/typescript-language-features/dist/browser/node_modules/typescript/lib/zh-cn/diagnosticMessages.generated.json: Filename too long
error: unable to create file webview/node_modules/@vscode/webview-ui-toolkit/.vscode-test-web/vscode-web-insider-b7730c807c1e0fdacdcc91a854bb963715b2a89b/extensions/typescript-language-features/dist/browser/node_modules/typescript/lib/zh-tw/diagnosticMessages.generated.json: Filename too long
Updating ebf23c3..94450f9
```

haneefdm, 2022-04-07:
Webview is a module from Microsoft. I checked in the snapshot of it as I did not want it to change. But looks like Windows Git has serious problems with it. I was able to duplicate it.

It will take me some time to re-jig and update docs on how to clone a repo and do an `npm install`. When I do that, two `npm installs` will be required...unfortunately. One for everything else and one for webview from MS. Can't remember the recipe totally, it I have to refer to my time machine :-)

haneefdm, 2022-04-08:
I updated the build instructinos in the [README](https://github.com/Marus/cortex-debug/blob/master/README.md) file as well as the appropriate .gitignore. Pushed the changes. Can you try it out and let me know. Especially the build instructions

bmcdonnell-fb, 2022-04-11:
I wasn't trying to build it; I only wanted to clone, checkout, and poke through the code. I verified that this resolved the issue for my purpose, but you or someone else may still want to check to ensure that you can build and run as expected.

Thanks. :)

closed 2022-04-11 by haneefdm