Marus/cortex-debug #631: Failed to launch OpenOCD GDB Server: Timeout

PaulDelestrac, 2022-04-08:
Hi, 

Configuration: 
- host: Dell x86, Ubuntu 20.04
- target: STM32F072RB-Nucleo (Cortex M0)
- probe: stlink based
- VSCode v1.66.1 + cortex-debug v1.4.4

I've been trying to debug using this extension, but I did not manage to get it working despite the fact that I can debug using OpenOCD (v0.10.0) + GDB (arm-none-eabi-gdb v10.3)...
When I launch the debugging on VSCode, the terminal shows that OpenOCD has been launched, printing these lines:
```bash
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/paul/Dev/hello -f /home/paul/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/stlink-v2-1.cfg -f target/stm32f0x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v38 API v2 SWIM v27 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.251053
Info : stm32f0x.cpu: hardware has 4 breakpoints, 2 watchpoints
```
Then, after a few seconds, these two lines appear: 
```shell
[2022-04-08T22:04:22.513Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
On the Debug Console (with "both" DevDebugOutput), I get the following lines:
```shell
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/paul/Dev/hello/target/thumbv6m-none-eabi/debug/blink
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/paul/Dev/hello/target/thumbv6m-none-eabi/debug/blink
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /home/paul/Dev/hello/target/thumbv6m-none-eabi/debug/blink
1-gdb-set target-async on
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/paul/Dev/hello -f /home/paul/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/stlink-v2-1.cfg -f target/stm32f0x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 17 ms
Finished reading symbols from nm: Time: 18 ms
-> =thread-group-added,id="i1"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
-> ~"Reading symbols from /home/paul/Dev/hello/target/thumbv6m-none-eabi/debug/blink...\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from /home/paul/Dev/hello/target/thumbv6m-none-eabi/debug/blink...\n"}]}
Reading symbols from /home/paul/Dev/hello/target/thumbv6m-none-eabi/debug/blink...
-> 1^done
GDB -> App: {"output":"","token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
2-gdb-version
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"}]}
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Copyright (C) 2021 Free Software Foundation, Inc.\n"}]}
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."}]}
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nType \"show copying\" and \"show warranty\" for details.\n"}]}
-> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"}]}
-> ~"Type \"show configuration\" for configuration details.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"show configuration\" for configuration details.\n"}]}
-> ~"For bug reporting instructions, please see:\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For bug reporting instructions, please see:\n"}]}
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"<https://www.gnu.org/software/gdb/bugs/>.\n"}]}
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."}]}
-> ~"\n\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\n\n"}]}
-> ~"For help, type \"help\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For help, type \"help\".\n"}]}
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"apropos word\" to search for commands related to \"word\".\n"}]}
-> 2^done
GDB -> App: {"output":"","token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
3-interpreter-exec console "set print demangle on"
-> 3^done
GDB -> App: {"output":"","token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","print asm-demangle"],["value","on"]]}]}
-> 4^done
GDB -> App: {"output":"","token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
Failed to launch OpenOCD GDB Server: Timeout.
```
As I understand it, no errors appeared before the last line: `Failed to launch OpenOCD GDB Server: Timeout.`.

My `launch.json` configuration is the following:
```json
{
    "type": "cortex-debug",
    "request": "launch",
    "name": "Debug with OpenOCD",
    "servertype": "openocd",
    "cwd": "${workspaceRoot}",
    "runToEntryPoint": "main",
    "executable": "${workspaceFolder}/target/thumbv6m-none-eabi/debug/blink",
    "configFiles": [
        "interface/stlink-v2-1.cfg",
        "target/stm32f0x.cfg"
    ],
    "showDevDebugOutput": "both",
},
```

Could you help me understand what I'm doing wrong?

haneefdm, 2022-04-08:
As you can see, openocd just quit. Not much we can do. You can run that command yourself from a shell and investigate why? Maybe wrong board. I don't see openocd ever open a connection to gdb either.

Chances are you are using some version of openocd that is not ST's version distributed by the ST. `OpenOCD (v0.10.0)` means nothing unfortunately. There are many custom versions and ST has one too.

On gdb's side, it barely even got started and never had a chance to talk to openocd.

haneefdm, 2022-04-08:
Btw, 0.10.0 is super old and will not work.

closed 2022-04-08 by haneefdm