Marus/cortex-debug #632: OpenOCD GDB Server Quit Unexpectedly

zenghaicang, 2022-04-10:
I was able to use opendocd normally a month ago.  Now update to the latest version of cortex-debug, it's not working properly.

launch.json
{
            "type": "cortex-debug",
            "request": "launch",
            "name": "Openocd jlink",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "showDevDebugOutput": true,    
            "runToMain": true,           
            "preLaunchTask": "compile",
            "executable": "Debug/${workspaceFolderBasename}.elf",
            "device": "STM32F103C8",            
            "configFiles": [
                "interface/jlink.cfg",
                "target/stm32f1x.cfg"                
            ],            "svdFile":"${env:GNU_ARM_PACKAGE_HOME}/Keil/STM32F1xx_DFP/2.4.0/SVD/STM32F103xx.svd",             
 },

Debug output information：

Reading symbols from d:/tools\xpack-arm-none-eabi-gcc-10.3.1-2.3\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p D:\code\stm32f1\Debug\stm32f1.elf
Launching GDB: "D:\\tools\\xpack-arm-none-eabi-gcc-10.3.1-2.3\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "D:\\code\\stm32f1\\Debug\\stm32f1.elf"
1-gdb-set target-async on
Launching gdb-server: "D:\\tools\\xpack-openocd-0.11.0-4\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:\\code\\stm32f1" -f "c:/Users/zenghc/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl" -f interface/jlink.cfg -f target/stm32f1x.cfg
    Please check TERMINAL tab (gdb-server) for output from D:\tools\xpack-openocd-0.11.0-4\bin\openocd.exe
Finished reading symbols from objdump: Time: 166 ms
Finished reading symbols from nm: Time: 163 ms
D:\tools\xpack-arm-none-eabi-gcc-10.3.1-2.3\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
OpenOCD GDB Server Quit Unexpectedly. See gdb-server output for more details.

zenghaicang, 2022-04-10:
I still have problems after using absolute paths

launch.json
 {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Openocd jlink",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "showDevDebugOutput": true,    
            "runToMain": true,           
            "preLaunchTask": "编译",
            "executable": "Debug/${workspaceFolderBasename}.elf",
            "device": "STM32F103C8",            
            "configFiles": [
                "D:/tools/xpack-openocd-0.11.0-4/scripts/interface/jlink.cfg",
                "D:/tools/xpack-openocd-0.11.0-4/scripts/target/stm32f1x.cfg"                
            ],
            "svdFile":"${env:GNU_ARM_PACKAGE_HOME}/Keil/STM32F1xx_DFP/2.4.0/SVD/STM32F103xx.svd",             
        },

Debug output information：
Reading symbols from D:\tools\xpack-arm-none-eabi-gcc-10.3.1-2.3\bin\arm-none-eabi-objdump.exe --syms -C -h -w D:\code\stm32f1\Debug\stm32f1.elf
Reading symbols from d:/tools\xpack-arm-none-eabi-gcc-10.3.1-2.3\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p D:\code\stm32f1\Debug\stm32f1.elf
Launching GDB: "D:\\tools\\xpack-arm-none-eabi-gcc-10.3.1-2.3\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "D:\\code\\stm32f1\\Debug\\stm32f1.elf"
1-gdb-set target-async on
Launching gdb-server: "D:\\tools\\xpack-openocd-0.11.0-4\\bin\\openocd.exe" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "D:\\code\\stm32f1" -f "c:/Users/zenghc/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl" -f "D:/tools/xpack-openocd-0.11.0-4/scripts/interface/jlink.cfg" -f "D:/tools/xpack-openocd-0.11.0-4/scripts/target/stm32f1x.cfg"
    Please check TERMINAL tab (gdb-server) for output from D:\tools\xpack-openocd-0.11.0-4\bin\openocd.exe
Finished reading symbols from nm: Time: 253 ms
Finished reading symbols from objdump: Time: 274 ms
D:\tools\xpack-arm-none-eabi-gcc-10.3.1-2.3\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from D:\\code\\stm32f1\\Debug\\stm32f1.elf...\n"}]}
Reading symbols from D:\code\stm32f1\Debug\stm32f1.elf...
GDB -> App: {"output":"","token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
2-gdb-version
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"GNU gdb (xPack GNU Arm Embedded GCC x86_64) 10.2.90.20210621-git\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Copyright (C) 2021 Free Software Foundation, Inc.\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\nType \"show copying\" and \"show warranty\" for details.\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"This GDB was configured as \"--host=x86_64-w64-mingw32 --target=arm-none-eabi\".\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"show configuration\" for configuration details.\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For bug reporting instructions, please see:\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"<https://www.gnu.org/software/gdb/bugs/>.\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"\n\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"For help, type \"help\".\n"}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"Type \"apropos word\" to search for commands related to \"word\".\n"}]}
GDB -> App: {"output":"","token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
3-interpreter-exec console "set print demangle on"
GDB -> App: {"output":"","token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
4-interpreter-exec console "set print asm-demangle on"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","print asm-demangle"],["value","on"]]}]}
GDB -> App: {"output":"","token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
OpenOCD GDB Server Quit Unexpectedly. See gdb-server output for more details.

haneefdm, 2022-04-10:
If OpenOCD is failing our focus should also be on why? It did not even start or give a chance to connect
```
Please check TERMINAL tab (gdb-server) for output from D:\tools\xpack-openocd-0.11.0-4\bin\openocd.exe
```
Can you run it manually from a shell prompt? If not wny.

Also, we made multiple releases of cortex-debug, so I can't tell when it worked and when it stopped

zenghaicang, 2022-04-10:
Sorry, it was my mistake. 
I have installed two versions of jlink driver, one is V6.98 and the other is V7.22.
In lower versions, it is necessary to switch the JLINK driver.  https://wiki.segger.com/OpenOCD
When I used o version 7.22, everything was resolved.

 V6.98:
PS C:\Users\zenghc> openocd -f interface/jlink.cfg -f target/stm32f1x.cfg -c init -c halt -c reset -c exit
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-03-25-17:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Warn : Failed to open device: LIBUSB_ERROR_NOT_FOUND.
Error: No J-Link device found

closed 2022-04-10 by zenghaicang