Marus/cortex-debug #634: ECONNRESET by J-Link GDB when removing "runToMain": true

fusedFET, 2022-04-12:
Hello,

Using the latest version of Cortex-Debug (1.4.4), when I remove `"runToMain": true`, from the configuration, the debugger seems to flash and tries to start but fails immediately. This is a new bug which was not present in v1.4.3. Here is the output of the GDB Debugger console:

```
Read 2 bytes @ address 0x00017DE8 (Data = 0x2000)
[2022-04-12T13:06:10.308Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET
[2022-04-12T13:06:10.309Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

Here is the configuration file:
```
            "name": "IOT_Debug",
            "cwd": "${workspaceRoot}",
            "executable": "./build/zephyr/zephyr.elf",
            "device": "NRF52832_XXAA",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "rtos": "Zephyr",
            //"runToMain": true,
            "svdFile": "./nrf52.svd",
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "label": "RTT",
                        "port": 0,
                        "type": "console"
                    }
                ]
            }
        },
```
When adding "runToMain": true, it works again on v1.4.4.

haneefdm, 2022-04-12:
Full Debug output from the Debug Console and the gdb-server Terminal could be helpful.

Make sure you enable debug in your launch.json

fusedFET, 2022-04-12:
Sure, I should have added them in the first splace.

```
Waiting for gdb server to start...[2022-04-12T14:15:32.902Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C://Program Files/SEGGER//JLink//JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52832_XXAA -rtttelnetport 19021 -rtos GDBServer/RTOSPlugin_Zephyr.dll
SEGGER J-Link GDB Server V7.64b Command Line Version

JLinkARM.dll V7.64b (DLL compiled Apr  8 2022 15:30:15)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52832_XXAA -rtttelnetport 19021 -rtos GDBServer/RTOSPlugin_Zephyr.dll
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 NRF52832_XXAA
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

RTT Telnet Port set to 19021
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled Mar  3 2022 10:16:14
Hardware: V11.00
S/N: xxxxxxxxxx
OEM: SEGGER-EDU
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 3.04 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00017DE8 (Data = 0xF3802000)
Read 2 bytes @ address 0x00017DE8 (Data = 0x2000)
Loading RTOS plugin: GDBServer/RTOSPlugin_Zephyr.dll...
RTOS plugin (API v1.1) loaded successfully
RTOS plugin: Loaded
Received symbol: _kernel (0x20004870)
Received symbol: _kernel_thread_info_offsets (0x0003057C)
Received symbol: z_sys_post_kernel (0x200063A4)
All mandatory symbols successfully loaded.
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x00017DE8)
Received monitor command: reset
Resetting target
Downloading 220 bytes @ address 0x00000000
Downloading 16048 bytes @ address 0x000000E0
Downloading 15840 bytes @ address 0x00003F90
Downloading 16048 bytes @ address 0x00007D70
Downloading 16000 bytes @ address 0x0000BC20
Downloading 16064 bytes @ address 0x0000FAA0
Downloading 15984 bytes @ address 0x00013960
Downloading 16032 bytes @ address 0x000177D0
Downloading 16016 bytes @ address 0x0001B670
[... 252 lines cut ...]
GDB -> App: {"output":"","token":16,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
17-thread-list-ids
-> =thread-created,id="2",group-id="i1"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","2"],["group-id","i1"]]}]}
-> ~"[New Remote target]\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"[New Remote target]\n"}]}
[New Remote target]
-> 17^done,thread-ids={thread-id="1",thread-id="2"},current-thread-id="1",number-of-threads="2"
GDB -> App: {"output":"","token":17,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"],["thread-id","2"]]],["current-thread-id","1"],["number-of-threads","2"]]}}
18-thread-info 1
-> =thread-exited,id="1",group-id="i1"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-exited","output":[["id","1"],["group-id","i1"]]}]}
-> 18^done,threads=[]
GDB -> App: {"output":"","token":18,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[]]]}}
19-thread-info 2
-> 19^done,threads=[{id="2",target-id="Remote target",frame={level="0",addr="0x00017de8",func="z_arm_reset",args=[],file="C:/Users/MyWindowsUserName/ncs/v1.9.0/zephyr/arch/arm/core/aarch32/cortex_m\\reset.S",fullname="C:\\Users\\MyWindowsUserName\\ncs\\v1.9.0\\zephyr\\arch\\arm\\core\\aarch32\\cortex_m\\reset.S",line="69",arch="armv7e-m"},state="stopped"}]
GDB -> App: {"output":"","token":19,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","2"],["target-id","Remote target"],["frame",[["level","0"],["addr","0x00017de8"],["func","z_arm_reset"],["args",[]],["file","C:/Users/MyWindowsUserName/ncs/v1.9.0/zephyr/arch/arm/core/aarch32/cortex_m\\reset.S"],["fullname","C:\\Users\\MyWindowsUserName\\ncs\\v1.9.0\\zephyr\\arch\\arm\\core\\aarch32\\cortex_m\\reset.S"],["line","69"],["arch","armv7e-m"]]],["state","stopped"]]]]]}}
VSCode stackTraceHack triggered
Finished waiting for stacktrace request in 148 ms in 9 tries
20-exec-continue --all
-> 20^error,msg="Cannot execute this command without a live selected thread."
GDB -> App: {"output":"","token":20,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Cannot execute this command without a live selected thread."]]}}
21-thread-list-ids

```

haneefdm, 2022-04-12:
Do you mind doing the same (just the Debug Console) for 1.4.3 where it was working? Thanks.

haneefdm, 2022-04-12:
Not sure what JLink is doing but it has gotten GDB thoroughly confused.

* Your program is paused (in the Reset handler most likely)
* JLink reports two threads to gdb which is then reported to VSCode. And, current-thread is "1"
* VSCode queries for each thread's stack trace one by one. But while that is happening, JLink deleted the current thread "1" and gdb lost its current thread.
* We issue a continue from the reset-handler and the continue fails because gdb says the current thread is gone.

JLink is not supposed to report a change in program status while the program is paused. And, why didn't gdb go out of sync because it should have expected that (can happen in other instances). It should have determined a new current thread automatically. If it can't, we certainly can't.

The fact that it was working before maybe just a coincidence and the timing of queries.

Also, I don't understand why JLink quit. We did nothing to make it quit!!! It is possible gdb disconnected as well, but there is no indication of that.

fusedFET, 2022-04-13:
Thank you for looking into this, here is the same process (nothing changed) on 1.4.3

```

Waiting for gdb server to start...[2022-04-13T06:55:43.168Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C://Program Files/SEGGER//JLink//JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52832_XXAA -rtttelnetport 19021 -rtos GDBServer/RTOSPlugin_Zephyr.dll
SEGGER J-Link GDB Server V7.64b Command Line Version

JLinkARM.dll V7.64b (DLL compiled Apr  8 2022 15:30:15)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device NRF52832_XXAA -rtttelnetport 19021 -rtos GDBServer/RTOSPlugin_Zephyr.dll
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 NRF52832_XXAA
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

RTT Telnet Port set to 19021
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled Mar  3 2022 10:16:14
Hardware: V11.00
S/N: XXXXXXXX
OEM: SEGGER-EDU
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 3.04 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00017DE8 (Data = 0xF3802000)
Read 2 bytes @ address 0x00017DE8 (Data = 0x2000)
Loading RTOS plugin: GDBServer/RTOSPlugin_Zephyr.dll...
RTOS plugin (API v1.1) loaded successfully
RTOS plugin: Loaded
Received symbol: _kernel (0x20004870)
Received symbol: _kernel_thread_info_offsets (0x0003057C)
Received symbol: z_sys_post_kernel (0x200063A4)
All mandatory symbols successfully loaded.
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x00017DE8)
Received monitor command: reset
Resetting target
Downloading 220 bytes @ address 0x00000000
Downloading 16048 bytes @ address 0x000000E0
Downloading 15840 bytes @ address 0x00003F90
Downloading 16048 bytes @ address 0x00007D70
Downloading 16000 bytes @ address 0x0000BC20
Downloading 16064 bytes @ address 0x0000FAA0
Downloading 15984 bytes @ address 0x00013960
Downloading 16032 bytes @ address 0x000177D0
[... 219 lines cut ...]
Executed memU32 0x20005ff4 = 0
-> 15^done
GDB -> App: {"output":"","token":15,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
16-interpreter-exec console "monitor exec memU32 0x20005ff8 = 0"
-> @"Executed memU32 0x20005ff8 = 0\r\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"target","content":"Executed memU32 0x20005ff8 = 0\r\n"}]}
Executed memU32 0x20005ff8 = 0
-> 16^done
GDB -> App: {"output":"","token":16,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
17-interpreter-exec console "monitor exec SetRTTAddr 0x20005ff0"
-> @"Executed SetRTTAddr 0x20005ff0\r\n"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"target","content":"Executed SetRTTAddr 0x20005ff0\r\n"}]}
Executed SetRTTAddr 0x20005ff0
-> 17^done
GDB -> App: {"output":"","token":17,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
18-exec-continue
-> 18^running
GDB -> App: {"output":"","token":18,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
-> *running,thread-id="all"
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
mi2.status = running

```

I have not changed in anyway the physical setup and this is systematic. I checked the probe connection, but it is correctly connected.

haneefdm, 2022-05-13:
Can you try 1.5.0 and let me know if it fixes anything? It is a pre-release and offered via VSCode normal mechanism to install extensions.

fusedFET, 2022-05-16:
Hi thank you for the update!
I tested using 1.5.0: I no longer see the ECONNRESET error and I can hit breakpoints without forcing "runToMain": true
However I lost the embedded RTT console. I attached the log below.
[1_5_0_Bug_no_run_to_main.txt](https://github.com/Marus/cortex-debug/files/8698585/1_5_0_Bug_no_run_to_main.txt)

haneefdm, 2022-05-16:
Found the issue with the RTT. Will be fixed shortly. In 1.5.1 coming shortly. If you need an advanced build, I can provide it.

haneefdm, 2022-06-17:
1.5.1 is released

closed 2022-06-17 by haneefdm