Marus/cortex-debug #635: Debugging hangs until relaunch

saloid, 2022-04-15:
I'm trying to debug STM32F4 hardware with freeRTOS using J-link, but after succesfull launch and run, debug totally hangs when it hits one of breakpoints (hardware or `__asm__("BKPT")`). All action buttons (reset, pause and restart) became unresponsive (I can press them and even see command in debug console, but nothing else).

Before reaching breakpoint everything works fine (I can run and pause, do step-by-step debugging and everything else), but when breakpoint is reached everything hangs. But when I close and launch debug session again - I can see that execution stopped at my breakpoint.

<details>
  <summary>launch.json</summary>

```json
{
"version": "0.2.0",
    "configurations": [
        {
            "name": "Jlink",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/first_part.elf",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "jlink",
            "svdFile": "${workspaceRoot}/STM32F407.svd",
            "interface": "swd",
            "device": "STM32F407VG",
            "armToolchainPath": "/home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-10.3-2021.10/bin",
            "showDevDebugOutput": true,
            "preAttachCommands": [
                "set substitute-path /tmp/opentx ${workspaceRoot}/new_sources_path",
                "add-symbol-file ${workspaceRoot}/build/another_part.elf 0x080c0080",
            ],
            "rtos": "FreeRTOS",
            "moduleLoad": true,
        }
    ]
}
```
</details>

When brekpoint is reached (and debug hangs) I can see this in debug console:

```
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","3"],["group-id","i1"]]}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"[New Thread 268445668]\n"}]}
[New Thread 268445668]
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"breakpoint-modified","output":[["bkpt",[["number","1"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x080c38b8"],["func","main_loop_task"],["file","[/home/saloid/Documents/Projects/Crossfire/uhf/trunk/UHF_Firmware/src/main.c]()"],["fullname","[/home/saloid/Documents/Projects/Crossfire/uhf/trunk/UHF_Firmware/src/main.c]()"],["line","247"],["thread-groups",["i1"]],["times","1"],["original-location","[/home/saloid/Documents/Projects/Crossfire/uhf/trunk/UHF_Firmware/src/main.c:247]()"]]]]}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-exited","output":[["id","3"],["group-id","i1"]]}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","4"],["group-id","i1"]]}]}
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"[New Thread 4748040]\n"}]}
[New Thread 4748040]
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":true,"type":"console","content":"[Switching to Thread 268445668]\n"}]}
[Switching to Thread 268445668]
```

And buttons are still in "Running state" (but really device is stopped)
[image]

When I push any button (reset, pause or restart) this appears in debug console:
```
94-exec-interrupt 
GDB -> App: {"output":"","token":94,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

saloid, 2022-04-15:
Seems that it depends on GDB version: `GNU gdb (GNU Tools for ARM Embedded Processors) 7.4.1.20130913-cvs` works better, but disassembly doesn't work here

haneefdm, 2022-04-15:
As you can see, we are attempting to pause the program via GDB and neither is responding. We rely 100% on GDB to do the right thing which then relies on JLink to do the right thing. We have to pause GDB before we can issue any more commands. In the output above, it is says the `94-exec-interrupt` succeeded so not sure what is the problem if you see nothing after that.

If a program halts by itself, JLink is supposed to report that to GDB. It doesn't matter is that is due to a breakpoint or `__asm__("BKPT")`

Yes, gdb 7.4.1 is quite old. I have had a lot of issues with late 2021 releases but early 2021 releases were okay.

Also, I need a bit more of the output from the Debug Console -- from the point the breakpoint was hit. Try `"showDevDebugOutput":"raw"` for more compact output.

saloid, 2022-04-17:
@haneefdm, thank you for your reply. Here is debug output with `"showDevDebugOutput":"raw"` when program hits breakpoint:

<details>
  <summary> ❌ARM GDB 10.3-2021.10 (GUI not responding)</summary>

```
-> =thread-created,id="3",group-id="i1"
-> ~"[New Thread 268445668]\n"
[New Thread 268445668]
-> =thread-exited,id="3",group-id="i1"
-> =thread-created,id="4",group-id="i1"
-> ~"[New Thread 4748040]\n"
[New Thread 4748040]
-> ~"\nThread "

Thread 
-> ~"3 received signal SIGTRAP, Trace[/breakpoint]() trap.\n"
3 received signal SIGTRAP, Trace[/breakpoint]() trap.
-> ~"[Switching to Thread 268445668]\n"
[Switching to Thread 268445668]
```

</details>

<details>
  <summary> ❌ARM GDB 9-2020-q2-update (GUI not responding)</summary>

```
-> =thread-exited,id="2",group-id="i1"
-> =thread-created,id="3",group-id="i1"
-> ~"[New Thread 4748040]\n"
[New Thread 4748040]
-> =thread-created,id="4",group-id="i1"
-> ~"[New Remote target]\n"
[New Remote target]
-> ~"\nThread "

Thread 
-> ~"2 received signal SIGTRAP, Trace[/breakpoint]() trap.\n"
2 received signal SIGTRAP, Trace[/breakpoint]() trap.
-> ~"[Switching to Thread 268445668]\n"
[Switching to Thread 268445668]
```

</details>

<details>
  <summary> ✅ARM GDB 7.4.1.20130913-cvs (GUI works)</summary>

```
-> ~"[Switching to Thread 268445668]\n"
[Switching to Thread 268445668]
-> *stopped,reason="signal-received",signal-name="SIGTRAP",signal-meaning="Trace/breakpoint trap",frame={addr="0x08055a00",func="HardFault_Handler",args=[],file="/old/path/source1.cpp",fullname="/full/path/source1.cpp",line="755"},thread-id="1",stopped-threads="all"
mi2.status = stopped
40-thread-list-ids
-> =thread-exited,id="1",group-id="i1"
-> =thread-created,id="2",group-id="i1"
-> ~"[New Thread 4748040]\n"
[New Thread 4748040]
-> =thread-created,id="3",group-id="i1"
-> ~"[New Remote target]\n"
[New Remote target]
-> 40^done,thread-ids={thread-id="3",thread-id="2"},number-of-threads="2"
41-thread-select 3
-> 41^done,new-thread-id="3",frame={level="0",addr="0x00000000",func="??",args=[]}
42-data-read-memory-bytes "0xe000ed00" 64
-> 42^done,memory=[{begin="0xe000ed00",offset="0x00000000",end="0xe000ed40",contents="41c20f410308000000c00008000305fa000000000002000000000000000000000000f0f000000000000002000000004002000000f8ed00e0f8ed00e000000000"}]
43-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
-> 43^done
[... 39 lines cut ...]
60-stack-list-frames --thread 3 0 1
-> 60^done,stack=[frame={level="0",addr="0x00000000",func="??"},frame={level="1",addr="0x00000000",func="??"}]
61-thread-list-ids
-> 61^done,thread-ids={thread-id="3",thread-id="2"},current-thread-id="3",number-of-threads="2"
62-thread-info 3
-> 62^done,threads=[{id="3",target-id="Remote target",frame={level="0",addr="0x00000000",func="??",args=[]},state="stopped"}]
63-thread-info 2
-> 63^done,threads=[{id="2",target-id="Thread 4748040",details=" [P: 0]",frame={level="0",addr="0x08055a00",func="HardFault_Handler",args=[],file="/old/path/source1.cpp",fullname="/full/path/source1.cpp",line="755"},state="stopped"}]
Debug-39: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00000000","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":39}
Debug-39: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00000000 -e 0x00000320 -- 5      800 bytes 
Suppressing output for '64-data-disassemble -s 0x00000000 -e 0x00000320 -- 5'
Debug-40: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00000000","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":40}
Debug-40: ******** Waiting for previous request to complete
Error: GDB failed: -data-disassemble: Mode argument must be 0, 1, 2, or 3. (from data-disassemble -s 0x00000000 -e 0x00000320 -- 5)
Debug-40: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00000000 -e 0x00000320 -- 5      800 bytes 
Suppressing output for '65-data-disassemble -s 0x00000000 -e 0x00000320 -- 5'
Error: GDB failed: -data-disassemble: Mode argument must be 0, 1, 2, or 3. (from data-disassemble -s 0x00000000 -e 0x00000320 -- 5)

```

</details>

Let me know if you want me to find exact GDB version where it was broken

haneefdm, 2022-04-18:
@saloid Thanks for providing the info. But the ones that fail (let us just use "ARM GDB 9-2020-q2-update"), can I have the whole output. I don't see any transactions between Cortex-Debug and GDB.

saloid, 2022-04-18:
<details>
<summary>9-2020-q2-update</summary>

```
Reading symbols from /home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-objdump --syms -C -h -w /home/saloid/Documents/Projects/my_project/binary_1.elf
Reading symbols from /home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/saloid/Documents/Projects/my_project/binary_1.elf
Launching GDB: /home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/saloid/Documents/Projects/my_project/binary_1.elf
1-gdb-set target-async on
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F407VG -rtos GDBServer/RTOSPlugin_FreeRTOS.so
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
-> =thread-group-added,id="i1"
-> ~"Reading symbols from /home/saloid/Documents/Projects/my_project/binary_1.elf...\n"
Reading symbols from /home/saloid/Documents/Projects/my_project/binary_1.elf...
Finished reading symbols from objdump: Time: 50 ms
Error: objdump failed! statics/globals/functions may not be properly classified: TypeError: t.push is not a function    ENOENT means program not found. If that is not the issue, please report this problem.-> 1^done
2-gdb-version
-> ~"GNU gdb (GNU Arm Embedded Toolchain 9-2020-q2-update) 8.3.1.20191211-git\n"
-> ~"Copyright (C) 2019 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
WARNING: Cortex-Debug will deprecate use of GDB version 8 after July 2022. Please upgrade to version 9+
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Finished reading symbols from nm: Time: 154 ms
5-interpreter-exec console "source /home/saloid/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdbsupport.init"
-> 5^done
6-interpreter-exec console "source /home/saloid/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 6^done
7-interpreter-exec console "set output-radix 0x10"
-> ~"Output radix now set to decimal 16, hex 10, octal 20.\n"
Output radix now set to decimal 16, hex 10, octal 20.
-> =cmd-param-changed,param="output-radix",value="16"
-> 7^done
8-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"0x0805529a in function_1 () at /old/path/source.cpp:196\n"
0x0805529a in function_1 () at /old/path/source.cpp:196
-> &"196\t/old/path/source.cpp: No such file or directory.\n"
196	/old/path/source.cpp: No such file or directory.
-> *stopped,frame={addr="0x0805529a",func="function_1",args=[],file="/old/path/source.cpp",fullname="/old/path/source.cpp",line="196",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 9^connected
10-interpreter-exec console "set substitute-path /old/path /new/path"
-> 10^done
11-interpreter-exec console "add-symbol-file /home/saloid/Documents/Projects/my_project/build/binary_2.elf 0x080c0080"
-> ~"add symbol table from file \"/home/saloid/Documents/Projects/my_project/build/binary_2.elf\" at\n"
add symbol table from file "/home/saloid/Documents/Projects/my_project/build/binary_2.elf" at
-> ~"\t.text_addr = 0x80c0080\n"
	.text_addr = 0x80c0080
-> ~"(y or n) [answered Y; input not from terminal]\n"
[... 324 lines cut ...]
44-interpreter-exec console "monitor reset"
-> @"Resetting target\r\n"
Resetting target
-> 44^done
45-exec-continue --all
-> 45^running
-> *running,thread-id="all"
mi2.status = running
-> =thread-exited,id="2",group-id="i1"
-> =thread-created,id="3",group-id="i1"
-> ~"[New Thread 4748040]\n"
[New Thread 4748040]
-> =thread-created,id="4",group-id="i1"
-> ~"[New Remote target]\n"
[New Remote target]
-> ~"\nThread "

Thread 
-> ~"2 received signal SIGTRAP, Trace/breakpoint trap.\n"
2 received signal SIGTRAP, Trace/breakpoint trap.
-> ~"[Switching to Thread 268445668]\n"
[Switching to Thread 268445668]
```

</details>

saloid, 2022-04-19:
@haneefdm I downloaded another one arm toolchain:

<details>
<summary>10-2020-q4-major (GDB 10.1.90.20201028-git)</summary>
```
Reading symbols from /home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-objdump --syms -C -h -w /home/saloid/Documents/Projects/my_project/binary_1.elf
Reading symbols from /home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/saloid/Documents/Projects/my_project/binary_1.elf
Launching GDB: /home/saloid/Documents/Projects/tools/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/saloid/Documents/Projects/my_project/binary_1.elf
1-gdb-set target-async on
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F407VG -rtos GDBServer/RTOSPlugin_FreeRTOS.so
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
-> =thread-group-added,id="i1"
-> ~"Reading symbols from /home/saloid/Documents/Projects/my_project/binary_1.elf...\n"
Reading symbols from /home/saloid/Documents/Projects/my_project/binary_1.elf...
Finished reading symbols from objdump: Time: 51 ms
Error: objdump failed! statics/globals/functions may not be properly classified: TypeError: t.push is not a function    ENOENT means program not found. If that is not the issue, please report this problem.-> 1^done
2-gdb-version
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.1.90.20201028-git\n"
-> ~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
Finished reading symbols from nm: Time: 137 ms
5-interpreter-exec console "source /home/saloid/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdbsupport.init"
-> 5^done
6-interpreter-exec console "source /home/saloid/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 6^done
7-interpreter-exec console "set output-radix 0x10"
-> ~"Output radix now set to decimal 16, hex 10, octal 20.\n"
Output radix now set to decimal 16, hex 10, octal 20.
-> =cmd-param-changed,param="output-radix",value="16"
-> 7^done
8-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"HardFault_Handler () at /old/path/board.cpp:755\n"
HardFault_Handler () at /old/path/board.cpp:755
-> &"755\t/old/path/board.cpp: No such file or directory.\n"
755	/old/path/board.cpp: No such file or directory.
-> *stopped,frame={addr="0x08055a00",func="HardFault_Handler",args=[],file="/old/path/board.cpp",fullname="/old/path/board.cpp",line="755",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 9^connected
10-interpreter-exec console "set substitute-path /old/path /new/path"
-> 10^done
11-interpreter-exec console "add-symbol-file /home/saloid/Documents/Projects/my_project/binary_2.elf 0x080c0080"
-> ~"add symbol table from file \"/home/saloid/Documents/Projects/my_project/binary_2.elf\" at\n"
add symbol table from file "/home/saloid/Documents/Projects/my_project/binary_2.elf" at
-> ~"\t.text_addr = 0x80c0080\n"
	.text_addr = 0x80c0080
-> ~"(y or n) [answered Y; input not from terminal]\n"
[... 120 lines cut ...]
45-interpreter-exec console "monitor reset"
-> @"Resetting target\r\n"
Resetting target
-> 45^done
46-exec-continue --all
-> 46^running
-> *running,thread-id="all"
mi2.status = running
-> =thread-created,id="3",group-id="i1"
-> ~"[New Thread 268445668]\n"
[New Thread 268445668]
-> =thread-exited,id="3",group-id="i1"
-> =thread-created,id="4",group-id="i1"
-> ~"[New Thread 4748040]\n"
[New Thread 4748040]
-> ~"\nThread "

Thread 
-> ~"3 received signal SIGTRAP, Trace/breakpoint trap.\n"
3 received signal SIGTRAP, Trace/breakpoint trap.
-> ~"[Switching to Thread 268445668]\n"
[Switching to Thread 268445668]
```

</details>

saloid, 2022-04-22:
Also I noticed that debug more stable if I remove `"rtos": "FreeRTOS"` (but I loose some features in this case)

haneefdm, 2022-05-11:
Can you try our latest pre-release? You have to enable it on our extension in VSCode

haneefdm, 2022-05-18:
Here is another pre-release to try.

https://github.com/Marus/cortex-debug/releases/tag/v1.5.1-pre4

Please re-open with details if this is still a problem. Haven't heard back from previous message

saloid, 2022-06-05:
Sorry for long time to reply.
Tried pre-release version from vscode extensions shop ("Last updated 02/06/2022, 00:57:42") - still have this issue

haneefdm, 2022-06-05:
I don't know how to help you if this so gdb specific, but I don't recommend using 7.4. One more try. Can you provide the logs with `"showDevDebugOutput":"vscode"`?

Or maybe this issue is really a JLink issue where it is the one who is not responding properly to gdb. Do you wee anything range in the gd-server TERMINAL window (also attach please).

saloid, 2022-06-10:
Seems that config `#define configUSE_LIST_DATA_INTEGRITY_CHECK_BYTES   1` in FreeRTOSConfig.h file was causing such issue. Without this define everything works as it should.
Thank you for your patiance and support)

saloid, 2022-06-10:
Issue was when FreeRTOS is fully initialized and running.