Marus/cortex-debug #636: Chained configurations: 'folder' option is no longer working

AndriiLishchynskyi, 2022-04-17:
Hello,

I am using chained configurations in order to launch multi-core debugging. I have my second 'chained' configuration in a separate launch.json file. In a primary configuration, path to the launch.json where second config is defined, given to a 'folder' option. With the latest Cortex-Debug, launching of second chained config fails due to inability to find second config. Neither absolute nor relative path given to 'folder' option work. However, when I switch to Cortex-Debug 1.2.2 it starts working. Thus, It looks like a problem is somewhere inside of Cortex-Debug starting from 1.4.0 version.

My project is quite large, so I am attaching .vscode folders structure for your analysis. I am trying to launch 'Multi-Core Debug (KitProg3_MiniProg4)' configuration.
[Multi-Core-Prototype.zip](https://github.com/Marus/cortex-debug/files/8502426/Multi-Core-Prototype.zip)

I am using Infineon PSoC6 device on Windows machine.

haneefdm, 2022-04-17:
Only on Windows or other platforms as well?

AndriiLishchynskyi, 2022-04-17:
I don't have other machines to try. I can ask someone to try tomorrow and will let you know.

haneefdm, 2022-04-17:
Hmmm. I don't see any chained configs in the zip file launch.json(s). FYI: it works in my own example with 1.4.x on Intel MacOS

AndriiLishchynskyi, 2022-04-17:
Please have a look at "Multi-Core-Prototype/.vscode/launch.json". There is a "Multi-Core Debug (KitProg3_MiniProg4)" config that refers "Attach PSoC6 CM4 (KitProg3_MiniProg4)".
Sometimes strange things might happen with path processing on different platforms) so it can be windows-only issue, will check with team members tomorrow and let you know.

haneefdm, 2022-04-17:
Okay, I missed that launch.json. 

I see you are still using `workspaceRoot`. Deprecated long time ago by VSCode.

https://code.visualstudio.com/docs/editor/variables-reference#_why-isnt-workspaceroot-documented

We must restrict ourselves to these names

https://code.visualstudio.com/docs/editor/variables-reference#_predefined-variables

Since in the Debug Console, I print out what the chained config looks like (after translations) you can see what the final result is. Regardless, full path name should have worked but did you actually type in the full path name or use one of the variables defined by VSCode?

haneefdm, 2022-04-17:
See also `Variables scoped per workspace folder`

https://code.visualstudio.com/docs/editor/variables-reference#_variables-scoped-per-workspace-folder

Not sure it will be useful because that lanch.json is at the root.

AndriiLishchynskyi, 2022-04-18:
Hello Haneef,
Sorry, today I didn't have a chance to try it on other machine (Linux or Mac), but I have some additional info from my Windows machine.

I am trying different options, but neither of them work:
```
C:/Users/lishchynskyi/mtw_psoc632/Multi-Core-Prototype/app_cm4
./app_cm4
${workspaceFolder}/app_cm4
${workspaceFolder:app_cm4}
```
I am always getting from Debug Console in a failed scenario: `"folder": "C:/Users\\lishchynskyi\\mtw_psoc632\\Multi-Core-Prototype\\app_cm4"`. Could that forward slash be a reason of the problem I am facing?

With Cortex-Debug 1.2.2, where everything works properly, I do not see a "folder" option printed to the console, thus not sure how translation is happening there.

BTW, I have updated launch.json templates by replacing `workspaceRoot` in favor of `workspaceFolder`, thanks.

haneefdm, 2022-04-19:
I am in a 2-day meeting so, have not had a chance to look at it. We do not touch that `folder` entry. Whatever VSCode gives us, we pass it along back to VSCode. It may not be the folder that is the problem and something else instead.

Will take a look on Wed

haneefdm, 2022-05-13:
Fixed in v1.5.0

closed 2022-05-13 by haneefdm