Marus/cortex-debug #641: renode configuration

jhgorse, 2022-04-21:
Greetings,

I am trying to get `cortex-debug` to play nicely with [renode](https://github.com/renode/renode).

The latest configuration that I've tried and failed with was:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug Renode",
            "cwd": "${workspaceFolder}",
            "executable": "./test1.elf",
            "device": "STM32F429x",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "external",
            "gdbTarget": "localhost:3333",
            "toolchainPrefix": "arm-none-eabi-",
            "gdbPath": "arm-none-eabi-gdb",
            "showDevDebugOutput": "raw",
            "overrideLaunchCommands": ["monitor start", "continue"],
            "overrideRestartCommands": ["monitor pause","monitor runMacro $reset"],
            "postStartSessionCommands": ["break main"]
        }
    ]
}
```

The minimum project is setup here if you'd like to reproduce:
https://github.com/jhgorse/renode_stm32f429_vscode

The issues are:
1. it does not breakpoint on main, regardless
2. it does not handle resets (probably renode's fault)

What do you recommend?

xref: https://github.com/Marus/cortex-debug/issues/288

Cheers,
Joe

haneefdm, 2022-04-21:
Don't know anything or how I can help. No bandwidth. You should enable debug and see what is going on in the Debug Console. Everything is done via gdb...
```
            "overrideLaunchCommands": ["monitor start", "continue"],
```
I don't the is a continue is needed. We do the continue ourselves,  but if you do it then we never get a chance to set a breakpoint in main. We can't even get to `postStartSessionCommands` because we are waiting for the `overrideLaunchCommands` to finish and the program to stop.

Read https://github.com/Marus/cortex-debug/wiki/Overview

jhgorse, 2022-04-22:
Removing `continue` from
```
  "overrideLaunchCommands": ["monitor start", "continue"],
```
works.

Reset is slightly better.

Now synchronizing the left panes. Disassembly doesn't show. Guess that is a new thing, eh?

closed 2022-04-22 by haneefdm

bissonex, 2022-10-18:
Hi @jhgorse, your config helped me a lot. Thank you

P.S. I have added an entry to the wiki with this Renode Configuration.