Marus/cortex-debug #643: Make runToEntryPoint timeout configurable

bmcdonnell-fb, 2022-04-22:
Feature request: please make the `runToEntryPoint` timeout configurable.

## Details

According to [`debug_attributes.md`](https://github.com/Marus/cortex-debug/blob/80bba429b1b3f5047ead119c30620522a16678a1/debug_attributes.md),

> | Attribute | Applies To | Description |
> | --------- | ---------- | ----------- |
> | ... | ... | ... |
> | runToEntryPoint | Common | If enabled the debugger will run until the start of the given function. There is a 5 second timeout |

I'm debugging an application on a platform with a bootloader which takes longer than 5 seconds to complete, so the debugger is stopping at  a "random" place in the disassembly in the bootloader code. I'd like to be able to modify the `runToEntryPoint` timeout in `launch.json`, in this case to make it longer, so it doesn't actually stop before the specified entry point (`main`).

bmcdonnell-fb, 2022-04-22:
5 seconds seems like a reasonable default, though. 😃

haneefdm, 2022-04-22:
Sorry, don't want to add an option for that. Instead, set a breakpoint in your main and do not use 'runToEntryPoint'. The number of options has become a testing nightmare. The timeout was added by someone from Nordic and has got in my way several times in several ways but I was debugging the extension  most of the time.

I have to think of something else rather than another 20 lines to package.json and testing it.

bmcdonnell-fb, 2022-04-22:
How about an option to disable the timeout?

Or, would it be better to even get rid of it?

haneefdm, 2022-04-22:
Yes, I would rather get rid of it. If the `pause` button still works -- which may be a problem because the session has not really started yet so VSCode may not have enabled that button. I have to experiment with that.

If I were to increase the timeout, how much would you need?

bmcdonnell-fb, 2022-04-22:
> If I were to increase the timeout, how much would you need?

Long enough for it not to be a sensible default, I think. Currently > 30 seconds, subject to change.

haneefdm, 2022-04-24:
Timeout going away. There was an issue with VSCode pause not working (issue with VSCode actually) when waiting for the breakpoint to hit and this is now addressed. Now that the pause button works, users have control.

Should be in a preview release (1.5.0) soon.

closed 2022-06-17 by haneefdm