Marus/cortex-debug #644: Unable to use xPack build tool shims on Windows

grigorig, 2022-05-03:
xPack installs a bunch of wrapper scripts in `xpacks/.bin` by default since Windows lacks good support for symbolic links. So you may have a wrapper called `xpacks/.bin/arm-none-eabi.gdb` that calls the actual executable. There are `.cmd` and `.ps1` versions of the wrappers as well.  This doesn't appear to work correctly with cortex-debug, since it always expects an actual `.exe` file.

That means there is no portable way to refer to the xPack installed tools, as far as I can see. That's a bit of a bummer, since abstracting away these details is the point of xPack.

haneefdm, 2022-05-03:
See. https://github.com/Marus/cortex-debug/issues/563

But I can tell you this is unsustainable. To rename every executable we might use and having to create a custom setting for everything.

We now also need `nm` and we don't have a setting for it. We just modify the objdump path and use it as is. Each setting we add had a TON of overhead.

closed 2022-05-03 by haneefdm

grigorig, 2022-05-03:
Maybe there's an easy way to solve this problem. The windows shell resolves from the base name to the wrapper scripts just fine and the `ChildProcess` API of NodeJS has an option to run inside a shell. I did a little bit of testing with the Node REPL and it seems to work fine for me. I assume running in a shell has some overhead, but apart from that I don't see why it shouldn't be possible to always use this setting, at least on Windows.

That way it might also be feasible to get rid of a few special casing I noticed where the extension explicitly uses `.exe` on Windows.

haneefdm, 2022-05-03:
I thought we took care of this with #563. If everything uses the same extension, the current mechanism works. I did this for the author(s) of the xPack

But you seem to be using a unique extension for everything. That I don't want to deal with. Either use the system extension or use the same extension.

grigorig, 2022-05-03:
I'm not using any unique extensions, not sure what you mean... I'm just trying to use the packages installed by xPack as-is, and in a portable way, nothing more. Right now this doesn't really work, you have to add some special configuration on Windows.

Anyway, I just think it would be a better approach to rely on Windows' own mechanism to resolve binary and script names (using the shell as a runner). That way, a lot of configuration isn't needed anymore. Or maybe a simple wrapper for ChildProcess could replicate that shell functionality. As far as I can tell, there are special cases that add `.exe` and the like all over the codebase. These could go away with that.

grigorig, 2022-05-03:
By the way, the shim approach seems to be "as it good as it gets", as far as I can see. NPM seems to be use the same approach as xPack too, and so does Python's pip when installing script launchers. So I'd say it makes a lot of sense to support it.

haneefdm, 2022-05-03:
The xPack author asked for something. I did it and he is/was happy. So, I am not sure what you are asking for. 

And no, I am not adding wrappers for ChildProcess, etc. It is already too complicated. And, I don't know what npm or pip does and I am not going to investigate, unless you point to detailed description.

You are not telling me why what the xPack author did is wrong. It should work and work well. What you did was use the `.gdb` as an extension meaning that you have deviated from what the xPack author was doing. You have an idea that I cannot even see what that is.

You seem to have edited the original comment that I did not see, but I still don't understand.

grigorig, 2022-05-03:
The `.gdb` thing was a typo, sorry, I already fixed it. I'm using the standard file extensions for scripts of course (otherwise it wouldn't work at all).

My issue is with ergonomics and maintenance. Ideally, platform-specific overrides like the `"cortex-debug.gdbPath.windows"` setting shouldn't be needed. And that setting can only be put into VSCode workspace settings, I can't use it in launch configurations. But I can't commit `settings.json` to version control, that will cause a mess. So currently there is no way to ship the project with a configuration that works out of the box for developers working on Windows. At least I don't see how that could work.

Or maybe there's something obvious I'm missing...

haneefdm, 2022-05-03:
> Or maybe there's something obvious I'm missing...

Perhaps. You can put the gdb path in launch.json and use all the mechanism provided in there by VSCode. VSCode also allows you to refer to User/Workspace setting in launch.json. And, launchjson has its own OS specific sections. There are many ways to skin this cat. Also, we prefer to use User settings rather than Workspace settings in many cases to make it more portable between collaborators. YMMV

See also. https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

grigorig, 2022-05-03:
OK, I guess I can live with the extra "windows" section in launch configurations, thanks for the hint!

I would still love it if the extension would be a bit more clever regarding this. Like I said, I think enforcing running the tools through a shell on Windows has potential to significantly code. And that's as simple as adding `shell: true` to `spawn()` parameters.