Marus/cortex-debug #645: Debug STM32H755XX Dual Core

Flo10394, 2022-05-04:
Hi Everyone,
I´m trying to debug my stm32h755zi. It has a cm7 and a cm4 core.
First of all, I am using a NUCLEO-H755ZI.

What works:
1. With an example Project of eclipse (stm32CubeIde 1.8.0), I am able to debug the cm7 core and the cm4 core separately with STLink and everything works fine.
2. I am also able to debug the cm7 core with visual studio code cortex-debugging

**What does not work:**
3. I am **not** able to debug the cm4 core with with visual studio code cortex-debugging

The disassembly tells me, that I am debugging in a flash region of the cm7 core.
CM7 Flash base: 0x08000000
CM4 Flash base: 0x08100000
[image]

My questions:
1. I tried with "numberOfProcessors" and "targetProcessor", but I had no success. Should this tags be used for my problem?
2. Are there any tags like "AccessPort" like in eclipse?
3. Have you any hint for me, how I can get this to work?

Here my eclipse launch configurations:
[image]
[image]

Here my .launch for cm7 core:
`

      "name": "Debug ST-Link",
      "cwd": "${workspaceFolder}",
      "armToolchainPath": "C:/.conan/11291b/1/bin",
      "executable": "./build/bin/rsl30_microengine_cm7.elf",
      "request": "launch",
      "type": "cortex-debug",
      "runToEntryPoint": "main",
      "preLaunchTask": "build cm7",
      "device": "STM32H755ZI",
      "showDevDebugOutput": "none",
      "svdFile": "${workspaceFolder}/src/device/STM32H755_CM7.svd",
      "servertype": "stlink"
`

Here my .launch for cm4 core:
`

      "name": "Debug ST-Link",
      "cwd": "${workspaceFolder}",
      "armToolchainPath": "C:/.conan/11291b/1/bin",
      "executable": "./build/bin/rsl30_microengine_cm4.elf",
      "request": "launch",
      "type": "cortex-debug",
      "runToEntryPoint": "main",
      "preLaunchTask": "build cm4",
      "device": "STM32H755ZI",
      "showDevDebugOutput": "none",
      "svdFile": "${workspaceFolder}/src/device/STM32H755_CM4.svd",
      "servertype": "stlink"
`

haneefdm, 2022-05-04:
I have tried multi-core debug with gdb-servers like OpenOCD, pyOCD and JLink but never with STLink.

> I tried with "numberOfProcessors" and "targetProcessor", but I had no success. Should this tags be used for my problem?

Yes, for other gdb-servers, `numberOfProcessors` should be `2` and `targetProcessor` should be `0` for cm7 and `1` for cm4. Access port you see in Eclipse is the debug access port. It does not matter to us, we use this setting for selecting the right TCP port. STLink (if it is like the others) should have opened two sequential TCP ports for gdb(s) to connect to.

What can help is the output from the STLink server. You will find this in the TERMINAL tab (gdb-server). Could you paste the contents of that output?

I noticed in that both CM4 and CM7 are using the same GDB port. This is normal for use on the command line but it depends on how the stlink server is using this info. Can you click on the `Show Command Line` button and see the differences. They are probably identical?!?! We should compare this to what Cortex-Debug is doing (you can see this in the Debug Console)

[image]

I think there is a way to see the gdb transactions/command-line in Eclipse, we would like to compare those with what we do. This is the critical part. You can see our gdb transactions info in the Debug-Console. In your launch.json, do
```
  "showDevDebugOutput": "raw",
```

Flo10394, 2022-05-04:
Thanks for your reply.

Eclipse"Show Command Line" for CM7:
`C:\ST\STM32CubeIDE_1.8.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.100.202109301221\tools\bin\ST-LINK_gdbserver.exe -p 61234 -l 1 -d -t -s -cp C:\ST\STM32CubeIDE_1.8.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.100.202110141430\tools\bin -m 0 -k`

Eclipse "Show Command Line" for CM4:
`C:\ST\STM32CubeIDE_1.8.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.100.202109301221\tools\bin\ST-LINK_gdbserver.exe -p 61234 -l 1 -d -t -s -cp C:\ST\STM32CubeIDE_1.8.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.100.202110141430\tools\bin -m 3 -k`

Cortex-Debug gdb-server output for CM7:
`"C:\\ST\\STM32CubeIDE_1.8.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.100.202109301221\\tools\\bin\\ST-LINK_gdbserver.exe" -p 50000 -cp "C:\\ST\\STM32CubeIDE_1.8.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.100.202110141430\\tools\\bin" --swd --halt`

Cortex-Debug gdb-server output for CM4:
`"C:\\ST\\STM32CubeIDE_1.8.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.100.202109301221\\tools\\bin\\ST-LINK_gdbserver.exe" -p 50000 -cp "C:\\ST\\STM32CubeIDE_1.8.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.100.202110141430\\tools\\bin" --swd --halt`

I am not really familar to this command line interface. But maybe it´s possible for you, to detect some problems.

`I have tried multi-core debug with gdb-servers like OpenOCD, pyOCD and JLink but never with STLink.`
I am also able to use jlink. Could you provide any example launch config, that i can get this to work with jlink instead of stlink?

haneefdm, 2022-05-04:
I think I found the issue. In the STLink case, `numberOfProcessors` and `targetProcessor` may not be needed. Lets not use that. Instead, we have to specify the debug port to debug. For CM4, this should be `3` according to your screenshots.

I did not see this in their docs before.
```
• -m <apID>, --apid <apID>
Defines the apID to debug.
[Use this option for muti-core.]
```
See https://www.st.com/resource/en/user_manual/um2576-stm32cubeide-stlink-gdb-server-stmicroelectronics.pdf

Can you try the following in launch.json for the cm4 config.
```
    "serverArgs": [
        "-m", "3"
    ],
```
You can leave the cm7 config alone for now.

haneefdm, 2022-05-04:
It is not clear to me how to debug both cores at the same time like we do with other gdb-servers. Based on what we find here, we may have to update the following

https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging

Yes, you can debug both cores at the same time in the same window with other gdb-servers, but for STLink, the stars have to align :-)

Flo10394, 2022-05-04:
First of all, Thank you! I am now able to debug my cm4 code!!

I have some issues, because i think, i have another issue, because now both flash-memories are written. But i will try to resolve on myself.
I will also try to run this with jlink (start 2 gdb-server) and post the results here.

Thank you [haneefdm](https://github.com/haneefdm)!

haneefdm, 2022-05-04:
FYI: If you do the following, nothing will be uploaded to flash
```
"loadFiles": [
]
```
You can also list multiple files. The `executable` will only be used for debug symbols and is still needed.

I will leave this issue open for a little while. Please report back with what you may have learnt so I can update the Wiki.

Flo10394, 2022-05-04:
This did the job for stlink, if both .elf-files are entered
`"loadFiles": [
]`

For stlink, i have this launch.json for cm7 (the -m argument and loadFiles did the job)

    {
      "name": "Debug ST-Link",
      "cwd": "${workspaceFolder}",
      "armToolchainPath": "C:/.conan/11291b/1/bin",
      "executable": "./build/bin/rsl30_microengine_cm7.elf",
       "loadFiles": [
         "./build/bin/rsl30_microengine_cm7.elf",
         "./../cm4/build/bin/rsl30_microengine_cm4.elf"
       ],
      "request": "launch",
      "type": "cortex-debug",
      "runToEntryPoint": "main",
      "device": "STM32H745ZI",
      "showDevDebugOutput": "raw",
      "svdFile": "${workspaceFolder}/src/device/STM32H745_CM7.svd",
      "servertype": "stlink",
      "serverArgs": [
        "-m", "0"
      ],
    },

For jlink, i have these two launch.json

For CM7:

    {
      "name": "Debug jlink",
      "cwd": "${workspaceRoot}",
      "armToolchainPath": "C:/.conan/11291b/1/bin",
      "executable": "./build/bin/rsl30_microengine_cm7.elf",
      "loadFiles": [
        "./build/bin/rsl30_microengine_cm7.elf",
        // "./../cm4/build/bin/rsl30_microengine_cm4.elf" uncomment to flash cm4 firmware too (must be build separately)
      ],
      "request": "launch",
      "type": "cortex-debug",
      "runToEntryPoint": "main",
      "preLaunchTask": "build cm7",
      "device": "STM32H745ZI_M7",
      "showDevDebugOutput": "raw",
      "servertype": "jlink",
      "serverpath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
      "interface": "swd",
      "swoConfig":{
        "enabled":true,
        "source":"probe",
        "swoFrequency": 2000000,
        "cpuFrequency":480000000,
        "decoders": [
            {
                "port": 0,
                "type": "console",
                "label": "SWO output",
                "encoding":"ascii"
            }
        ]
      }
    },

For CM4:

    {
      "name": "Debug jlink",
      "cwd": "${workspaceRoot}",
      "armToolchainPath": "C:/.conan/11291b/1/bin",
      "executable": "./build/bin/rsl30_microengine_cm4.elf",
      "loadFiles": [
        "./build/bin/rsl30_microengine_cm4.elf",
        // "./../cm7/build/bin/rsl30_microengine_cm7.elf" uncomment to flash cm7 firmware too (must be build separately)
      ],
      "request": "launch",
      "type": "cortex-debug",
      "runToEntryPoint": "main",
      "preLaunchTask": "build cm4",
      "device": "STM32H745ZI_M4",
      "showDevDebugOutput": "raw",
      "servertype": "jlink",
      "serverpath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServerCL.exe",
      "interface": "swd",
      "swoConfig":{
        "enabled":true,
        "source":"probe",
        "swoFrequency": 2000000,
        "cpuFrequency":480000000,
        "decoders": [
            {
                "port": 0,
                "type": "console",
                "label": "SWO output",
                "encoding":"ascii"
            }
        ]
      }
    },

Thanks again to [haneefdm](https://github.com/haneefdm)

closed 2022-05-04 by Flo10394

haneefdm, 2022-09-12:
@dookei It is not the project that matters, what matters is which gdb-server you are using. OpenOCD, vs JLink vs STLink, they require totally different setups. You want the launch.json files for the projects and I see some posted above -- they may be incomplete or have things like SWO that you do not need

JLink will probably be the simplest, followed bymOpenOCD. STLink does not look that bad either.