Marus/cortex-debug #646: SWO support is not being fully enabled in pyOCD

grigorig, 2022-05-05:
When I try to enable SWO console output with pyOCD, I don't get any output. My configurations looks like this:

```
"swoConfig": {
    "enabled": true,
    "cpuFrequency": 32000000,
    "swoFrequency": 2000000,
    "source": "probe",
    "decoders": [
        {
            "type": "console",
            "port": 0,
            "label": "Output",
            "encoding": "ascii",
            "timestamp": true
        }
    ]
},
```
When I add `-Oenable_swv=1` to `serverArgs`, it starts working. I think this should be added when SWO is configured.

grigorig, 2022-05-05:
Yes, it seems to work both ways. No idea if it changed at some point, the flag was [introduced with the initial SWO support](https://github.com/pyocd/pyOCD/pull/543) in pyOCD. It sounds like it's supposed to be needed to enable SWO.

haneefdm, 2022-05-05:
Yes it was there originally, but when tested it, it did not require it. Now it looks like we need both `enable_swv` and `swv_raw_enable`

haneefdm, 2022-05-05:
I think we have a problem...
[image]

They are initialiing a bunch of stuff and so are we. These are the GDB macros that are executed
```
        const ratio = Math.floor(cpuFrequency / swoFrequency) - 1;
        const commands: string[] = [
            'EnableITMAccess',
            `BaseSWOSetup ${ratio}`,
            'SetITMId 1',
            'ITMDWTTransferEnable',
            'DisableITMPorts 0xFFFFFFFF',
            `EnableITMPorts ${portMask}`,
            'EnableDWTSync',
            'ITMSyncEnable',
            'ITMGlobalEnable'
        ];
```
This stuff predates me.

These macros are in the gdbsupport.init file in <extension-install-dir>/support directory. I don't think the first four lines are needed because it will trample on what pyOCD is doing. In fact, maybe none of it is needed except for the portmasks. You can probably edit that file in the install dir. and experiment. Those macros also use hard-coded addresses for TPIU/DWTetc. and with more recent MCU, they are no longer at ARM specified default addresses for some devices.

You can make a copy of the file and load it in your `preLaunchCommands` and it will override the original macros.

All of this is you have the time or inclination to help. It will be much appreciated.

grigorig, 2022-05-05:
Hm, yeah I noticed that TPIU/ITM configuration is a bit of a problem.

I guess in an ideal world, the debug server would know all device specific details and offer some high-level interface to configure TPIU/ITM, right? I wonder if it's worth the effort to work towards that.

haneefdm, 2022-05-13:
The original issue should be fixed in 1.5.0

closed 2022-05-13 by haneefdm