Marus/cortex-debug #647: Serial SWO source with OpenOCD

vankxr, 2022-05-09:
In a previous version, the property `source` for SWO was meant to be set to the device path for the serial port (e.g. `/dev/ttyUSB1`). In commit https://github.com/Marus/cortex-debug/commit/46b671656d6823bf07b72497b1c64eeaf79e5d9a, it seems to have changed to `serial` instead, but the serial port is still instantiated with the device set to the `source` property:
https://github.com/Marus/cortex-debug/blob/2798b220893cf8d5b9fb8c11876742c6274b16b4/src/openocd.ts#L250
This prevents it from working correctly, obviously.

haneefdm, 2022-05-09:
I had that wrong. Even the `external` uses `source` for the device. Makes no sense. I think I will change both to `swoPath` as you suggested. It was probably never used/reported and it would never have worked

vankxr, 2022-05-09:
Yes, I am aware of that script, infact, I remember reporting about the binary modules issue back then, before the script was born :)

Speaking of that, the API of `serialport` changed in version 10.x.x, and since that shell script does not install a specific version, it is crashing. For now I editted the shell script and manually force it to install version 9.2.8 of serialport. You might want to take a look.

Currently cortex-debug uses:
https://github.com/Marus/cortex-debug/blob/a5d4f5e91d0867a41647211e5cf3fbc5f3f69b5f/src/frontend/swo/sources/serial.ts#L33

`serialport` v10.x.x+ should be:
`const { SerialPort } = require('serialport')`

https://serialport.io/docs/9.x.x/guide-usage
https://serialport.io/docs/guide-usage

commit a4c31de2 (2022-05-09): updated missing/wrong props/descr. in package.json, fixed #647