Marus/cortex-debug #649: Debugger tries to read registers after having started the CPU.

s13n, 2022-05-12:
I am having difficulties with debugging, because the debugger seems to trip over its own feet. After flashing the code, it starts the CPU at the wrong point in time and then fails reading the registers. See the log below.

My launch configuration is the following:
```
        { // Run a local GDB server that connects to the SEGGER remote server on the host
            "name": "Debug (SEGGER remote)",
            "type": "cortex-debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "servertype": "jlink",  
            "interface": "swd",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",  
            "ipAddress": "192.168.1.71",
            "device": "STM32H745XI_M7",
            "svdFile": "${command:cmake.launchTargetDirectory}/../_deps/pelua-src/stm32/hw/STM32H757_CM7.svd",
            "runToEntryPoint": "ResetISR",
            "demangle": true,
            "showDevDebugOutput": true
        },
```

As you can see, the Segger GDB Server is used, and it talks to a Segger probe via the network.

GDB Server output is:
```
Waiting for gdb server to start...[2022-05-12T15:53:06.689Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/opt/SEGGER/JLink/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32H745XI_M7 -select ip=192.168.1.71
SEGGER J-Link GDB Server V7.52d Command Line Version

JLinkARM.dll V7.52d (DLL compiled Aug 17 2021 17:14:32)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32H745XI_M7 -select ip=192.168.1.71
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         IP
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32H745XI_M7
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V10 compiled Nov  2 2021 12:14:50
Hardware: V10.00
S/N: 600000275
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.32 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 2 bytes @ address 0x08042648 (Data = 0xB508)
Read 2 bytes @ address 0x0807303C (Data = 0x4770)
Read 4 bytes @ address 0x08000518 (Data = 0x4D1F4A1E)
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x08000518)
Received monitor command: reset
Resetting target
Downloading 16032 bytes @ address 0x08000000
Downloading 16240 bytes @ address 0x08003EA0
Downloading 16112 bytes @ address 0x08007E10
Downloading 16096 bytes @ address 0x0800BD00
Downloading 16096 bytes @ address 0x0800FBE0
Downloading 16176 bytes @ address 0x08013AC0
Downloading 16176 bytes @ address 0x080179F0
Downloading 16208 bytes @ address 0x0801B920
Downloading 16160 bytes @ address 0x0801F870
Downloading 16208 bytes @ address 0x08023790
Downloading 16192 bytes @ address 0x080276E0
Downloading 16192 bytes @ address 0x0802B620
Downloading 16208 bytes @ address 0x0802F560
Downloading 16208 bytes @ address 0x080334B0
Downloading 16112 bytes @ address 0x08037400
Downloading 16112 bytes @ address 0x0803B2F0
Downloading 16096 bytes @ address 0x0803F1E0
Downloading 15984 bytes @ address 0x080430C0
Downloading 16080 bytes @ address 0x08046F30
Downloading 16096 bytes @ address 0x0804AE00
Downloading 15968 bytes @ address 0x0804ECE0
Downloading 16048 bytes @ address 0x08052B40
Downloading 16032 bytes @ address 0x080569F0
Downloading 16080 bytes @ address 0x0805A890
Downloading 16000 bytes @ address 0x0805E760
Downloading 16064 bytes @ address 0x080625E0
Downloading 16128 bytes @ address 0x080664A0
Downloading 16096 bytes @ address 0x0806A3A0
Downloading 16112 bytes @ address 0x0806E280
Downloading 16096 bytes @ address 0x08072170
Downloading 16080 bytes @ address 0x08076050
Downloading 16128 bytes @ address 0x08079F20
Downloading 16112 bytes @ address 0x0807DE20
Downloading 16048 bytes @ address 0x08081D10
Downloading 16032 bytes @ address 0x08085BC0
Downloading 10348 bytes @ address 0x08089A60
Downloading 16100 bytes @ address 0x0808C2CC
Downloading 9804 bytes @ address 0x080901B0
Downloading 13824 bytes @ address 0x080927FC
Downloading 16272 bytes @ address 0x08095E00
Downloading 16240 bytes @ address 0x08099D90
Downloading 16320 bytes @ address 0x0809DD00
Downloading 16288 bytes @ address 0x080A1CC0
Downloading 16224 bytes @ address 0x080A5C60
Downloading 8264 bytes @ address 0x080A9BC0
Downloading 3092 bytes @ address 0x080ABC08
Read 2 bytes @ address 0x08042648 (Data = 0xB508)
Read 2 bytes @ address 0x0807303C (Data = 0x4770)
Received monitor command: reset
Comparing flash   [....................] Done.
Resetting target
Read 4 bytes @ address 0x08000518 (Data = 0x4D1F4A1E)
Reading 64 bytes @ address 0x08007840
Read 4 bytes @ address 0x08007B80 (Data = 0x080A27C0)
Read 2 bytes @ address 0x08007A30 (Data = 0xF240)
Read 2 bytes @ address 0x08000518 (Data = 0x4A1E)
Read 4 bytes @ address 0x08000518 (Data = 0x4D1F4A1E)
Setting breakpoint @ address 0x08007A30, Size = 2, BPHandle = 0x0001
Performing single step...
...Target halted (Vector catch, PC = 0x0800051A)
Reading all registers
Setting breakpoint @ address 0x08000518, Size = 2, BPHandle = 0x0002
Starting target CPU...
ERROR: Cannot read register 15 (R15) while CPU is running
Reading all registers
ERROR: Cannot read register 0 (R0) while CPU is running
ERROR: Cannot read register 1 (R1) while CPU is running
ERROR: Cannot read register 2 (R2) while CPU is running
ERROR: Cannot read register 3 (R3) while CPU is running
ERROR: Cannot read register 4 (R4) while CPU is running
ERROR: Cannot read register 5 (R5) while CPU is running
ERROR: Cannot read register 6 (R6) while CPU is running
ERROR: Cannot read register 7 (R7) while CPU is running
...
```

haneefdm, 2022-05-12:
It would  help to know which version of Cortex-Debug you are using and if you have tried the latest. Also, the output in the Debug Console. Note that `showDevDebugOutput` format has changed with more options....

Even if we did read the registers while CPU is running, we do that with GDB and never directly with JLink. But generally GDB does not request registers if it knows the CPU is busy. That is a negotiation between GDB and JLink-gdb-server.

s13n, 2022-05-13:
The version of cortex-debug is 1.4.4. Unfortunately I haven't access to the target hardware at the moment, so I can't help with the DebugConsole output right now.

In general, I had the impression from working with cortex-debug, that there might be stale actions involved which cause requests to the target at inappropriate times, and for data that's not needed. I had recently an instance where I had opened a memory view window showing a large amount of memory, and even after I had closed it again, the memory content would still be requested from the target, slowing everything down considerably.

I can try to recreate this, too, when I have access to the hardware again. I don't know how cortex-debug interacts with GDB to determine what needs to be read from the target, so I can't say which side is responsible.

haneefdm, 2022-05-18:
Would you be willing to try out the following

https://github.com/Marus/cortex-debug/releases/tag/v1.5.1-pre4

There are a lot of things that occur in VSCode asynchronously. Like requests for memory/register/stack/variables/watch reads coming in when the target just did a continue. The whole thing is asynchronous and is impossible to do otherwise. It can happen even with simple/fast step button presses.

Things are distributed across 6 different Unix/Windows processes and nearly impossible to have the state exactly the same in all of them. This is not our design. This is how VSCode is designed. VSCode is the one that makes requests and pumps them furiously.

s13n, 2022-05-18:
I won't be able to check this soon, unfortunately. I'm heading into a two-week holiday, and thereafter there's a conference I have to attend, so the earliest chance is going to be in more than 3 weeks. Please don't wait for me with anything you're doing. I will come back to this issue when the "distractions" are gone.

peterbone-bo, 2022-06-01:
I have the same issue.

**Software**

Ubuntu 22.04
VS Code 1.67.2
Cortex-debug 1.5.0 (also tried switching to pre-release)
J-Link GDB server 7.60f

**Hardware**

Intel Core i9 PC
J-Link PLUS compact
Arm Cortex M33 based board

**Log output**

```
JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device cortex-m33
SEGGER J-Link GDB Server V7.60f Command Line Version

JLinkARM.dll V7.60f (DLL compiled Jan 25 2022 16:40:59)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device cortex-m33
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 cortex-m33
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled May  2 2022 08:59:57
Hardware: V11.00
S/N: 851003949
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 1.82 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x08216C06 (Data = 0xBDF0B003)
Reading 64 bytes @ address 0x0421F4C0
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x08216C06)
Received monitor command: reset
ERROR: Failed to halt CPU.
Resetting target
Downloading 16352 bytes @ address 0x06000000
Downloading 16352 bytes @ address 0x06003FE0
Downloading 16352 bytes @ address 0x06007FC0
[... 433 lines cut ...]
```

**Debug console output**

```
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /xxx.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /xxx.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /xxx.elf
    Set "showDevDebugOutput": "raw" in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device cortex-m33
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
Reading symbols from /xxx.elf...
Finished reading symbols from objdump: Time: 200 ms
check_power_off_and_reboot () at /xxx.c:195
195	            APPS_LOG_MSGID_I(UI_SHELL_IDLE_BT_CONN_ACTIVITY", check_and_do_power_off reboot ret = %d", 1, ret);
Program stopped, probably due to a reset and/or halt issued by debugger
Resetting target
Finished reading symbols from nm: Time: 5271 ms
Resetting target

Program
 received signal SIGTRAP, Trace/breakpoint trap.
0xdeadbeee in ?? ()
warning: value truncated
```

haneefdm, 2022-06-01:
Let us latest build we got. We have moved on even since the build I mentioned above. Download the vsix asset/file mentioned from the latest.

https://github.com/Marus/cortex-debug/releases

What you are seeing may have nothing to do with register reading at all. We may be conflating things. The following is not an error
```
Program
 received signal SIGTRAP, Trace/breakpoint trap.
0xdeadbeee in ?? ()
warning: value truncated
```
That is what GDB prints whenever a halt occurs (depending on how the Jlink/OpenOCD SW reports). The PC `0xdeadbeee` does not seem valid which may be a problem in your firmware. If your PC or SP are wrong when processor is halted, then not much works after that.

peterbone-bo, 2022-06-02:
It also seems like something is this sequence is messed up:

```
Starting target CPU...
ERROR: CPU is not halted
ERROR: Cannot read register 15 (R15) while CPU is running
Reading all registers
ERROR: Cannot read register 0 (R0) while CPU is running
ERROR: Cannot read register 1 (R1) while CPU is running
ERROR: Cannot read register 2 (R2) while CPU is running
ERROR: Cannot read register 3 (R3) while CPU is running
ERROR: Cannot read register 4 (R4) while CPU is running
```

Why is it trying to start the CPU first, and then read some registers that it cannot read while running? Is this intentional that this might occur?

haneefdm, 2022-06-02:
This is because there are a dozen or so async event source and sinks. It is far too complex to explain for me. But I will try

* Processor halts for whatever reason, gdb-server notifies gdb
* gdb notifies cortex-debug backend, but also reads registers to determine the location of PC/SP. This is a case where gdb makes a request without us asking
* cortex-debug backend notifies VSCode
* There are a dozen+ windows all get active and start sending requests to update themselves. No window/process knows what the other is doing, some requests even come far too late
* While the above is happening, the program continues -- either due to runToEntryPoint, user action to step/continue, etc.
* **But some requests were already in the queue (in flight)**
* Now, gdb could have rejected the request itself as it does for other requests (like memory reads, variable reads, etc.) But it forwards the requests from Registers Window to gdb-server anyway. We can eliminate the Registers window and this particular problem goes away but not entirely. There are other things wanting to read registers/memory
* Most gdb-servers simply notify gdb that the request has failed, while JLink 

Because of the buffering and queueing, things in flight cannot be stopped/withdrawn but everyone making requests expect failures and adjust accordingly. There are a dozen different SW modules, processes all have to get notified of the current state of things (running vs stopped) and it is not possible for that to be atomic.

An alternative is to make everything synch. eliminate all queues (meaning very slow with tones of dead time in between requests) but we cannot change how VSCode makes requests. It does not have a queue and one window does not know what the other is doing. VSCode is super responsive with this programming model. VSCode even makes duplicate requests back to back.

Note: The sequence of requests/responses are entirely different between attach and launch. That is the nature of the request. You can see that for yourself in the Debug Console (enable debug as instructed in the Debug Console)

Now, my question to all of you in this thread. Has anyone tried the latest pre-release (1.5.1) where I made some changes to hopefully reduce this (not eliminate)? It is now available from the marketplace. No one has responded my request to try even private builds. **If no one wants to work with me**, I and going to shut this conversation down and I will not respond anymore.

My biggest issue is that this does not happen to me anymore and even before that, it was intermittent. Mine looks like this.
```
...
Read 2 bytes @ address 0x08000D9A (Data = 0xD034)
Setting breakpoint @ address 0x080004F6, Size = 2, BPHandle = 0x0001
Starting target CPU...
...Breakpoint reached @ address 0x080004F6
Reading all registers
Removing breakpoint @ address 0x080004F6, Size = 2
Read 4 bytes @ address 0x080004F6 (Data = 0x607B2300)
Reading 64 bytes @ address 0x2002FFC0
Reading register (MSP = 0x2002FFF0)
Reading register (PSP = 0x       0)
Reading register (PRIMASK = 0x       0)
Reading register (BASEPRI = 0x       0)
Reading register (FAULTMASK = 0x       0)
Reading register (CONTROL = 0x       0)
Reading register (FPSCR = 0x       0)
...
```

You can also fix this yourself if you so desire and submit a PR.

peterbone-bo, 2022-06-03:
Thanks a lot for your very detailed explanation, and for putting time into trying to fix this.

I think i will need more time to narrow down the exact cause of this, and why it is only happening when doing `launch` and not `attach`. If there is anything i can test or try for you, please let me know.

Thanks in advance, and keep up the good work @haneefdm.

haneefdm, 2022-06-03:
Please use the pre-release and let me know if this is still happening. You can enable this in VSCode Extensions panel. No one wants to try the potential fix and give feedback

[image]

Again, I repeat that this is a harmless issue. If a register read fails, we simply don't use it, we wait for the next opportunity to read. What is your real issue and does it have anything to do with registers?

s13n, 2022-06-15:
I now have finally had the opportunity to recheck with the newest prerelease. I still see register reads failing, but it doesn't seem to cause problems, so I accept this as not being a bug.

However, there is still the behavior I mentioned in my second message above:

I opened a memory view window, and after closing the window, the memory is still being read from the target whenever I step through the code. This can be hindrance especially when the memory view covered a large memory area. It seems that when closing the window, somebody forgets to tell gdb that the memory contents aren't needed anymore, and it keeps reading it each time it regains control over the target.

But that may be a different issue altogether.

haneefdm, 2022-06-15:
About memory windows, that is a totally different thing. This is VSCode doing it and it manages the window -- we only provide content when asked. It appears that although we closed the window, we are never notified, and VSCode keeps it in the list of active windows. This was written many years ago, never touched. I will see if there is anything in VSCode APIs to see if it to possible to detect a closed window -- has to be

This is why I don't like mixing different issues up.

haneefdm, 2022-06-15:
Looked at it. VSCode "truly" destroys windows in a lazy method and unless we have a custom editor, the memory windows making requests cannot be avoided.

closed 2022-06-15 by haneefdm

s13n, 2022-06-16:
Thanks for looking into this. It is a VSCode "feature" then, but I think they should destroy windows somewhat more eagerly, as the effect can be rather annoying.

s13n, 2022-06-16:
Just a thought: Would it be possible to determine whether a window is visible, and make its requests dependent on that? This wouldn't just apply to a memory view window, it is equally futile to request the register contents when the register window isn't visible - only that the amount of data requested is much more benign here.

haneefdm, 2022-06-16:
We thought so. Of course, we were doing this. when debugger pausess
```typescript
        vscode.workspace.textDocuments.filter((td) => td.fileName.endsWith('.cdmem')).forEach((doc) => {
            if (!doc.isClosed) {
                this.memoryProvider.update(doc);
            }
        });
```
Doesn't work. Not only is the document still in in the list, is also marked as not closed. Even though we closed it in the UI. Sometime later, it gets closed. The timing was indeterminate.

s13n, 2022-06-16:
Hmm, that doesn't look like a dependable scheme to rely upon. :-)

Thanks again for your efforts, and for cortex-debug in general!