Marus/cortex-debug #650: Failed to launch OpenOCD GDB Server: Error: spawn /usr/local/bin EACCES

VBruceHunt, 2022-05-12:
Issue Type: <b>Bug</b>

**1. Environment:** 

1. Raspberry Pi 4 up-to-date Raspberry Pi OS with development essentials package installed.
2. installed Raspberry Pi Pico SDK for C/C++
3. installed Raspberry Pi version of OpenOCD
4. installed Raspberry Pi Picoprobe.
5. Pi 4 connected via USB to Pico with Picoprobe software loaded.
6. Picoprobe connected to target Raspberry Pi Pico via SWD interface
7. Target Application compiled  on Pi4 using GCC 8.3.1 arm-none-eabi using VScode IDE 1.66.2 
8. VScode click on "Run and Debug" icon.
9. Click on Pico Debug to start up and debug the application.
10. Summary:  Compiled and ran debugger many times up until mid April developing myproject. I am _fairly new_ at using vscode so I've worked to isolate the problem to VScode as the following shows.  I can run each of the individual steps ( gdb-multiarch and openocd ) so the components work fine manually using the (displayed) parameters that VScode used according to the DEBUG CONSOLE.

**2.The following shows the suspected error output:**

_VSCode Debug Console Output:_

Reading symbols from objdump-multiarch --syms -C -h -w /home/myself/projects/myproject/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf
Reading symbols from nm-multiarch --defined-only -S -l -C -p /home/myself/projects/myproject/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf
Launching GDB: gdb-multiarch -q --interpreter=mi2 /home/myself/projects/myproject/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.elf
1-gdb-set target-async on
Launching gdb-server: /usr/local/bin -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s ~/pico/openocd/tcl -f /home/myself/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
    Please check TERMINAL tab (gdb-server) for output from /usr/local/bin
Finished reading symbols from objdump: Time: 257 ms
Finished reading symbols from nm: Time: 238 ms
**_Failed to launch OpenOCD GDB Server: Error: spawn /usr/local/bin EACCES_**
----------------------------------------------------------End -------------------------------------

_VSCode Terminal Output::_

[2022-05-10T23:58:00.145Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/usr/local/bin -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s ~/pico/openocd/tcl -f /home/myself/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
[2022-05-10T23:58:01.216Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
----------------------------------------------------End -------------------------------------------

_The file in ~/projects/myproject/.vscode/launch.json:_

{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Pico Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            // This may need to be arm-none-eabi-gdb depending on your system
            "gdbPath" : "gdb-multiarch", // was arm-none-eabi-gdb
            "device": "RP2040",
            "configFiles": [
                "interface/picoprobe.cfg",
                "target/rp2040.cfg"
            ],
            "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
            "runToMain": true,
            // Work around for stopping at main on restart
            "postRestartCommands": [
                "break main",
                "continue"
            ],
            "searchDir": ["~/pico/openocd/tcl"],
            "showDevDebugOutput": "both",
        }
    ]
}
----------------------------------------------------End -------------------------------------------
[... 170 lines cut ...]
vswsl492cf:30256860
pythonvspyl392:30443607
pythontb:30283811
pythonptprofiler:30281270
vshan820:30294714
vstes263:30335439
vscorecescf:30445987
pythondataviewer:30285071
vscod805:30301674
pythonvspyt200:30340761
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
vsaa593cf:30376535
vsc1dst:30438360
pythonvs932:30410667
wslgetstarted:30449410
pythonvsnew555:30457759
vscscmwlcmt:30465135
cppdebugcf:30475217
vsclangdf:30486550

```

</details>

haneefdm, 2022-05-13:
your OpenOCDPath is incomplete...pointing to a directory rather than an executable

```
"cortex-debug.openocdPath": "/usr/local/bin",
```

Which resulted in this a bad launch
```
Launching gdb-server: /usr/local/bin -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s ~/pico/openocd/tcl -f /home/myself/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/picoprobe.cfg -f target/rp2040.cfg
```

Change it to 
```
"cortex-debug.openocdPath": "/usr/local/bin/openocd",
```

[image]

closed 2022-05-13 by haneefdm

VBruceHunt, 2022-05-13:
Thank you for the assistance and quick response.  I was confused by the overloaded term "Path".  Sometimes it means the pathways only excluding the endpoint(i.e. just the directory sequence leading to the endpoint) and sometimes (like this time) it includes the endpoint.  Perhaps the term "path" could be reserved for the pathway to the endpoint and a new term such as "pathend" could be introduced to be the "path" including the endpoint. Note that the sentence "Path to the OpenOCD GDB Server executable." is ambiguous and in light of the following sentence it seems logical that the OpenOCD path would follow the same pattern as the system path.  Clarity in this documentation would have saved time for both of us.  Again my thanks for your quick observation and response.