Marus/cortex-debug #651: Debugger did not start from main - using st-util & arm-none-eabi-gdb

Benasking7124, 2022-05-13:
Hi, I'm trying to use VSCode along with this extension, st-util, and arm-none-eabi-gdb to debug my stm32 board. 

But **every time I start the debug sesion, it won't stop at "main".** Instead, it will run for a random amount of cycles then pause. 

It will ignore any break point before while(1) loop. And for the break point inside the while(1) function, it will not stop but not at the first iteration. Therfore, I assume it stopped after running for a while.

the arm-none-eabi-gdb version is 11.2-2022.02 (GNU gdb (Linaro_GDB-2019.12) 11.2.90.20220202-git)
The arm gnu toolchain I use is installed from source following the release note instruction.
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads

the st-util version is v1.7.0-186-gc4762e6
https://github.com/stlink-org/stlink

I'm on Ubuntu 20.04

And below is my settiong.json:
```
{
    "makefile.extensionOutputFolder": "./.vscode",
    "files.associations": {
        "main.h": "c"
    },
    "cortex-debug.gdbPath": "/home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-gdb",
    "cortex-debug.stutilPath": "/usr/local/bin/st-util"
}
```

And below is my launch.json:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ST-Link Debug",
            "cwd": "${workspaceFolder}",
            "executable": "./build/CPP_Test.elf",
            "device": "STM32F407VE",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "stutil",
            "interface": "swd"
        }
    ]
}
```

And below is my main function:
```
int main(void)
{
  int n = 0;

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* Configure the system clock */
  SystemClock_Config();

  MX_GPIO_Init();
  MX_USART2_UART_Init();

  while (1)
  {
    n++;
  }
}
```

And below is my debug console output:
```
Reading symbols from /home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-objdump --syms -C -h -w /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf
Reading symbols from /home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf
Launching GDB: /home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf
    Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Launching gdb-server: /usr/local/bin/st-util -p 50000 --no-reset
    Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/st-util
Finished reading symbols from objdump: Time: 35 ms
Finished reading symbols from nm: Time: 41 ms
Reading symbols from /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.

0x08004250 in main () at Core/Src/main.cpp:98
98	    n++;
Program stopped, probably due to a reset and/or halt issued by debugger
Note: automatically using hardware breakpoints for read-only addresses.
Run to 'main' timed out. Trying to pause program
```

Thank you for making such gread extension! And Thank you in advance for helping me solve this issue ~~

haneefdm, 2022-05-13:
Please follow the instructions printed for advanced information and report back with details. We don't use st-util that much but maybe you/me/others can see what is wrong.
```
    Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
    Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/st-util
```
Most of the time, the issue is with a reset+halt not happening properly.

Benasking7124, 2022-05-14:
Here is the gdb-server terminal output:

```
Waiting for gdb server to start...[2022-05-14T02:39:29.243Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/usr/local/bin/st-util -p 50000 --no-reset
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32F4x5_F4x7
# Reference Manual: RM0090           // RM0090 (Rev. 2)
#
chip_id 0x413
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x4000
sram_size 0x30000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x40023c14
option_size 0x4
flags 2

st-util 1.7.0-186-gc4762e6
2022-05-14T10:39:29 INFO common.c: STM32F4x5_F4x7: 192 KiB SRAM, 512 KiB flash in at least 16 KiB pages.
2022-05-14T10:39:29 INFO gdb-server.c: Listening at *:50000...

detected chip_id parametres

# Device Type: STM32F4x5_F4x7
# Reference Manual: RM0090           // RM0090 (Rev. 2)
#
chip_id 0x413
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x4000
sram_size 0x30000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x40023c14
option_size 0x4
flags 2

2022-05-14T10:39:29 INFO common.c: STM32F4x5_F4x7: 192 KiB SRAM, 512 KiB flash in at least 16 KiB pages.
2022-05-14T10:39:29 INFO gdb-server.c: Found 6 hw breakpoint registers
2022-05-14T10:39:29 INFO gdb-server.c: GDB connected.
2022-05-14T10:39:29 WARN common.c: NRST is not connected
2022-05-14T10:39:29 INFO gdb-server.c: Found 6 hw breakpoint registers

detected chip_id parametres

# Device Type: STM32F4x5_F4x7
# Reference Manual: RM0090           // RM0090 (Rev. 2)
#
chip_id 0x413
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x4000
sram_size 0x30000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x40023c14
option_size 0x4
flags 2

2022-05-14T10:39:29 INFO common.c: STM32F4x5_F4x7: 192 KiB SRAM, 512 KiB flash in at least 16 KiB pages.
2022-05-14T10:39:29 INFO gdb-server.c: flash_erase: block 08000000 -> 8000
2022-05-14T10:39:29 INFO gdb-server.c: flash_erase: page 08000000
EraseFlash - Sector:0x0 Size:0x4000 2022-05-14T10:39:29 INFO gdb-server.c: flash_erase: page 08004000
EraseFlash - Sector:0x1 Size:0x4000 2022-05-14T10:39:30 INFO flashloader.c: Starting Flash write for F2/F4/F7/L4
[... 103 lines cut ...]
98	    n++;
GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"stopped","output":[["reason","signal-received"],["signal-name","SIGTRAP"],["signal-meaning","Trace/breakpoint trap"],["frame",[["addr","0x08004252"],["func","main"],["args",[]],["file","Core/Src/main.cpp"],["fullname","/home/ben/stm32_ws/CPP_Test/Core/Src/main.cpp"],["line","98"],["arch","armv7e-m"]]],["thread-id","1"],["stopped-threads","all"]]}]}
mi2.status = stopped
16-thread-list-ids
GDB -> App: {"output":"","token":16,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["thread-ids",[["thread-id","1"]]],["current-thread-id","1"],["number-of-threads","1"]]}}
17-thread-info 1
GDB -> App: {"output":"","token":17,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["threads",[[["id","1"],["target-id","Thread <main>"],["frame",[["level","0"],["addr","0x08004252"],["func","main"],["args",[]],["file","Core/Src/main.cpp"],["fullname","/home/ben/stm32_ws/CPP_Test/Core/Src/main.cpp"],["line","98"],["arch","armv7e-m"]]],["state","stopped"]]]]]}}
18-data-list-register-names
GDB -> App: {"output":"","token":18,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-names",["r0","r1","r2","r3","r4","r5","r6","r7","r8","r9","r10","r11","r12","sp","lr","pc","","","","","","","","","","xpsr","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","msp","psp","control","faultmask","basepri","primask","s0","s1","s2","s3","s4","s5","s6","s7","s8","s9","s10","s11","s12","s13","s14","s15","s16","s17","s18","s19","s20","s21","s22","s23","s24","s25","s26","s27","s28","s29","s30","s31","fpscr"]]]}}
19-stack-info-depth --thread 1 10000
GDB -> App: {"output":"","token":19,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["depth","1"]]}}
20-stack-list-frames --thread 1 0 0
GDB -> App: {"output":"","token":20,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x08004252"],["func","main"],["file","Core/Src/main.cpp"],["fullname","/home/ben/stm32_ws/CPP_Test/Core/Src/main.cpp"],["line","98"],["arch","armv7e-m"]]]]]]}}
21-data-list-register-values N
GDB -> App: {"output":"","token":21,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["register-values",[[["number","0"],["value","0"]],[["number","1"],["value","100"]],[["number","2"],["value","32"]],[["number","3"],["value","129930697"]],[["number","4"],["value","536873196"]],[["number","5"],["value","0"]],[["number","6"],["value","0"]],[["number","7"],["value","537001968"]],[["number","8"],["value","0"]],[["number","9"],["value","0"]],[["number","10"],["value","0"]],[["number","11"],["value","0"]],[["number","12"],["value","-520093696"]],[["number","13"],["value","0x2001fff0"]],[["number","14"],["value","134230335"]],[["number","15"],["value","0x8004252 <main()+30>"]],[["number","25"],["value","16777216"]],[["number","91"],["value","0x2001fff0"]],[["number","92"],["value","0x0"]],[["number","93"],["value","0 '\\000'"]],[["number","94"],["value [... cut ...]
GDB -> App: {"output":"","token":23,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["variables",[[["name","n"],["type","int"],["value","129930696"]]]]]}}
24-var-update --thread 1 --frame 0 --all-values var_n_256
GDB -> App: {"output":"","token":24,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
25-var-create --thread 1 --frame 0 var_n_256 * "n"
GDB -> App: {"output":"","token":25,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","var_n_256"],["numchild","0"],["value","129930696"],["type","int"],["thread-id","1"],["has_more","0"]]}}

```

haneefdm, 2022-05-14:
Thank you, that was helpful. Here is the problem...
```
Run to 'main' timed out. Trying to pause program
```
We waited for 20 seconds for the program to reach main. It didn't so we paused it. This implies that the program did not halt after a second reset that was done after programming. It takes a bit of learning you can see each command sent to gdb and the response. Setting "showDevDebugOutput" to "raw" is a bit more compact and has same info, easier to read. Please try that.

In the newer release (1.5.x) thugh, we wait forever. It is in pre-release but you can download it in VSCode. You can always switch back to production release

haneefdm, 2022-05-14:
From the gdb-server, I see
```
2022-05-14T10:39:30 WARN common.c: NRST is not connected
```
Okay, then not sure a reset is actually doing what it is supposed to do. It is supposed Halt at the Reset Vector. Do you have another way to do this other using the "reset" command on stutil?

Benasking7124, 2022-05-14:
Thank you~ 
This is the debug console output after setting `"showDevDebugOutput": "raw"`
```
Reading symbols from /home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-objdump --syms -C -h -w /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf
Reading symbols from /home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf
Launching GDB: /home/ben/stm32_tools/abe_build/builds/destdir/x86_64-pc-linux-gnu/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf
1-gdb-set target-async on
Launching gdb-server: /usr/local/bin/st-util -p 50000 --no-reset
    Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/st-util
Finished reading symbols from objdump: Time: 31 ms
Finished reading symbols from nm: Time: 39 ms
-> =thread-group-added,id="i1"
-> ~"Reading symbols from /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf...\n"
Reading symbols from /home/ben/stm32_ws/CPP_Test/build/CPP_Test.elf...
-> ~"Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.\n"
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
-> ~"Use 'set mi-async'.\n\n"
Use 'set mi-async'.
-> 1^done
2-gdb-version
-> ~"GNU gdb (Linaro_GDB-2019.12) 11.2.90.20220202-git\n"
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-pc-linux-gnu --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-interpreter-exec console "source /home/ben/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdbsupport.init"
-> 5^done
6-interpreter-exec console "source /home/ben/.vscode/extensions/marus25.cortex-debug-1.4.4/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 6^done
7-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"0x08004250 in main () at Core/Src/main.cpp:98\n"
0x08004250 in main () at Core/Src/main.cpp:98
-> ~"98\t    n++;\n"
98	    n++;
-> *stopped,frame={addr="0x08004250",func="main",args=[],file="Core/Src/main.cpp",fullname="/home/ben/stm32_ws/CPP_Test/Core/Src/main.cpp",line="98",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 7^connected
8-interpreter-exec console "monitor halt"
-> 8^done
9-interpreter-exec console "monitor reset"
-> 9^done
10-target-download
-> 10+download,{section=".isr_vector",section-size="392",total-size="210168"}
-> 10+download,{section=".isr_vector",section-sent="392",section-size="392",total-sent="392",total-size="210168"}
-> 10+download,{section=".text",section-size="16944",total-size="210168"}
-> 10+download,{section=".rodata",section-size="24",total-size="210168"}
-> 10+download,{section=".ARM.extab",section-size="48",total-size="210168"}
-> 10+download,{section=".ARM",section-size="208",total-size="210168"}
-> 10+download,{section=".init_array",section-size="4",total-size="210168"}
-> 10+download,{section=".fini_array",section-size="4",total-size="210168"}
-> 10+download,{section=".data",section-size="2124",total-size="210168"}
-> 10^done,address="0x0800434c",load-size="19748",transfer-rate="126184",write-rate="2194"
[... 22 lines cut ...]
-> ~"98\t    n++;\n"
98	    n++;
-> *stopped,reason="signal-received",signal-name="SIGTRAP",signal-meaning="Trace/breakpoint trap",frame={addr="0x08004250",func="main",args=[],file="Core/Src/main.cpp",fullname="/home/ben/stm32_ws/CPP_Test/Core/Src/main.cpp",line="98",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
16-thread-list-ids
-> 16^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
17-thread-info 1
-> 17^done,threads=[{id="1",target-id="Thread <main>",frame={level="0",addr="0x08004250",func="main",args=[],file="Core/Src/main.cpp",fullname="/home/ben/stm32_ws/CPP_Test/Core/Src/main.cpp",line="98",arch="armv7e-m"},state="stopped"}]
18-data-list-register-names
-> 18^done,register-names=["r0","r1","r2","r3","r4","r5","r6","r7","r8","r9","r10","r11","r12","sp","lr","pc","","","","","","","","","","xpsr","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","","msp","psp","control","faultmask","basepri","primask","s0","s1","s2","s3","s4","s5","s6","s7","s8","s9","s10","s11","s12","s13","s14","s15","s16","s1 [... cut ...] ,{number="127",value="0"},{number="128",value="0"},{number="129",value="0"}]
22-stack-select-frame --thread 1 0
-> 22^done
23-stack-list-variables --thread 1 --frame 0 --simple-values
-> 23^done,variables=[{name="n",type="int",value="123209548"}]
24-var-update --thread 1 --frame 0 --all-values var_n_256
-> 24^error,msg="Variable object not found"
25-var-create --thread 1 --frame 0 var_n_256 * "n"
-> 25^done,name="var_n_256",numchild="0",value="123209548",type="int",thread-id="1",has_more="0"
```

Benasking7124, 2022-05-14:
> From the gdb-server, I see
> 
> ```
> 2022-05-14T10:39:30 WARN common.c: NRST is not connected
> ```
> 
> Okay, then not sure a reset is actually doing what it is supposed to do. It is supposed Halt at the Reset Vector. Do you have another way to do this other using the "reset" command on stutil?
> 
> Note: any command sent to gdb that starts with the word "monitor" is sent by gdb directly to the gdb-server -- whichever that is.

I can use `st-flash reset` to reset the board. And the wire I have connected are "SWDIO", "GND, "SWCLK", and "3.3V", do I miss any port that I should connect?

haneefdm, 2022-05-14:
@Benasking7124 The point is that the "monitor reset" command is supposed to reset the device and halt at the reset vector. Instead it already ran the program before we even set a temporary breakpoint in main, so stop in main will never work. That is not a reliable reset+halt mechanism.

```
11-interpreter-exec console "monitor reset"
-> 11^done
12-enable-pretty-printing
-> 12^done
13-break-insert -t --function main
```
Between commands 11 and 13, the FW has already run away past main

You may want to consult the stutil folks on how to halt at the reset vector given your situation. One way to do it perhaps to set the PC, SP, LR to the right/reset values after a 'monitor halt'...meaning doing a manual/partial reset.

haneefdm, 2022-05-14:
Just to be sure you understand from your title "Debugger did not start from main". The debugger never starts at the main. It **always** attempts to start at the reset vector and optionally stops in main. It cannot start at main

Benasking7124, 2022-05-14:
@haneefdm OK I see! Sorry that I miss use the word. I'll ask at the st-link project to see how to halt at the reset vector. Thank you so much for helping me! ~!!

haneefdm, 2022-05-14:
Good luck. If you show them your program and the last trace you posted here, they will know what to do as this something they would be super familiar with. If you do not have a reset pin, they  may be able to show you how to do a SW reset (depending on which version of ARM arch.). Please post the method back here as it can be helpful to others.

Benasking7124, 2022-05-14:
@haneefdm OK!! As soon as I get the solution, I will post it back here~ Thanks again for your help!!

Benasking7124, 2022-05-15:
@haneefdm Sorry to bother you again. I saw in README there is an alternative gdb server to use called "ST-LINK GDB server", provided by ST. I assume it's more stable and reliable. And while searching for it, I found out there is a application called [ST-LINK server](https://www.st.com/en/development-tools/st-link-server.html). 

Can I use this as the "ST-LINK GDB server" in the extension? This way I don't need to download the entire CubeIDE. But I'm not sure how to setup. Can you help me out? Thank you ~

haneefdm, 2022-05-15:
Absolutely. That would ST-Link be the preferred server along with OpenOCD, both supported/distributed by ST. The setup for stlink can be a bit difficult if you don't do it the standard way. Install STMCube32IDE at

Linux: `~/st` or `/opt/st`
Mac: `/Applications`
Windows: `C:/ST`

And you may not even have to setup your paths for the st-link server. Otherwise there are a set of paths for cortex-debug you have to create (like you did for cortex-debug.stutilPath)

When editing your launch.json, use this button (Add Configuration)

[image]

I was wondering why you were using st-util, but some people insist on wanting open source but so is openocd -- officially supported by ST.

Question: Your HW truly does not have a Reset pin?

Benasking7124, 2022-05-15:
Do you know where the "ST-LINK GDB server" is located in "ST-Link server" ? I cannot find any executable name "ST-LINK_gdbserver" if I only install "ST-Link server".  Perhapes it is in a different name?

OMG I don't know that OpenOCD is maintained by ST. I thought it is also an open source library. Do you recommend it over "ST-LINK_gdbserver"?

The reason I use st-util is because at the first time I learn stm32, the senior classmate use the this tools and the video I found online use this tool too. Therefore, I assume it is quite comman to use that.

The emulator I use is this [one](https://hallroad.org/images/detailed/34/ST-LINK_V2_ST_Link_V2_STLINK_V2_Programmer_Lahore_Karachi_Islamabad_Pakistan.jpg). There is a RST pin, but after I connect it to the NRST pin, it did not work.

haneefdm, 2022-05-15:
I prefer STLink over OpenOCD, but OpenOCD gives you more features.

Install STM32CubeIDE (instead of the standalone stlink server) and st-link gdb-server is included in the install. We will find it if you install in the default location. You can always go to the install dir and do a
```sh
cd <install-dir>
find . -name ST-LINK_gdbserver -print
```
You can also look in the Debug Configurations dialog box in Eclipse to see what it is.

They also ship their own version of OpenOCD. Warning: sometimes, they ship multiple versions of OpenOCD. If you use OpenOCD, make sure it is the latest. No idea why they ship multiple versions

No, ST does not maintain OpenOCD. Every device manufacturer adds their own device support to it but the core is maintained by a large set of individual developers (me included) as well. It is used by non-Cortex-MCU vendors as well (RISC-V, FPGAs, Xtensa, etc.)

Benasking7124, 2022-05-16:
I see!! Then I'll use STLink then.

The CubeIDE solution work perfectly! Thank you~ I just thought that ST-Link server contains ST-LINK GDB server! 
May I ask what ST-Link server is for? It seems very similar to "STM32CubeProgrammer". All I can see is that they can flash binary into board and read memory.

OK I will keep that in mind!!
I see!! Thank you for explaining to me ~~

haneefdm, 2022-05-16:
> May I ask what ST-Link server is for? It seems very similar to "STM32CubeProgrammer". All I can see is that they can flash binary into board and read memory.

Confused me too and wasted a LOT of my time a couple of years ago. You have to ask ST :-)

closed 2022-05-16 by haneefdm