Marus/cortex-debug #652: How to define an offset to the base address when flashing my program?

basprins, 2022-05-13:
Hi,

I just received my SEGGER JLink and currently busy migrating from MCUXpresso to VScode with cortex-debug. Everything seems to work. We use position independent code on our board (so that we can rely on over the air (OTA) updates where we can switch between app1 / app2). 

How can I configure my launch target to upload by binary to an offset location ? 

In eclipse I can do something like:

[image]

My current VScode launch configuration is

```
      {
        "type": "cortex-debug",
        "request": "launch",
        "name": "iobox-1020-windows",
        "cwd": "${workspaceRoot}",
        "executable": "${workspaceRoot}/iobox-imx-rt-1020/build/iobox-imx-rt-1020.axf",
        "serverpath": "C:\\Program Files (x86)\\SEGGER\\JLink\\JLinkGDBServerCL.exe", 
        "servertype": "jlink",
        "device": "MIMXRT1021xxx5A",
        "interface": "swd",
        "serialNumber": "", //If you have more than one J-Link probe, add the serial number here.
        //"jlinkscript":"${workspaceRoot}/BSP/SEGGER/K66FN2M0_emPower/Setup/Kinetis_K66_Target.js",
        "runToMain": true,
        "svdFile": "${workspaceRoot}/svd/MIMXRT1021.svd",
        },    
```

haneefdm, 2022-05-14:
Two ways maybe

## Use gdb (everything we do is via gdb)
1. If you have a hex/bin file, you can load that with `loadFiles` in launch.json
2. If not, first convert your ELF file to a bin/hex format using utilities like GNU utilities like objcopy. Keil has fromelf utility

But above may not be enough. Because symbols may also have to be offset
3. You may also have to use the `symbolFiles` in launch.json (must use 1.5.x) to offset the symbols. Or else gdb will think your symbol files are at a different location that your bin/hex files

## Use `jlink` monitor commands and `overrideLaunchCommandfs . I think it is something like `monitor program ...` See JLinik gdb server documentation.

## Use an external script to program the device and run it with `preLaunchTask` and disable load in Cortex-Debug using `loadFiles`

https://stackoverflow.com/questions/49680382/objcopy-elf-to-bin-file

You can also avoid all of this and create a proper hex file in the first place using a linker script.

I don't what you are working with. Enable debug in launch.json so you can detailed transactions between gdb and cortex-debug.

basprins, 2022-05-14:
@haneefdm 
Thx for the reply. 
I have both the .elf file, and I grab the .bin file in my cmake post build steps. 

Can you point to some example of a `launch.json` which uses `loadFiles` and `symbolFiles` and how an offset can be applied?

Is there documentation on this or should I just clone this repo and find the anwers in code? (I was hoping to find some documentation for each property allowed to set in `launch.json`

Thx in advance!

haneefdm, 2022-05-14:
Intellisense is your friend while editing launch.json. Just hover over an attribute, or start typing with a double-quote and you will see help for possible completions, etc.

You may have missed the [Usage secion](https://github.com/Marus/cortex-debug#usage) of our top level readme. You will see a reference to

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

EvilEli, 2022-05-17:
@basprins 
For J-Link this should work when added to launch.json:
```
"overrideLaunchCommands": [
        "load <path-to-elf> <offset>",
        (optional: set $pc=<start-address>)
],
```

However, as @haneefdm already pointed out I'd recommend to add a proper placing in the linkerscript (see [Linker scripts](https://users.informatik.haw-hamburg.de/~krabat/FH-Labor/gnupro/5_GNUPro_Utilities/c_Using_LD/ldLinker_scripts.html#Entry_point)).

haneefdm, 2022-05-17:
I am wondering how a load with offset will even work. The GDB machine interface doesn't even support it. GDB docs say it may not work for all formats. Also, the way most of us compile/link, the executables are *not re-locatable*. Many branches/calls have absolute addresses.

Wondering about use cases:

Okay, you have the case where we store the binary somewhere (say external FLASH) but load it into RAM to run. This is the case where you want to load the file at a different address. This is what LMA and VMA does and this what the linker script magic does including code to copy from LMA to VMA at startup. And all the GNU tools understand this.

haneefdm, 2022-05-17:
> Also when using dedicated bootloaders you might have a usecase:

But, even in that case, the application is compiled with that offset in mind.

basprins, 2022-05-17:
Hi all,
Thanks for the comments and tips. Will try them asap.

I understand the confusion about the need for all this. 

I have a custom bootloader, compiled position dependent. Always resident at 0x6000 0000.

Then i have one app, compiled position independent. Linker defined location is 0x6002000. I can upload that either to it's normal location, or to any other location. In case of the latter, during startup the got section gets patched and the app runs on the offset location. 

My bootloader decided to jump to either app1 or app2.

Anyway, in case i relocated my app, i occasionally also need to debug through stuff. Currently i am doing that by issueing gdb commands (add-symbol-file) where i give the address of relocated text section.

Thats the part i want to have automated in the launch json. I will try the above tips. Thx!

basprins, 2022-05-17:
Ah ok, i do that via Jlink but good to know, then i wont have to look for that :)

Thx again!

haneefdm, 2022-05-17:
> Ah ok, i do that via Jlink but good to know, then i wont have to look for that :)

May I ask what gdb monitor command you use? Or some other way? I can look it up but...

With 1.5.1, you will have the capability to apply offsets to individual sections of an ELF file (what gdb allows). Crossing fingers. Your code offset may be different from data offsets for instance, or you could have multiple code sections (secure, non-secure) with different offsets. The combinations can be mind-boggling.

I may know a bit more than others but I am in no way an expert who knows everything. Do my best to keep this extension going.

basprins, 2022-05-18:
Hi, the only command I use is

`add-symbol-file path-to-my-elf 0x600xxxxx`.

Xxxxx is the originaal linked start address of my app 0x60020000 plus the offset of my text section found in readelf -WS path-to-elf.

Gdb then calculaties the offset and applies it to all text sections i thought. I only have one, so i dont really need more. I think it automatically leaves all vma addressed sections alone. So i dont need to apply offsets to individual section (yet :)).

I am sure i am less expert, 2 months ago i never touched a linker file and never used any gdb command. Haha. And your extension is great. It allows me to delete mcuxpresso all together. And that alone is priceless!

usmanmehmoodSSP, 2023-10-27:
> @basprins For J-Link this should work when added to launch.json:
>
> ```
> "overrideLaunchCommands": [
>         "load <path-to-elf> <offset>",
>         (optional: set $pc=<start-address>)
> ],
> ```
>
> However, [...]

I was having the same issue and this worked for me. Of course I tried all the solutions mentioned here but this is the one that worked. Thanks a lot 😄

haneefdm, 2023-10-27:
You can also use `symbolFiles` to specify offsets.

```JSON
"symbolFiles": [
      { "file": "path to elf", offset: "0x600xxxxx", ... }
]
```

Let Intellisense help you. There are many options. You can even supply offsets for individual sections.

But, this is not how I do it. Because, these tricks are only for gdb and will not for production. I use the linker script to properly set the LMA, VMA addresses and have my startup code copy the code. This is what I do when using external flash, but run code out of internal RAM. Also, I need my program to work without a debugger or with other debuggers.

If you artificially change addresses, then things like disassembly won't work

If you really want to change the offset of your hex file, you should use proper tools to do that during your build process. Like `objcopy`

There are so many situations and use cases, so I can't tell you exactly what methods are best.